espressif/esp-idf #2754: Link failure on Linux (cannot open linker script file)

detly, 2018-11-28:
## Environment

- Development Kit: none
- Module or chip used: ESP32-WROOM-32
- IDF version: v3.3-dev-236-gfa59b1b1c
- Build System: CMake
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux (Ubuntu 18.04 amd64)
- Power Supply: USB

## Problem Description

Trying to build any project eg. the `hello_world` example project fails with:

```
hello_world $ idf.py fullclean && idf.py build
...
ld: cannot open linker script file /home/me/Code/esp-libs/esp-idf/examples/get-started/hello_world/build/esp32/esp32.common.ld: No such file or directory
```

I notice this error earlier in the build process:

```
esptool.py v2.6-beta1
[460/750] Generating esp32.common.ld
linker script generation failed for /home/hearing.net/heerij/Code/esp-libs/esp-idf/components/esp32/ld/esp32.common.ld.in
```

Git bisect output:

```
63411fc556fb2e71985db50dacbf42a6532b3993 is the first bad commit
commit 63411fc556fb2e71985db50dacbf42a6532b3993
Author: Renz Bagaporo <renz@espressif.com>
Date:   Wed Apr 18 10:57:45 2018 +0800

    tools: implement linker script generation

:100644 100644 73c2cf577e8fdfcf13374c1428c9072f6c70bb40 4bfea8f6471a0c2e62cd53ba2652e0fe21e4dd15 M	.gitlab-ci.yml
:040000 040000 129a171d511a8363bcc1f39ed32ca4717a13962f 8a90992678b62f9c03624217e616bc97d9f51b16 M	components
:040000 040000 d8d55e907b021a4bb02db9054fd1e405e1c1d90c 09b07429ffebde3ba24eb7c2d1562e1934ea6318 M	docs
:040000 040000 5ea18803d05061c350347c2c3c8d3bb12db3bea0 b3afe2c2fa4cd49a893b782f7e4097d9b76be0e8 M	make
:040000 040000 834d8a06e2abc91fe07ee26f896eb3539df6b3b8 6d08c1d2cf9819ea5c28b873b410e404a9e4938a M	tools
```

This doesn't make a lot of sense to me, since it was working two weeks ago.

[idf_build_log.txt](https://github.com/espressif/esp-idf/files/2622633/idf_build_log.txt)

The full build log is attached. Ignore the warnings about out-of-date submodules, that's a result of bisecting. It happens even when everything is up to date as well.

projectgus, 2018-11-28:
Thanks for the comprehensive bug report, @detly

The linker script generation feature was under development since earlier this year but was only merged recently (in 8915f482089d2666d81174e9d671af300036472b), so it seems like that's the trigger of the failures.

The linker script generation failure you point out should be failing the build at that stage (as the linker script is required for linking), but isn't failing it until link time.

It's not clear from the log what the root cause of the actual linker script generation failure is. Could you please run `idf.py -v fullclean build` and post the log from this as well?

detly, 2018-11-28:
@projectgus I'll do that soon (I'm working out of Windows right now). But I'm also curious as to whether you or anyone else can reproduce the issue?

projectgus, 2018-11-28:
No, we haven't reproduced this yet (including on our CI runners which are all Linux-based.) I know some Espressif developers use Ubuntu 18.04 as well.

detly, 2018-11-28:
Just a hunch, but what architecture are your runners?

detly, 2018-11-28:
Here's the log from `idf.py -v fullclean build`.

[idf_v_build_log.txt](https://github.com/espressif/esp-idf/files/2626934/idf_v_build_log.txt)

detly, 2018-11-30:
Just about to do a PR to fix this and I notice it's now working on `master`! Great work, thanks.

projectgus, 2018-11-30:
Excellent news. OK if we close this issue, then?

Another fix is coming shortly which will make these parser errors output correctly in the build log, hopefully shouldn't need pdb from now on.

detly, 2018-11-30:
Yep, go ahead and close it.

closed 2018-11-30 by projectgus