espressif/esp-idf #2755: error: while compiling the project from mingw32 or make -j4 from eclipse

KrrishnaShah, 2018-11-28:
xtensa-esp32-elf-gcc -std=gnu99 -Og -ggdb -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -DESP_PLATFORM -D IDF_VER=\"v3.1.1-dirty\" -MMD -MP   -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -I /c/esp32/esp-idf-v3.1.1/components/bootloader_support/include_priv -I /c/esp32/esp-idf-v3.1.1/components/bootloader_support/include -I /c/esp32/esp-idf-v3.1.1/components/app_trace/include -I /c/esp32/esp-idf-v3.1.1/components/app_update/include -I /c/esp32/esp-idf-v3.1.1/components/console -I /c/esp32/esp-idf-v3.1.1/components/driver/include -I /c/esp32/esp-idf-v3.1.1/components/esp32/include -I /c/esp32/esp-idf-v3.1.1/components/esp_adc_cal/include -I /c/esp32/esp-idf-v3.1.1/components/ethernet/include -I /c/esp32/esp-idf-v3.1.1/components/expat/port/include -I /c/esp32/esp-idf-v3.1.1/components/expat/include/expat -I /c/esp32/esp-idf-v3.1.1/components/fatfs/src -I /c/esp32/esp-idf-v3.1.1/components/freertos/include -I /c/esp32/esp-idf-v3.1.1/components/heap/include -I /c/esp32/esp-idf-v3.1.1/components/idf_test/include -I /c/esp32/esp-idf-v3.1.1/components/jsmn/include -I /c/esp32/esp-idf-v3.1.1/components/json/cJSON -I /c/esp32/esp-idf-v3.1.1/components/libsodium/libsodium/src/libsodium/include -I /c/esp32/esp-idf-v3.1.1/components/libsodium/port_include -I /c/esp32/esp-idf-v3.1.1/components/log/include -I /c/esp32/esp-idf-v3.1.1/components/lwip/include/lwip -I /c/esp32/esp-idf-v3.1.1/components/lwip/include/lwip/port -I /c/esp32/esp-idf-v3.1.1/components/lwip/include/lwip/posix -I /c/esp32/esp-idf-v3.1.1/components/lwip/apps/ping -I /m/eclipse-workspace/Infrastructure_Monitoring_System/main/include -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/devices/sht3x -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/devices-lib/rtc-pcf85063tp -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/devices-lib/vs1053b -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/esp32_wrapper/fio -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/esp32_wrapper/i2c -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/esp32_wrapper/spi -I /m/eclipse-workspace/Infrastructure_Monitoring_System/Infrastructure/platform -I /m/eclipse-workspace/Infrastructure_Monitoring_System/Infrastructure/device -I /m/eclipse-workspace/Infrastructure_Monitoring_System/Infrastructure/device/flow-sensor -I /m/eclipse-workspace/Infrastructure_Monitoring_System/Infrastructure/device/rtc-memory -I /m/eclipse-workspace/Infrastructure_Monitoring_System/Infrastructure/device-manager -I /m/eclipse-workspace/Infrastructure_Monitoring_System/Infrastructure/infrastructure-common -I /m/eclipse-workspace/Infrastructure_Monitoring_System/Infrastructure/infrastructure-common/infrastructure-task -I /m/eclipse-workspace/Infrastructure_Monitoring_System/Infrastructure/infrastructure-common/infrastructure-task/task -I /m/eclipse-workspace/Infrastructure_Monitoring_System/Infrastructure/infrastructure-common/jsmn-app -I /m/eclipse-workspace/Infrastructure_Monitoring_System/Infrastructure/infrastructure-common/local-server -I /m/eclipse-workspace/Infrastructure_Monitoring_System/Infrastructure/infrastructure-common/battery-monitor -I /m/eclipse-workspace/Infrastructure_Monitoring_System/Infrastructure/infrastructure-common/settings-common -I /m/eclipse-workspace/Infrastructure_Monitoring_System/Infrastructure/infrastructure-common/tss -I /m/eclipse-workspace/Infrastructure_Monitoring_System/Infrastructure/infrastructure-common/udi-app -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/general -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/general/aes -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/general/debug -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/general/geolocation -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/general/jsmn -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/general/mbedtls -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/general/mqtt -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/general/smtp -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/general/sntp -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/general/communicator -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/general/udi -I /m/eclipse-workspace/Infrastructure_Monitoring_System/rts_lib/general/webserver -I /c/esp32/esp-idf-v3.1.1/components/mbedtls/port/include -I /c/esp32/esp-idf-v3.1.1/components/mbedtls/mbedtls/include -I /c/esp32/esp-idf-v3.1.1/components/mdns/include -I /c/esp32/esp-idf-v3.1.1/components/micro-ecc/micro-ecc -I /c/esp32/esp-idf-v3.1.1/components/newlib/platform_include -I /c/esp32/esp-idf-v3.1.1/components/newlib/include -I /c/esp32/esp-idf-v3.1.1/components/nvs_flash/include -I /c/esp32/esp-idf-v3.1.1/components/pthread/include -I /c/esp32/esp-idf-v3.1.1/components/sdmmc/include -I /c/esp32/esp-idf-v3.1.1/components/smartconfig_ack/include -I /c/esp32/esp-idf-v3.1.1/components/soc/esp32/include -I /c/esp32/esp-idf-v3.1.1/components/soc/include -I /c/esp32/esp-idf-v3.1.1/components/spi_flash/include -I /c/esp32/esp-idf-v3.1.1/components/spiffs/include -I /c/esp32/esp-idf-v3.1.1/components/tcpip_adapter/include -I /c/esp32/esp-idf-v3.1.1/components/ulp/include -I /c/esp32/esp-idf-v3.1.1/components/vfs/include -I /c/esp32/esp-idf-v3.1.1/components/wear_levelling/include -I /c/esp32/esp-idf-v3.1.1/components/wpa_supplicant/include -I /c/esp32/esp-idf-v3.1.1/components/wpa_supplicant/port/include -I /c/esp32/esp-idf-v3.1.1/components/xtensa-debug-module/include -I /m/eclipse-workspace/Infrastructure_Monitoring_System/build/include  -I src -c /c/esp32/esp-idf-v3.1.1/components/bootloader_support/src/secure_boot.c -o src/secure_boot.o
xtensa-esp32-elf-ar cru libconsole.a linenoise/linenoise.o argtable3/argtable3.o commands.o split_argv.o
make[1]: Leaving directory '/m/eclipse-workspace/Infrastructure_Monitoring_System/build/console'
/usr/bin/make -C /m/eclipse-workspace/Infrastructure_Monitoring_System/build/driver -f /c/esp32/esp-idf-v3.1.1/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/esp32/esp-idf-v3.1.1/components/driver/component.mk COMPONENT_NAME=driver build
In file included from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\random:49:0,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algo.h:66,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\algorithm:62,
                 from C:/esp32/esp-idf-v3.1.1/components/cxx/cxx_guards.cpp:20:
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\random.h: In constructor 'std::bernoulli_distribution::param_type::param_type(double)':
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\random.h:3585:54: error: '_GLIBCXX_DEBUG_ASSERT' was not declared in this scope
  _GLIBCXX_DEBUG_ASSERT((_M_p >= 0.0) && (_M_p <= 1.0));
                                                      ^
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\random.h: In constructor 'std::geometric_distribution<_IntType>::param_type::param_type(double)':
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\random.h:4025:54: error: there are no arguments to '_GLIBCXX_DEBUG_ASSERT' that depend on a template parameter, so a declaration of '_GLIBCXX_DEBUG_ASSERT' must be available [-fpermissive]
    _GLIBCXX_DEBUG_ASSERT((_M_p > 0.0) && (_M_p < 1.0));
                                                      ^
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\random.h:4025:54: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\random.h: In constructor 'std::poisson_distribution<_IntType>::param_type::param_type(double)':
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\random.h:4447:39: error: there are no arguments to '_GLIBCXX_DEBUG_ASSERT' that depend on a template parameter, so a declaration of '_GLIBCXX_DEBUG_ASSERT' must be av [... cut ...] t/port/include -I /c/esp32/esp-idf-v3.1.1/components/xtensa-debug-module/include -I /m/eclipse-workspace/Infrastructure_Monitoring_System/build/include  -I . -c /c/esp32/esp-idf-v3.1.1/components/driver/rtc_module.c -o rtc_module.o
make[1]: *** [/c/esp32/esp-idf-v3.1.1/make/component_wrapper.mk:286: cxx_guards.o] Error 1
make[1]: Leaving directory '/m/eclipse-workspace/Infrastructure_Monitoring_System/build/cxx'
make: *** [C:\esp32\esp-idf-v3.1.1/make/project.mk:468: component-cxx-build] Error 2

KrrishnaShah, 2018-11-28:
More clear IMAGE:
----------------------
[image]

KrrishnaShah, 2018-11-28:
More Clear Image with red color error:
-------------------------------------------
[image]

jas39, 2018-11-30:
Building on more than one core fails with new linking process in eclipse. Use -j1 instead

KrrishnaShah, 2018-12-02:
problem has been solved.