espressif/esp-idf #2757: Change BLUETOOTH Passkey programmatically

lettierimarco, 2018-11-28:
I'm testing the "new" feature to set passkey to BLE.
All works (i'm basing on security gatts example) but, if i try to change the pass key from bonded device i have several issues.

```
uint32_t newPassKey = 123457;
esp_ble_gap_set_security_param(ESP_BLE_SM_SET_STATIC_PASSKEY, &newPassKey, sizeof(uint32_t));
remove_all_bonded_devices();
esp_ble_gatts_close(gl_profile_tab[PROFILE_A_APP_ID].gatts_if, gl_profile_tab[PROFILE_A_APP_ID].conn_id);
```

After this the client can anyway connect to server (**it connects also with no password**) and i receive some errors like:

```
I (19516) BLE: pair status = fail
I (19526) BLE: fail reason = 0x51
W (19526) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB
```

```

E (36436) BT_BTM: Device not found
W (36436) BT_APPL: BTA got unregistered event id 31
W (36436) BT_APPL: BTA got unregistered event id 31
```

Weijian-Espressif, 2018-11-29:
@lettierimarco, Please describe your test process in detail.

lettierimarco, 2018-11-29:
@Weijian-Espressif 
THE BLE CONFIGURATION:
`uint32_t newPassKey = 123456;
esp_ble_auth_req_t auth_req = ESP_LE_AUTH_BOND;     //bonding with peer device after authentication
	esp_ble_io_cap_t iocap = ESP_IO_CAP_OUT;           //set the IO capability to No output No input
	uint8_t key_size = 16;      //the key size should be 7~16 bytes
	uint8_t init_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
	uint8_t rsp_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
	//set static passkey
	uint8_t auth_option = ESP_BLE_ONLY_ACCEPT_SPECIFIED_AUTH_DISABLE;
	//uint8_t oob_support = ESP_BLE_OOB_DISABLE;
	esp_ble_gap_set_security_param(ESP_BLE_SM_SET_STATIC_PASSKEY, &passKey, sizeof(uint32_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &auth_req, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_MAX_KEY_SIZE, &key_size, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_ONLY_ACCEPT_SPECIFIED_SEC_AUTH, &auth_option, sizeof(uint8_t));
	//esp_ble_gap_set_security_param(ESP_BLE_SM_OOB_SUPPORT, &oob_support, sizeof(uint8_t));
	/* If your BLE device act as a Slave, the init_key means you hope which types of key of the master should distribut to you,
	and the response key means which key you can distribut to the Master;
	If your BLE device act as a master, the response key means you hope which types of key of the slave should distribut to you,
	and the init key means which key you can distribut to the slave. */
	esp_ble_gap_set_security_param(ESP_BLE_SM_SET_INIT_KEY, &init_key, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_SET_RSP_KEY, &rsp_key, sizeof(uint8_t));
	for(int i=0;i<ESP_BLE_PWR_TYPE_NUM;i++){
		ret = esp_ble_tx_power_set(i, ESP_PWR_LVL_P9);
		if (ret) {
			ESP_LOGE(GATTS_TABLE_TAG, "%i Unable to set BLE Tx power level. err: %x \n", i,ret);
			return;
		}
	}`

Try to connect via nRF connect app for Android.
When connected you'll have a prompt for "Associate" and input the passkey.
If you click "Associate" and leave empty the passkey and press "CANCEL", you receive CONNECT_EVT and this log:
```
I (33234) BLE: pair status = fail
I (33234) BLE: fail reason = 0x63
```
but the connection is up and you're able to write on characteristics.

So i change the passkey via:
```
uint32_t newPassKey = 123457;
esp_ble_gap_set_security_param(ESP_BLE_SM_SET_STATIC_PASSKEY, &newPassKey, sizeof(uint32_t));
remove_all_bonded_devices();
esp_ble_gatts_close(gl_profile_tab[PROFILE_A_APP_ID].gatts_if, gl_profile_tab[PROFILE_A_APP_ID].conn_id);
```
I try with 123456 and i connect correctly, so the passkey doesn't change.

Weijian-Espressif, 2018-11-29:
@lettierimarco 
1. After Bluetooth connection, You can  decide whether to start encryption. even if the encryption fails, it will not affect the Bluetooth connection.
2. Please set static passkey when Bluetooth initializes. If you clear the bonded keys, please delete the pairing record in the phone Bluetooth settings.

lettierimarco, 2018-11-29:
the second, is solved,thanks.

For the first the problem still live.
i've setted the encryption on connect:
` esp_ble_set_encryption(param->connect.remote_bda, ESP_BLE_SEC_ENCRYPT_MITM);
`
but i try to set  notify  after connection and it passes, and after i receive GATT_INSUF_AUTHENTICATION.
`
I (5519) BLE: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 7a:0e:1c:25:cb:80:
W (5519) BT_L2CAP: l2cble_start_conn_update, the last connection update command still pending.
I (6569) BLE: GATT_WRITE_EVT, conn_id 0, trans_id 1, handle 43 prepare 0
I (6569) BLE: notify enable
E (6679) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION
I (35529) BLE: remote BD_ADDR: 7e0e1e25bb80
I (35529) BLE: address type = 1
I (35529) BLE: pair status = fail
I (35529) BLE: fail reason = 0x63
W (35539) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB
I (35549) BLE: ESP_GATTS_DISCONNECT_EVT
E (35679) BT_BTM: Device not found
W (35679) BT_APPL: BTA got unregistered event id 31
```

chegewara, 2018-11-29:
Do you have setup all characteristics and descriptors with permission:
`ESP_GATT_PERM_READ_ENCRYPTED` and/or `ESP_GATT_PERM_WRITE_ENCRYPTED`

lettierimarco, 2018-11-29:
ESP_GATT_PERM_READ and ESP_GATT_PERM_WRITE_ENCRYPTED, because 
 seems ESP_GATT_PERM_READ_ENCRYPTED has some problem with iOS
```
    case ESP_GATTS_CREATE_EVT:
        //ESP_LOGI(GATTS_TABLE_TAG, "CREATE_SERVICE_EVT, status %d,  service_handle %d\n", param->create.status, param->create.service_handle);
        gl_profile_tab.service_handle = param->create.service_handle;
        gl_profile_tab.char_uuid.len = ESP_UUID_LEN_16;
        gl_profile_tab.char_uuid.uuid.uuid16 = GATTS_CHAR_UUID_TEST_A;

        esp_ble_gatts_start_service(gl_profile_tab.service_handle);
        a_property = ESP_GATT_CHAR_PROP_BIT_READ | ESP_GATT_CHAR_PROP_BIT_WRITE | ESP_GATT_CHAR_PROP_BIT_NOTIFY;
        esp_err_t add_char_ret = esp_ble_gatts_add_char(gl_profile_tab.service_handle, &gl_profile_tab.char_uuid,
        												ESP_GATT_PERM_WRITE_ENCRYPTED|ESP_GATT_PERM_READ,
                                                        a_property,
                                                        &gatts_demo_char1_val, NULL);
        if (add_char_ret){
            ESP_LOGE(GATTS_TABLE_TAG, "add char failed, error code =%x",add_char_ret);
        }
        break;
```

chegewara, 2018-11-29:
Remember to set this to CCC descriptor too.

lettierimarco, 2018-11-29:
thanks @chegewara
solved

closed 2018-11-29 by lettierimarco