espressif/esp-idf #2758: NDEBUG and assert() old change landed in esp32-arduino / breaks sqlite3

tobozo, 2018-11-28:
- Development Kit:      ESP32-Wrover-Kit v4
- IDF version : the one bundled with Arduino 1.0.1-rc
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
- Operating System:     [Windows/Linux]
- xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0
- esp-idf version : bundled with arduino-esp32 1.0.1-rc (built on nov 28th, yesterday)

### Summary
The *new* version of 'assert.h' (although last change was in January 2017) just landed in Arduino-esp32 and somehow breaks compilation for [sqlite3 library](https://github.com/siara-cc/esp32_arduino_sqlite3_lib/)

@siara-cc found a [workaround](https://gitter.im/espressif/arduino-esp32?at=5bfe9a7ff59704348e197fc8) but @me-no-dev suggests we create an issue here.

### Expected Behavior
Maybe this is the expected behaviour, please advise what approach is best for external lib (e.g. implement own version of assert ?)

### Actual Behavior
Can't compile a sketch with the library using the Arduino IDE bundled with the latest esp-idf.
Many errors come up like this one, all pointing to assert.h:

    ~/.arduino15/packages/esp32/hardware/esp32/1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
    # define assert(__e) ((void) sizeof(__e))
                                     ^

### Steps to repropduce

1. Arduino IDE with the latest ESP32 SDK installed from the boards manager
2. Try to compile the [SPIFFS example](https://github.com/siara-cc/esp32_arduino_sqlite3_lib/blob/master/examples/sqlite3_spiffs/sqlite3_spiffs.ino) (or any other)

## Debug Logs

<details>
  <summary> Compile log (...)</summary>

```
In file included from ~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:13476:0:
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'pcache1TruncateUnsafe':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:48838:11: error: 'nPage' undeclared (first use in this function)
   assert( nPage<0 || pCache->nPage==(unsigned)nPage );
           ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                    ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:48838:11: note: each undeclared identifier is reported only once for each function it appears in
   assert( nPage<0 || pCache->nPage==(unsigned)nPage );
           ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'pcache1Fetch':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:49201:11: error: 'pCache' undeclared (first use in this function)
   assert( pCache->bPurgeable || createFlag!=1 );
           ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'sqlite3PagerUnrefNotNull':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:55770:33: error: 'pPager' undeclared (first use in this function)
   assert( sqlite3PcacheRefCount(pPager->pPCache)>0 );
                                 ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'cellSizePtr':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:63969:18: error: 'debuginfo' undeclared (first use in this function)
   assert( nSize==debuginfo.nSize || CORRUPT_DB );
                  ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'cellSizePtrNoPayload':
[... 158 lines cut ...]
               ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:106230:11: error: 'nLookaside' undeclared (first use in this function)
   assert( nLookaside==0 || nLookaside==sqlite3LookasideUsed(db,0) );
           ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'sqlite3InitCallback':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:122300:25: error: 'rcp' undeclared (first use in this function)
     assert( (rc&0xFF)==(rcp&0xFF) );
                         ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
```

</details>
<br><br>

As a reference, the initial problem description started here : https://github.com/siara-cc/esp32_arduino_sqlite3_lib/issues/2 (or see debug log below)

Ping @siara-cc (maintainer of the project)

siara-cc, 2018-11-28:
The problem seems to be that `newlib/assert.h` of `1.0.1-rc` has changed since `1.0.0`.

Specifically `1.0.1-rc assert.h` defines `assert()` as `# define assert(__e) ((void) sizeof(__e))` at line number 14. Since NDEBUG is defined, this makes assert evaluate the expression and throws compilation error.

`1.0.0` does not have the problem because if NDEBUG is defined, `assert()` is not evaluated.. line number 25 of `1.0.0`.

We can say that the assert expressions in `sqlite3.c` should not have error when NDEBUG is defined, but since assert() is a compile time macro, I think this is argumentative.

Problem occurs at Line number 48800 of sqlite3.c and many other places as shown below:
```
  TESTONLY( int nPage = 0; )  // nPage is defined only when NDEBUG is not defined
  ...
  ...
  // But assert tries to evaluate nPage and throws compilation error because NDEBUG is defined.
  assert( nPage<0 || pCache->nPage==(unsigned)nPage );
```

siara-cc, 2018-12-31:
Please see also https://github.com/espressif/esp-idf/issues/2901

closed 2024-08-26 by espressif-bot