espressif/esp-idf #2759: native_ota_example: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT

James-sjec, 2018-11-28:
[image]

I tried with : simple_ota_example & native_ota_example  but I get the same result

James-sjec, 2018-11-28:
server IP address = https://192.168.43.79:8070/hello-world.bin
I am able to download the bin file from browser

chegewara, 2018-11-28:
`https://192.168.43.79:8070/hello-world.bin`
Do you have server with generated ca certificate and did you provide certificate in ota example code?

James-sjec, 2018-11-28:
I run the following cmd in the example folder :  espressif\hello_world\build
"openssl req -x509 -newkey rsa:2048 -keyout ca_key.pem -out ca_cert.pem -days 365 -nodes"
and started the server with the cmd 
" openssl s_server -WWW -key ca_key.pem -cert ca_cert.pem -port 8070"

Next  copied the "ca_cert.pem" file
from espressif\hello_world\build to : espressif\native_ota_example\server_certs location 
and flashed the example code

jitin17, 2018-12-04:
@James-sjec I reckon the problem is with your openssl server. Try the python HTTPS server shared in the thread of this issue: #2625. Additionally, can you tell what is the version of your openssl?

James-sjec, 2018-12-16:
I use openssl verison = "OpenSSL 1.0.2l  25 May 2017"

I installed esp-idf from beginning , currently I am using version 3.1 , now I do not get HTTP connection issue and able to update the firmware.

I integrated the simple ota example code with my application firmware.
I had issue with mbedtls : Memory allocation failed at funtions "mbedtls_ssl_setup" , so reduced the "TLS maximum message content length" from 16K to 4K , and memory allocation issue got solved. 
But I get the following issue , I don't known  how to processed further

I (24231) https_ota: Starting OTA example...
D (24231) HTTP_CLIENT: Begin connect to: https://192.168.43.79:8070
D (24241) TRANS_SSL: Connect to 192.168.43.79:8070
D (24261) TRANS_SSL: Performing the SSL/TLS handshake...
D (30871) TRANS_SSL: Verifying peer X.509 certificate...
D (30871) TRANS_SSL: Certificate verified.
D (30871) TRANS_SSL: Cipher suite is TLS-ECDHE-RSA-WITH-AES-256-GCM-SHA384
D (30871) https_ota: HTTP_EVENT_ON_CONNECTED
D (30881) HTTP_CLIENT: Write header[3]: GET /hello-world.bin HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: 192.168.43.79
Content-Length: 0

I (30921) esp_https_ota: Starting OTA...
I (30921) esp_https_ota: Writing to partition subtype 16 at offset 0x190000
I (30921) esp_https_ota: esp_ota_begin succeeded
I (30921) esp_https_ota: Please Wait. This may take time
D (30921) HTTP_CLIENT: is_data_remain=0, is_chunked=0
I (30931) esp_https_ota: Connection closed,all data received
D (30931) https_ota: HTTP_EVENT_DISCONNECTED
D (30941) TRANS_SSL: Cleanup mbedtls
D (30941) https_ota: HTTP_EVENT_DISCONNECTED
D (30951) esp_https_ota: Total binary data length writen: 0
E (30951) esp_https_ota: Error: esp_ota_end failed! err=0x258. Image is invalid
E (30961) https_ota: Firmware Upgrades Failed
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Core 0 register dump:
PC      : 0x400e4b54  PS      : 0x00060230  A0      : 0x00000000  A1      : 0x3fff6f40
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800e4b54  A9      : 0x3fff6ef0
A10     : 0x400de1c0  A11     : 0x3ffc50c0  A12     : 0x3f4237c0  A13     : 0x0000001f
A14     : 0x00000001  A15     : 0x00000005  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd

Backtrace: 0x400e4b54:0x3fff6f40 0x7ffffffd:0x3fff6fb0

Rebooting...
ets Jun  8 2016 00:22:57

and I also tried reducesing  the "TLS maximum message content length" from 16K to 4K in the simple ota example I get the following error : 

I (2808) simple_ota_example: Connect to Wifi ! Start to Connect to Server....
D (2818) HTTP_CLIENT: Begin connect to: https://192.168.43.79:8070
D (2828) TRANS_SSL: Connect to 192.168.43.79:8070
D (3128) TRANS_SSL: Performing the SSL/TLS handshake...
D (9778) TRANS_SSL: Verifying peer X.509 certificate...
D (9778) TRANS_SSL: Certificate verified.
D (9778) TRANS_SSL: Cipher suite is TLS-ECDHE-RSA-WITH-AES-256-GCM-SHA384
D (9778) simple_ota_example: HTTP_EVENT_ON_CONNECTED
D (9788) HTTP_CLIENT: Write header[3]: GET /hello-world.bin HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: 192.168.43.79
Content-Length: 0

I (9828) esp_https_ota: Starting OTA...
I (9828) esp_https_ota: Writing to partition subtype 16 at offset 0x190000
I (9828) esp_https_ota: esp_ota_begin succeeded
I (9828) esp_https_ota: Please Wait. This may take time
D (9828) HTTP_CLIENT: is_data_remain=0, is_chunked=0
I (9838) esp_https_ota: Connection closed,all data received
D (9838) simple_ota_example: HTTP_EVENT_DISCONNECTED
D (9848) TRANS_SSL: Cleanup mbedtls
D (9848) simple_ota_example: HTTP_EVENT_DISCONNECTED
D (9858) esp_https_ota: Total binary data length writen: 0
E (9858) esp_https_ota: Error: esp_ota_end failed! err=0x258. Image is invalid
E (9868) simple_ota_example: Firmware Upgrades Failed

Is it necessary to have "TLS maximum message content length"  to 16K for ota update to work and help me to fix the above issue 

After integrating ota example with my code . my ESP-WROOM-32  ram usage : 

Used DATA_FLASH: 273KB out of 8192KB (3%)
Used INSTR_FLASH: 1061KB out of 3264KB (32%)
Used INSTR_RAM: 87KB out of 128KB (68%)
Used DATA_RAM: 73KB out of 320KB (22%)

mahavirj, 2018-12-20:
> Is it necessary to have "TLS maximum message content length" to 16K for ota update to work and help me to fix the above issue

This depends on server configuration, and generally it is recommended to keep incoming TLS fragment length to 16384, however it is possible to tune outgoing fragment length. Please try enabling config option at https://github.com/espressif/esp-idf/blob/master/components/mbedtls/Kconfig#L58

mahavirj, 2019-03-05:
Closing, assuming no further questions here.

closed 2019-03-05 by mahavirj