espressif/esp-idf #2761: sntp example unit test failure

fullerth, 2018-11-29:
## Environment

- Development Kit: N/A, issue is with the build
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used:  ESP-32
- IDF version (run ``git describe --tags`` to find it): v3.3-dev-278-gcc5673435, but has existed for a long time
- Build System: Make TESTS_ALL=1
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:  Linux
- Power Supply: N/A this is a build issue

## Problem Description
When building with `make TESTS_ALL=1` to attempt to build the unit test app it fails with the following error:
`esp-idf/components/app_update/test/test_ota_ops.c:9:24: fatal error: test_utils.h: No such file or directory`

### Expected Behavior
Build succeeds

### Actual Behavior
Build fails

### Steps to repropduce
1. Get esp-idf
2. Navigate to `esp-idf/examples/protocols/sntp/`
3. Run `make TESTS_ALL=1`
4. Exit the menuconfig
5. Observe the build fail with the build being unable to find `test_utils.h`

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue
No code needed, happens with the default example.

I've tried it with both the `Include ESP-IDF test registration/running helpers` enabled and disabled, as well as the `Include Unity test fixture` enabled and disabled.

## Debug Logs
N/A 
## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2627606/sdkconfig.txt)

KonstantinKondrashov, 2018-11-29:
Hi @fullerth.  If you want to build app_update tests. You need:
1) `cd ~/esp/esp-idf/tools/unit-test-app/`
2) make TEST_COMPONENTS=app_update
3) make flash monitor

Alvin1Zhang, 2019-03-22:
@fullerth Hi, would you help share if any updates for the issue? Thanks.

Alvin1Zhang, 2019-04-18:
@fullerth Feel free to reopen if the issue still exists, thanks.

closed 2019-04-18 by Alvin1Zhang