espressif/esp-idf #2763: Strapping pins and circuit design

Victor-ghcn, 2018-11-29:
Hi,

According to the specification: 
• Firmware can configure register bits to change the settings of ”Voltage of Internal LDO (VDD_SDIO)” and ”Timing of SDIO Slave”, after booting.

Suppose the voltage levels at MTDO and GPIO5 are affected by the external circuitry such that wrong timing of slave SDIO is selected, will it be corrected after booting? Is that carried out inside the bootloader?

Best Regards,
Victor

igrr, 2018-11-29:
This depends on the desired boot mode of the ESP32.

If the ESP32 is supposed to run an application from flash, then you need to ensure that VDD_SDIO voltage is correct at boot time. If the voltage is incorrect, loading of application from flash will likely fail, and the application will not have a chance to reconfigure the registers.

On the other hand, if the ESP32 is supposed to start in SDIO download mode (where the host MCU uploads some code into ESP32 RAM) then SDIO timing configuration needs to be correct at boot time, otherwise the host will not be able to communicate with the ESP32.

Victor-ghcn, 2018-12-03:
Noted with thanks!

closed 2018-12-03 by Victor-ghcn