espressif/esp-idf #2766: stack smashing + wifi + ble = crash

chegewara, 2018-11-30:
With stack smashing = NONE app is working, stack smashing = NORMAL i have crash:

```
I (424) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
Heap summary for capabilities 0x00000400:
  At 0x3f800000 len 4194303 free 2124268 allocated 2069160 min_free 2124184
    largest_free_block 2124268 alloc_blocks 209 free_blocks 1 total_blocks 210
  Totals:
    free 2125964 allocated 2067460 min_free 2124184 largest_free_block 2124268
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x40094484  PS      : 0x00060034  A0      : 0x80092eda  A1      : 0x3ffbe1f0
0x40094484: uxPortCompareSetExtram at D:/msys32/home/imper/esp/esp-idf/components/freertos/por
t.c:452

A2      : 0x3f8018bc  A3      : 0xb33fffff  A4      : 0x3ffbe22c  A5      : 0xbad00bad
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000002  A11     : 0x3ffe32c0  A12     : 0x80085ff7  A13     : 0x3ffe32c0
A14     : 0x0000a560  A15     : 0x3ffe347c  SAR     : 0x00000017  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4009beb5  LEND    : 0x4009becf  LCOUNT  : 0xfffffffc
0x4009beb5: strncpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/
xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strncp
y.S:180

0x4009becf: strncpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/
xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strncp
y.S:206

Core 0 was running in ISR context:
EPC1    : 0x4009909e  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40094484
0x4009909e: esp_rom_spiflash_read_data at D:/msys32/home/imper/esp/esp-idf/components/spi_flas
h/spi_flash_rom_patch.c:412

0x40094484: uxPortCompareSetExtram at D:/msys32/home/imper/esp/esp-idf/components/freertos/por
t.c:452

Backtrace: 0x40094484:0x3ffbe1f0 0x40092ed7:0x3ffbe220 0x40086ffa:0x3ffbe250 0x40088a15:0x3ffb
e270 0x4008810d:0x3ffbe2b0 0x4008b3f6:0x3ffbe2d0 0x40084e11:0x3ffbe300 0x4009909b:0x00000000
0x40094484: uxPortCompareSetExtram at D:/msys32/home/imper/esp/esp-idf/components/freertos/por
t.c:452

0x40092ed7: vPortCPUAcquireMutexIntsDisabledExtram at D:/msys32/home/imper/esp/esp-idf/compone
nts/freertos/tasks.c:4466
 (inlined by) vPortCPUAcquireMutexIntsDisabled at D:/msys32/home/imper/esp/esp-idf/components/
freertos/portmux_impl.h:95
 (inlined by) vTaskEnterCritical at D:/msys32/home/imper/esp/esp-idf/components/freertos/tasks
.c:4258

0x40086ffa: wifi_int_disable_wrapper at D:/msys32/home/imper/esp/esp-idf/components/esp32/wifi
_os_adapter.c:187

0x40088a15: pp_post at ??:?

0x4008810d: lmacPostTxComplete at ??:?

0x4008b3f6: wDev_ProcessFiq at ??:?

0x40084e11: _xt_lowint1 at D:/msys32/home/imper/esp/esp-idf/components/freertos/xtensa_vectors
.S:1105

0x4009909b: esp_rom_spiflash_read_data at D:/msys32/home/imper/esp/esp-idf/components/spi_flas
h/spi_flash_rom_patch.c:412

CPU halted.
```

TianHao-Espressif, 2018-12-20:
@chegewara , do you use PSRAM and enable CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY in menuconfig? It's a cache access error while cache is diable.
Could you please provide your sdkconfig?
@TimXia please take a look.

chegewara, 2018-12-20:
Hi @TianHao-Espressif 
sorry, but dont remember what board ive been using in this test. It could be wroom rev 0 devKitC or wrover mini with psram. For sure i have turned on psram in menuconfig. Sorry.

TimXia, 2019-02-01:
@chegewara Could you please provide the sdkconfig file to us to check?

chegewara, 2019-02-01:
Hi, im not sure i will be able to find settings and/or project that causes this and recreate it. If i do i will provide sdkconfig.

Alvin1Zhang, 2019-03-15:
Would you help provide any updates? Thanks @chegewara

chegewara, 2019-03-15:
I think it could be my fault. I will close this topic for now.

closed 2019-03-15 by chegewara