espressif/esp-idf #2767: Crash upon restart when calling time() and using SNTP

X-Ryl669, 2018-11-30:
To follow #2756, here is the pattern I've isolated that's crashing the ESP32 upon restart:

1. First boot, initialize Wifi as station, wait for connection and IP assigned in the main event loop, and check RTC time. Since it's (close to) Epoch, start LWIP's SNTP client.
2. Wait for SNTP client to set the time, and send an event to the main event loop. 
3. In the main event loop, show the received time (to ensure it's working) and call `esp_restart`.

1. Second boot, initialize WIFI as station, wait for connection and IP assigned in the main event loop, and check what is the RTC time. Since it's in current year, send a event to the event loop and start LWIP SNTP client.
4. In the main event loop, call `time`, `gmtime_r` to show the time to user, then call `time` again => enter `double exception vector`. It's reproducible 100% of the time.

The backtrace is: 
```
Guru Meditation Error: Core  0 panic'ed (Double exception)
Core 0 register dump:
PC      : 0x400803c0  PS      : 0x00060436  A0      : 0x00000016  A1      : 0x3ffb2150
0x400803c0: _DoubleExceptionVector at /Users/cyril/esp/esp-idf/components/freertos/xtensa_vectors.S:512

A2      : 0x3ffb21c0  A3      : 0x00000000  A4      : 0x3ffc21e8  A5      : 0x3ffc21e0
A6      : 0x00002aaa  A7      : 0x3ffb21f0  A8      : 0x80001853  A9      : 0x00000000
A10     : 0x3ffb23cc  A11     : 0x3ffb2210  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x00000aaa  A15     : 0x3ffb21d0  SAR     : 0x0000001a  EXCCAUSE: 0x00000002
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x400803c0:0x3ffb2150 0x00000013:0x3ffb2210 0x400803bd:0x3ffb20f0 0x4008e871:0x00000000
0x400803c0: _DoubleExceptionVector at /Users/cyril/esp/esp-idf/components/freertos/xtensa_vectors.S:512

0x400803bd: _UserExceptionVector at ??:?

0x4008e871: pvTaskIncrementMutexHeldCount at /Users/cyril/esp/esp-idf/components/freertos/tasks.c:4601

```
For information, if not sending the event before SNTP is started but after SNTP has "polled" the time (thus probably from a different task), this works 100% of the time.

Also, `pvTaskIncrementMutexHeldCount` is only used in `xQueueXXXReceived` but this does not make much sense since the crash happens exactly at the second call to `time(X)` (confirmed by testing with:
```
    switch(eventID)
    {
    [...]
     case TimeSet:
     {
          time_t n;
          time(&n);
          struct timeinfo ti;
          gmtime_r(n, &ti);
          Log("Current time: %d/%d/%d %02d:%02d:%02d", ti.tm_year + 100, ti.tm_mon + 1, ti.tm_mday, ti.tm_hour, ti.tm_min, ti.tm_sec);
           abort(); // First test, use this, it fails here everytime, then comment this line
           time(&n);
           abort(); // Second test, this is never reached, double exception occurs in the time above
     }
     break;
    }
```

costaud, 2018-12-20:
Is this one related to https://github.com/espressif/esp-idf/issues/2756 ?
Does this still happen after https://github.com/espressif/esp-idf/issues/2756 has been fixed ?

X-Ryl669, 2018-12-20:
It's related because I observed a crash while restarting but while fixing the former (timer still triggering after restart), this did not fix this one. I *guess* there's some leaking state that's not cleared upon restart correctly, and this breaks the time management routines. 
Because I was not able to get more information correctly (I don't have a *working* JTAG setup) and the gdb's given backtrace is useless, I decided to improve gdbstub to return more information (like the other tasks callstack & variables), and this is subject to PR #2828.

Even with this PR, I've still not figured this out... yet, because I did not have time to investigate further. I'll resume this as soon as possible.

koobest, 2018-12-20:
Hi, @X-Ryl669 
In issue https://github.com/espressif/esp-idf/issues/2756 , you are using the hardware timer group,and here, you are using gtimer, right?

X-Ryl669, 2018-12-20:
I don't know what gtimer is. Here, when calling the C library's `time` function to get the number of second elapsed from Epoch, it crashes (after restart). It does this only if I start SNTP client.
I don't know what timer the ESP32 uses underneath to hold/maintain time, I suspect it's one of the FRC since I selected this option in menuconfig, but with no guarantee, someone more involved in the code could confirm.

koobest, 2018-12-20:
OK, thanks, we  will continue to debug this issue.

KonstantinKondrashov, 2019-07-19:
Hi @X-Ryl669!
I am trying to solve this issue. I am very late, do you have a solution for this? If not could you give an example of how to reproduce it? It will be useful.
You did not fill the error template. Did you see the error on the ESP-IDF : v3.2-dev-1602-g4d021d7 form your repo?
The sdkconfig file will be useful as well.

In the comments above, you linked this issue to the SNTP operation and time() function. I'm trying to reproduce it, and I am thinking, how can I catch it, what components are associated with it. What do you think  if `initialization of Wifi as station` will be excluded from your code, will the problem still occur? 
Also I have notice that you have the modified `settimeofday` function. You add into it a callback function to get notify of update time.
`	if (time_updated_cb) time_updated_cb(time_updated_cb_arg);`
What is the code inside this callback function?

Thanks.

X-Ryl669, 2019-07-19:
Hi, yes it's a old issue now.
I seems to happen because the SNTP server calls settimeofday while the other code is in the time function holding a mutex (as far as I remember). In the same time, I had the issue (that's now fixed) that running timer survived the reboot and calling my functions while object were not initialized yet.
So it might be related to a stack or memory corruption (sadly, at the time, I haven't improved the gdbstub yet and the backtrace was useless).

Right now, like you said I've set up a callback in settimeofday to trigger my timing code when the time is actually modified and it works correctly. You've added a similar mechanism meanwhile, so I'll probably drop my changes. 

At the time, the timing to get this issue was hard to reproduce (it has to do with rebooting and having the code calling time() at the right moment to get the crash) and it's likely it's fixed now with the change in the timing code.

For now, I think we can close the issue as I have no code to reproduce and won't spend time on bissecting for this.

Alvin1Zhang, 2019-08-12:
@X-Ryl669 Thanks for reporting the issue, feel free to reopen the ticket if you still have the issue, thanks.

closed 2019-08-12 by Alvin1Zhang