espressif/esp-idf #2768: How do I find out ESP32 transceiver state?

MegaGarynych, 2018-11-30:
Hello guys,

We are going to use external RF power amplifier. We need to know which state ESP32  RF transceiver  has currently (transmit or receive) so we could change RF amplifier mode. Does esp-idf has some functions to find out ESP32  transceiver state?

Thank you.

negativekelvin, 2018-11-30:
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/wifi.html#wi-fi-multiple-antennas

But I don't think it works for bluetooth yet?

Alvin1Zhang, 2018-12-06:
@MegaGarynych 

Hi, you could try the following out:

void gpio_out_tx_status()

{

    wifi_ant_gpio_config_t config_gpio;

    config_gpio.gpio_cfg[0].gpio_select = 1;

    config_gpio.gpio_cfg[0].gpio_num = 21;

    config_gpio.gpio_cfg[1].gpio_select = 1;

    config_gpio.gpio_cfg[1].gpio_num = 22;

    wifi_ant_config_t config_ant = {

        .rx_ant_mode = WIFI_ANT_MODE_ANT1,

        .rx_ant_default = WIFI_ANT_MODE_ANT1,

        .tx_ant_mode = WIFI_ANT_MODE_ANT0,

        .enabled_ant0 = 1,

        .enabled_ant1 = 2

    };

    esp_wifi_set_ant_gpio(&config_gpio);

    esp_wifi_set_ant(&config_ant);

}

The gpio_out_tx_status() function listed above needs to be called after initialise_wifi() . It can achieve:

  ESP32 transceiver state when GPIO21 is in high level 

  ESP32 receiver state when GPIO22 is in high level.

You could refer to here for details: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/wifi.html#wi-fi-multiple-antennas

Thanks.

Alvin
Best Regards

Alvin1Zhang, 2018-12-20:
@MegaGarynych Hi, could you help share if any updates for this issue? Feel free to reopen if there is still issue. Thanks.

closed 2018-12-20 by Alvin1Zhang

MegaGarynych, 2018-12-20:
@Alvin1Zhang sorry for delay.
You code works perfect!!! Big thanks!