espressif/esp-idf #2769: Windows: using existing MSYS2, strange issues (localization an others) of windows_install_prerequisites.sh

dsptech, 2018-11-30:
Not an hardware/driver issues.
Environment: Windows 8.1 - MSYS2 32bit installed and fully updated, included gcc i686 and gccx86_64, and gtkmm (i686 and x64) packages only. MSYS2 is automatically localized to the italian language. 
I was trying to add the esp32 environment when "windows_install_prerequisites" is launched, the following error is generated:
(I produced the original output. Translate "attenzione" as "warning",
 and "è aggiornato e sarà ignorato" as "is up to date -- skipping".
It is not possibile to produce the same english output. See below.)
"
ndy@AndyHPNB2 MINGW32 /c/myprj/gitother/esp-idf/tools/windows
$ ./windows_install_prerequisites.sh
:: Sincronizzazione dei database in corso...
 mingw32 è aggiornato
 mingw64 è aggiornato
 msys è aggiornato
:: Starting core system upgrade...
 non ci sono aggiornamenti disponibili
:: Aggiornamento del sistema in corso...
 non ci sono aggiornamenti disponibili
attenzione: gettext-devel-0.19.8.1-1 è aggiornato e sarà ignorato
attenzione: gcc-7.3.0-3 è aggiornato e sarà ignorato
attenzione: git-2.19.1-1 è aggiornato e sarà ignorato
attenzione: make-4.2.1-1 è aggiornato e sarà ignorato
attenzione: ncurses-devel-6.1.20180908-1 è aggiornato e sarà ignorato
attenzione: flex-2.6.4-1 è aggiornato e sarà ignorato
attenzione: bison-3.2.2-1 è aggiornato e sarà ignorato
attenzione: gperf-3.1-1 è aggiornato e sarà ignorato
attenzione: vim-8.1.0500-1 è aggiornato e sarà ignorato
attenzione: mingw-w64-i686-python2-pip-18.1-1 è aggiornato e sarà ignorato
attenzione: mingw-w64-i686-python2-cryptography-2.4.2-1 è aggiornato e sarà ignorato
attenzione: unzip-6.0-2 è aggiornato e sarà ignorato
attenzione: winpty-0.4.3-1 è aggiornato e sarà ignorato
 non ci sono aggiornamenti disponibili
Traceback (most recent call last):
  File "C:/msys2/mingw32/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "C:/msys2/mingw32/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:/msys2/mingw32/lib/python2.7/site-packages/pip/__main__.py", line 16, in <module>
    from pip._internal import main as _main  # isort:skip # noqa
  File "C:/msys2/mingw32/lib/python2.7/site-packages/pip/_internal/__init__.py", line 40, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "C:/msys2/mingw32/lib/python2.7/site-packages/pip/_internal/cli/autocompletion.py", line 8, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "C:/msys2/mingw32/lib/python2.7/site-packages/pip/_internal/cli/main_parser.py", line 12, in <module>
    from pip._internal.commands import (
  File "C:/msys2/mingw32/lib/python2.7/site-packages/pip/_internal/commands/__init__.py", line 6, in <module>
    from pip._internal.commands.completion import CompletionCommand
  File "C:/msys2/mingw32/lib/python2.7/site-packages/pip/_internal/commands/completion.py", line 6, in <module>
    from pip._internal.cli.base_command import Command
  File "C:/msys2/mingw32/lib/python2.7/site-packages/pip/_internal/cli/base_command.py", line 25, in <module>
    from pip._internal.req.constructors import (
  File "C:/msys2/mingw32/lib/python2.7/site-packages/pip/_internal/req/__init__.py", line 5, in <module>
    from .req_install import InstallRequirement
  File "C:/msys2/mingw32/lib/python2.7/site-packages/pip/_internal/req/req_install.py", line 16, in <module>
    from pip._vendor.pep517.wrappers import Pep517HookCaller
ImportError: No module named pep517.wrappers
"

-------------------------------------------------------------------------------
By forcing an english output (to prepare this post) I seen that there are no errors related to  pep517.wrappers, but a new one is produced instead  (related to requirements.txt)  : 
"
$ LANG=en ./windows_install_prerequisites.sh
:: Synchronizing package databases...
 mingw32 is up to date
 mingw64 is up to date
 msys is up to date
:: Starting core system upgrade...
 there is nothing to do
:: Starting full system upgrade...
 there is nothing to do
warning: gettext-devel-0.19.8.1-1 is up to date -- skipping
warning: gcc-7.3.0-3 is up to date -- skipping
warning: git-2.19.2-1 is up to date -- skipping
warning: make-4.2.1-1 is up to date -- skipping
warning: ncurses-devel-6.1.20180908-1 is up to date -- skipping
warning: flex-2.6.4-1 is up to date -- skipping
warning: bison-3.2.2-1 is up to date -- skipping
warning: gperf-3.1-1 is up to date -- skipping
warning: vim-8.1.0500-1 is up to date -- skipping
warning: mingw-w64-i686-python2-pip-18.1-2 is up to date -- skipping
warning: mingw-w64-i686-python2-cryptography-2.4.2-1 is up to date -- skipping
warning: unzip-6.0-2 is up to date -- skipping
warning: winpty-0.4.3-1 is up to date -- skipping
 there is nothing to do
Could not open requirements file: [Errno 2] No such file or directory: 'C:/msys2/requirements.txt'

"
-------------------------------------------------------------
It seems to me that " if [ -n $IDF_PATH ] ; then ...", inside "windows_install_prerequisites.sh", produce a true condition even if IDF_PATH is still not defined. 
I'm not a bash expert, but commenting out the full "if" command, the installation continue successfully.

Regards

chegewara, 2018-11-30:
try this:
https://esp32.com/viewtopic.php?f=13&t=7419&p=32601#p32601

dsptech, 2018-12-02:
UPDATE:
-About Language related issue:
In the last three days I've done several installations, each time on a  newly installed  msys2 environment (from https://www.msys2.org/). I got errors up to yesterday, but now it seems that the language related issue was fixed by itself. From this morning I'm not longer able to replicate it (a package update ?)

-About requirements.txt issue:
The "requirements.txt" problem is still on, due the check:

'... if [ -n $IDF_PATH ]; then ....' (in windows_install_prerequisite.sh)

that happen to fail in its purpose. So, requirements.txt is processed even if IDF_PATH is not set.  Adding a double quote inside che check

'... if [ -n "$IDF_PATH"]; then ....' 

fix the issue in my installation.
Regards.

projectgus, 2018-12-04:
Thanks for the update, @dsptech .

Fix for the `[ -n $IDF_PATH ]` test is coming shortly. Glad to hear the other issue resolved itself.

dsptech, 2018-12-04:
I'm glad too to hear about the forthcoming fix. 
Thank you.
Regards.

closed 2018-12-20 by igrr