espressif/esp-idf #2771: with latest idf, following example for CMake fails on idf.py menuconfig

buckybots, 2018-12-01:
I did a fresh build on a PC, using MINGW64 (bash from GIT install).  Followed example to try CMake on example hello_world.  Menuconfig fails on example with output below.  I was able to idf.py build, and that had no problem.  This is all with a fresh install, following directions in Latest.  I show failed menuconfig first, and the working build next.

I just downloaded the espidf tools using exe from Latest, and then espidf, log here 

$ git log
commit 1cffc90e58c84a7a4daa801c5864110210c4d767 (HEAD -> master, origin/master, origin/HEAD)
Merge: 7458c1c1e e621e0af8
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Fri Nov 30 21:42:57 2018 +0800

    Merge branch 'feature/optimize_wifi_log' into 'master'

    wifi: add code for setting wifi log level and module

    See merge request idf/esp-idf!3710

---------------------------------------------------------------------------------------------------------

 MINGW64 ~/esp/hello_world
$ C:/Python27/python ~/esp/esp-idf/tools/idf.py menuconfig
Python requirements from C:/Users/eric/esp/esp-idf/requirements.txt are satisfied.
[0/1] cmd.exe /C "cd /D C:\Users\eric\esp\hello_world\build\esp-idf && C:\Python27\python.exe C:/Users/eric/esp/esp-idf/tools/kconfig_new/confgen.py --kconfig C:/Users/eric/esp/esp-idf/Kconfig --config C:/Users/eric/esp/hello_world/sdkconfig --create-config-if-missing --env "COMPONENT_KCONFIGS= C:/Users/eric/esp/esp-idf/components/log/Kconfig C:/Users/eric/esp/esp-idf/components/heap/Kconfig C:/Users/eric/esp/esp-idf/components/app_trace/Kconfig C:/Users/eric/esp/esp-idf/components/freertos/Kconfig C:/Users/eric/esp/esp-idf/components/vfs/Kconfig C:/Users/eric/esp/esp-idf/components/driver/Kconfig C:/Users/eric/esp/esp-idf/components/esp_event/Kconfig C:/Users/eric/esp/esp-idf/components/ethernet/Kconfig C:/Users/eric/esp/esp-idf/components/lwip/Kconfig C:/Users/eric/esp/esp-idf/components/tcpip_adapter/Kconfig C:/Users/eric/esp/esp-idf/components/spi_flash/Kconfig C:/Users/eric/esp/esp-idf/components/mbedtls/Kconfig C:/Users/eric/esp/esp-idf/components/nvs_flash/Kconfig C:/Users/eric/esp/esp-idf/components/pthread/Kconfig C:/Users/eric/esp/esp-idf/components/esp32/Kconfig C:/Users/eric/esp/esp-idf/components/aws_iot/Kconfig C:/Users/eric/esp/esp-idf/components/bt/Kconfig C:/Users/eric/esp/esp-idf/components/esp_adc_cal/Kconfig C:/Users/eric/esp/esp-idf/components/esp_http_client/Kconfig C:/Users/eric/esp/esp-idf/components/esp_http_server/Kconfig C:/Users/eric/esp/esp-idf/components/openssl/Kconfig C:/Users/eric/esp/esp-idf/components/wear_levelling/Kconfig C:/Users/eric/esp/esp-idf/components/fatfs/Kconfig C:/Users/eric/esp/esp-idf/components/freemodbus/Kconfig C:/Users/eric/esp/esp-idf/components/libsodium/Kconfig C:/Users/eric/esp/esp-idf/components/mdns/Kconfig C:/Users/eric/esp/esp-idf/components/mqtt/Kconfig C:/Users/eric/esp/esp-idf/components/spiffs/Kconfig C:/Users/eric/esp/esp-idf/components/unity/Kconfig" --env "COMPONENT_KCONFIGS_PROJBUILD= C:/Users/eric/esp/esp-idf/components/partition_table/Kconfig.projbuild C:/Users/eric/esp/esp-idf/components/bootloader/Kconfig.projbuild C:/Users/eric/esp/esp-idf/components/esptool_py/Kconfig.projbuild" --env IDF_CMAKE=y --env IDF_TARGET=esp32 --output config C:/Users/eric/esp/hello_world/sdkconfig && "C:\Program Files\CMake\bin\cmake.exe" -E env "COMPONENT_KCONFIGS= C:/Users/eric/esp/esp-idf/components/log/Kconfig C:/Users/eric/esp/esp-idf/components/heap/Kconfig C:/Users/eric/esp/esp-idf/components/app_trace/Kconfig C:/Users/eric/esp/esp-idf/components/freertos/Kconfig C:/Users/eric/esp/esp-idf/components/vfs/Kconfig C:/Users/eric/esp/esp-idf/components/driver/Kconfig C:/Users/eric/esp/esp-idf/components/esp_event/Kconfig C:/Users/eric/esp/esp-idf/components/ethernet/Kconfig C:/Users/eric/esp/esp-idf/components/lwip/Kconfig C:/Users/eric/esp/esp-idf/components/tcpip_adapter/Kconfig C:/Users/eric/esp/esp-idf/components/spi_flash/Kconfig C:/Users/eric/esp/esp-idf/components/mbedtls/Kconfig C:/Users/eric/esp/esp-idf/components/nvs_flash/Kconfig C:/Users/eric/esp/esp-idf/components/pthread/Kconfig C:/Users/eric/esp/esp-idf/components/esp32/Kconfig C:/Users/eric/esp/esp-idf/components/aws_iot/Kconfig C:/Users/eric/esp/esp-idf/components/bt/Kconfig C:/Users/eric/esp/esp-idf/components/esp_adc_cal/Kconfig C:/Users/eric/esp/esp-idf/components/esp_http_client/Kconfig C:/Users/eric/esp/esp-idf/components/esp_http_server/Kconfig C:/Users/eric/esp/esp-idf/components/openssl/Kconfig C:/Users/eric/esp/esp-idf/components/wear_levelling/Kconfig C:/Users/eric/esp/esp-idf/components/fatfs/Kconfig C:/Users/eric/esp/esp-idf/components/freemodbus/Kconfig C:/Users/eric/esp/esp-idf/components/libsodium/Kconfig C:/Users/eric/esp/esp-idf/components/mdns/Kconfig C:/Users/eric/esp/esp-idf/components/mqtt/Kconfig C:/Users/eric/esp/esp-idf/components/spiffs/Kconfig C:/Users/eric/esp/esp-idf/components/unity/Kconfig" "COMPONENT_KCONFIGS_PROJBUILD= C:/Users/eric/esp/esp-idf/components/partition_table/Kconfig.projbuild C:/Users/eric/esp/esp-idf/components/bootloader/Kconfig.projbuild C:/Users/eric/esp/esp-idf/components/esptool_py/Kconfig.projbuild" IDF_CMAKE=y KCONFIG_CONFIG=C:/Users/eric/esp/hello_world/sdkconfig IDF_TARGET=esp32 "C:/Users/eric/esp/hello_world/build/C:/Program Files/Git/usr/bin/winpty.exe" "C:/Program Files/Espressif/ESP-IDF Tools/mconf-idf/mconf-idf.exe" C:/Users/eric/esp/esp-idf/Kconfig"
The system cannot find the file specified
FAILED: esp-idf/CMakeFiles/menuconfig
cmd.exe /C "cd /D C:\Users\eric\esp\hello_world\build\esp-idf && C:\Python27\python.exe C:/Users/eric/esp/esp-idf/tools/kconfig_new/confgen.py --kconfig C:/Users/eric/esp/esp-idf/Kconfig --config C:/Users/eric/esp/hello_world/sdkconfig --create-config-if-missing --env "COMPONENT_KCONFIGS= C:/Users/eric/esp/esp-idf/components/log/Kconfig C:/Users/eric/esp/esp-idf/components/heap/Kconfig C:/Users/eric/esp/esp-idf/components/app_trace/Kconfig C:/Users/eric/esp/esp-idf/components/freertos/Kconfig C:/Users/eric/esp/esp-idf/components/vfs/Kconfig C:/Users/eric/esp/esp-idf/components/driver/Kconfig C:/Users/eric/esp/esp-idf/components/esp_event/Kconfig C:/Users/eric/esp/esp-idf/components/ethernet/Kconfig C:/Users/eric/esp/esp-idf/components/lwip/Kconfig C:/Users/eric/esp/esp-idf/components/tcpip_adapter/Kconfig C:/Users/eric/esp/esp-idf/components/spi_flash/Kconfig C:/Users/eric/esp/esp-idf/components/mbedtls/Kconfig C:/Users/eric/esp/esp-idf/components/nvs_flash/Kconfig C:/Users/eric/esp/esp-idf/components/pthread/Kconfig C:/Users/eric/esp/esp-idf/components/esp32/Kconfig C:/Users/eric/esp/esp-idf/components/aws_iot/Kconfig C:/Users/eric/esp/esp-idf/components/bt/Kconfig C:/Users/eric/esp/esp-idf/components/esp_adc_cal/Kconfig C:/Users/eric/esp/esp-idf/components/esp_http_client/Kconfig C:/Users/eric/esp/esp-idf/components/esp_http_server/Kconfig C:/Users/eric/esp/esp-idf/components/openssl/Kconfig C:/Users/eric/esp/esp-idf/components/wear_levelling/Kconfig C:/Users/eric/esp/esp-idf/components/fatfs/Kconfig C:/Users/eric/esp/esp-idf/components/freemodbus/Kconfig C:/Users/eric/esp/esp-idf/components/libsodium/Kconfig C:/Users/eric/esp/esp-idf/components/mdns/Kconfig C:/Users/eric/esp/esp-idf/components/mqtt/Kconfig C:/Users/eric/esp/esp-idf/components/spiffs/Kconfig C:/Users/eric/esp/esp-idf/components/unity/Kconfig" --env "COMPONENT_KCONFIGS_PROJBUILD= C:/Users/eric/esp/esp-idf/components/partition_table/Kconfig.projbuild C:/Users/eric/esp/esp-idf/components/bootloader/Kconfig.projbuild C:/Users/eric/esp/esp-idf/components/esptool_py/Kconfig.projbuild" --env IDF_CMAKE=y --env IDF_TARGET=esp32 --output config C:/Users/eric/esp/hello_world/sdkconfig && "C:\Program Files\CMake\bin\cmake.exe" -E env "COMPONENT_KCONFIGS= C:/Users/eric/esp/esp-idf/components/log/Kconfig C:/Users/eric/esp/esp-idf/components/heap/Kconfig C:/Users/eric/esp/esp-idf/components/app_trace/Kconfig C:/Users/eric/esp/esp-idf/components/freertos/Kconfig C:/Users/eric/esp/esp-idf/components/vfs/Kconfig C:/Users/eric/esp/esp-idf/components/driver/Kconfig C:/Users/eric/esp/esp-idf/components/esp_event/Kconfig C:/Users/eric/esp/esp-idf/components/ethernet/Kconfig C:/Users/eric/esp/esp-idf/components/lwip/Kconfig C:/Users/eric/esp/esp-idf/components/tcpip_adapter/Kconfig C:/Users/eric/esp/esp-idf/components/spi_flash/Kconfig C:/Users/eric/esp/esp-idf/components/mbedtls/Kconfig C:/Users/eric/esp/esp-idf/components/nvs_flash/Kconfig C:/Users/eric/esp/esp-idf/components/pthread/Kconfig C:/Users/eric/esp/esp-idf/c [... cut ...]
esptool.py v2.6-beta1
Checking Python dependencies...
Running ninja in directory C:\Users\eric\esp\hello_world\build
Executing "ninja all"...

Project build complete. To flash, run this command:
..\esp-idf\components\esptool_py\esptool\esptool.py -p (PORT) -b 460800 --after hard_reset write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x10000 build\hello-world.bin
or run 'idf.py -p (PORT) flash'

igrr, 2018-12-02:
Is this the same issue as https://github.com/espressif/esp-idf/issues/2765?

buckybots, 2018-12-02:
It looks like the same result, but I am not sure that it is the same cause.

 In this case, I was redoing the setup, the only thing I reused on the PC
is that it already had GIT and its MINGW64 already installed.  I then
reinstalled esp tools per latest instructions, using
esp-idf-tools-setup-1.2.exe, ran this exe on the PC, and that loaded
ESP-IDF Tools into C:\ProtramFiles\Espressif, I also allowed the installer
to download and run CMake 3.11.14 installer and Python 2.7.14 installer (64
bit).  (Python was already on this machine, and the only glitch I had was
that I had a Python error message that told me to update packages, I ran it
with a PIP install against esp-idf/requirements.txt, and the packages
updated fine fixing the Python problem).

With the esp-idf tools loaded, I then, per the Latest instructions,
downloaded esp-idf into ~/esp.  I then added IDF_PATH and PATH to a new
.profile, restarted MINGW64 window, and checked that the env was updated,
and it was.  Lastly, I copied the example program, hello_world, into
~/esp/hello_world, and tried to run idf.py menuconfig.  This is where it
broke.  The problem is that it says it cannot find the file specified, but
from the error, I can not tell what file it is looking for.  So, I think
this is a similar problem to #2765, but not sure if it is the same cause.

----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Here is the run again, the error message is shorter this time (less
compilations are needed), but the same.

MINGW64 ~/esp/hello_world
$ idf.py menuconfig
Python requirements from C:/Users/eric/esp/esp-idf/requirements.txt are
satisfied.
[0/1] cmd.exe /C "cd /D C:\Users\eric\esp\hello_world\build\esp-idf &&
C:\Python27\python.exe
C:/Users/eric/esp/esp-idf/tools/kconfig_new/confgen.py --kconfig
C:/Users/eric/esp/esp-idf/Kconfig --config
C:/Users/eric/esp/hello_world/sdkconfig --create-config-if-missing --env
"COMPONENT_KCONFIGS= C:/Users/eric/esp/esp-idf/components/log/Kconfig
C:/Users/eric/esp/esp-idf/components/heap/Kconfig
C:/Users/eric/esp/esp-idf/components/app_trace/Kconfig
C:/Users/eric/esp/esp-idf/components/freertos/Kconfig
C:/Users/eric/esp/esp-idf/components/vfs/Kconfig
C:/Users/eric/esp/esp-idf/components/driver/Kconfig
C:/Users/eric/esp/esp-idf/components/esp_event/Kconfig
C:/Users/eric/esp/esp-idf/components/ethernet/Kconfig
C:/Users/eric/esp/esp-idf/components/lwip/Kconfig
C:/Users/eric/esp/esp-idf/components/tcpip_adapter/Kconfig
C:/Users/eric/esp/esp-idf/components/spi_flash/Kconfig
C:/Users/eric/esp/esp-idf/components/mbedtls/Kconfig
C:/Users/eric/esp/esp-idf/components/nvs_flash/Kconfig
C:/Users/eric/esp/esp-idf/components/pthread/Kconfig
C:/Users/eric/esp/esp-idf/components/esp32/Kconfig
C:/Users/eric/esp/esp-idf/components/aws_iot/Kconfig
C:/Users/eric/esp/esp-idf/components/bt/Kconfig
C:/Users/eric/esp/esp-idf/components/esp_adc_cal/Kconfig
C:/Users/eric/esp/esp-idf/components/esp_http_client/Kconfig
C:/Users/eric/esp/esp-idf/components/esp_http_server/Kconfig
C:/Users/eric/esp/esp-idf/components/openssl/Kconfig
C:/Users/eric/esp/esp-idf/components/wear_levelling/Kconfig
C:/Users/eric/esp/esp-idf/components/fatfs/Kconfig
C:/Users/eric/esp/esp-idf/components/freemodbus/Kconfig
C:/Users/eric/esp/esp-idf/components/libsodium/Kconfig
C:/Users/eric/esp/esp-idf/components/mdns/Kconfig
C:/Users/eric/esp/esp-idf/components/mqtt/Kconfig
C:/Users/eric/esp/esp-idf/components/spiffs/Kconfig
C:/Users/eric/esp/esp-idf/components/unity/Kconfig" --env
"COMPONENT_KCONFIGS_PROJBUILD=
C:/Users/eric/esp/esp-idf/components/partition_table/Kconfig.projbuild
C:/Users/eric/esp/esp-idf/components/bootloader/Kconfig.projbuild
C:/Users/eric/esp/esp-idf/components/esptool_py/Kconfig.projbuild" --env
IDF_CMAKE=y --env IDF_TARGET=esp32 --output config
[... 99 lines cut ...]
C:/Users/eric/esp/esp-idf/components/esp_adc_cal/Kconfig
C:/Users/eric/esp/esp-idf/components/esp_http_client/Kconfig
C:/Users/eric/esp/esp-idf/components/esp_http_server/Kconfig
C:/Users/eric/esp/esp-idf/components/openssl/Kconfig
C:/Users/eric/esp/esp-idf/components/wear_levelling/Kconfig
C:/Users/eric/esp/esp-idf/components/fatfs/Kconfig
C:/Users/eric/esp/esp-idf/components/freemodbus/Kconfig
C:/Users/eric/esp/esp-idf/components/libsodium/Kconfig
C:/Users/eric/esp/esp-idf/components/mdns/Kconfig
C:/Users/eric/esp/esp-idf/components/mqtt/Kconfig
C:/Users/eric/esp/esp-idf/components/spiffs/Kconfig
C:/Users/eric/esp/esp-idf/components/unity/Kconfig"
"COMPONENT_KCONFIGS_PROJBUILD=
C:/Users/eric/esp/esp-idf/components/partition_table/Kconfig.projbuild
C:/Users/eric/esp/esp-idf/components/bootloader/Kconfig.projbuild
C:/Users/eric/esp/esp-idf/components/esptool_py/Kconfig.projbuild"
IDF_CMAKE=y KCONFIG_CONFIG=C:/Users/eric/esp/hello_world/sdkconfig
IDF_TARGET=esp32 "C:/Users/eric/esp/hello_world/build/C:/Program
Files/Git/usr/bin/winpty.exe" "C:/Program Files/Espressif/ESP-IDF
Tools/mconf-idf/mconf-idf.exe" C:/Users/eric/esp/esp-idf/Kconfig"
ninja: build stopped: subcommand failed.
Checking Python dependencies...
Running ninja in directory C:\Users\eric\esp\hello_world\build
Executing "ninja menuconfig"...
ninja failed with exit code 1

projectgus, 2018-12-03:
Hi @buckybots ,

I think this is different to #2765.

From the log:
`"C:/Users/eric/esp/hello_world/build/C:/Program Files/Git/usr/bin/winpty.exe" "C:/Program Files/Espressif/ESP-IDF Tools/mconf-idf/mconf-idf.exe" C:/Users/eric/esp/esp-idf/Kconfig"
The system cannot find the file specified`

It looks like the build system has detected you need winpty.exe (as it's a MSYS/MINGW terminal) but derived the path wrong.

Oddly, this is the built-in `find_program` functionality in CMake which does this part. It's worked for me in the past on MSYS2, although I didn't try under MINGW64 specifically:
https://github.com/espressif/esp-idf/blob/master/tools/cmake/kconfig.cmake#L119

Can you please try `grep -i winpty build/CMakeCache.txt` and post any result? CMake should cache the result of `find_program(winpty)` there.

(You might also be able to fix the build by editing that cache entry, although clearly that's not a permanent fix.)

buckybots, 2018-12-03:
Here it is 
MINGW64 ~/esp/hello_world
$ grep -i winpty build/CMakeCache.txt
WINPTY:FILEPATH=C:/Program Files/Git/usr/bin/winpty.exe

closed 2018-12-05 by buckybots