espressif/esp-idf #2772: CMake and IDE integration issues

huming2207, 2018-12-01:
## Environment

- Development Kit:      TTGO T8
- Kit version (for WroverKit/PicoKit/DevKitC): v1.1
- Module or chip used:  ESP32-WROVER-I
- IDF version: v3.3-dev-97-gfc1995437
- Build System: CMake
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux, Manjaro 18.0 (at home) and Ubuntu 18.10 (at work)
- Power Supply: USB

## Problem Description

Looks like ESP-IDF's CMake build system cannot integrate with some IDEs. I have tested CLion, KDevelop and VSCode with CMake Tools plugin, only CLion works mostly fine with no significant issues.

KDevelop has some false alarms on semantic analysis like this:

[image]

In the example above, **I have properly included the `esp-log.h` while KDevelop doesn't think so.**

VSCode with CMake Tools plugin also has some false alarms. But I guess it might also have some other issues with the plugin itself. So I have created another issue on that plugin project. Please refer to here for more details: https://github.com/vector-of-bool/vscode-cmake-tools/issues/604

### Expected Behavior

All the source files are properly indexed with no strange warnings and/or false alarms.

### Actual Behavior

A lot of strange warnings and/or false alarms pop up like I described above...

### Steps to repropduce

1. Create a project (make a copy from the `$IDF_PATH/examples/get-started/hello_world`)
2. Open it by VSCode + CMake Tools plugin or KDevelop, follow what it asks me to do, leave it by default and continue.
3. Some warnings pop up

### Code to reproduce this issue

You can use any projects to reproduce, or try one of my personal project: https://github.com/huming2207/QueSera 

## Debug Logs

Not available so far. If you need me to provide any, please point out.

## Other items if possible

N/A

Milan991, 2018-12-01:
@huming2207 I'm using Microsoft Visual Code, which is completely free. On the link below you can find instruction for setting it up for build and terminal monitoring:
[https://www.youtube.com/watch?v=VPgEc8FUiqI](url)

huming2207, 2018-12-01:
@Milan991 Thanks for your reply. But this is not a perfect approach as you need to configure the include path manually.

PerMalmberg, 2018-12-08:
I've been using CLion (under Linux). It has worked OK-ish; some include directories etc have been missing causing code completion etc  to break. I've always compiled using CLI to compile and link.

I just tried to get things working with VS Code (also under Linux), but hit the same problem; the tests performed on the compilers by CMake fails, which prevent any further processing. This issue was discussed on the forum a while back too, but nothing ever came of that thread.

If the tool chain could be modified so that it passes the CMake tests I think we'd have full IDE integration. @projectgus , you have been involved a lot in the CMake implementation, do you think there is a chance to make this happen? Id' be such a good thing for the community to get IDE integration working. 

This is the output from the CMake tests, showing what xtensa-gcc has issues with:
```[cms-client] Configuring using the "Unix Makefiles" CMake generator
[cmake] The C compiler identification is GNU 5.2.0
[cmake] The CXX compiler identification is GNU 5.2.0
[cmake] Check for working C compiler: /home/permal/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
[cmake] Check for working C compiler: /home/permal/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- broken
[cmake] CMake Error at /usr/local/share/cmake-3.12/Modules/CMakeTestCCompiler.cmake:52 (message):
[cmake]   The C compiler
[cmake] 
[cmake]     "/home/permal/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc"
[cmake] 
[cmake]   is not able to compile a simple test program.
[cmake] 
[cmake]   It fails with the following output:
[cmake] 
[cmake]     Change Dir: /home/permal/electronics/IO-Card-G3/software/build/CMakeFiles/CMakeTmp
[cmake]     
[cmake]     Run Build Command:"/usr/bin/make" "cmTC_7afe6/fast"
[cmake]     /usr/bin/make -f CMakeFiles/cmTC_7afe6.dir/build.make CMakeFiles/cmTC_7afe6.dir/build
[cmake]     make[1]: Entering directory '/home/permal/electronics/IO-Card-G3/software/build/CMakeFiles/CMakeTmp'
[cmake]     Building C object CMakeFiles/cmTC_7afe6.dir/testCCompiler.c.o
[cmake]     /home/permal/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc    -o CMakeFiles/cmTC_7afe6.dir/testCCompiler.c.o   -c /home/permal/electronics/IO-Card-G3/software/build/CMakeFiles/CMakeTmp/testCCompiler.c
[cmake]     Linking C executable cmTC_7afe6
[cmake]     /usr/local/bin/cmake -E cmake_link_script CMakeFiles/cmTC_7afe6.dir/link.txt --verbose=1
[cmake]     /home/permal/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc      -rdynamic CMakeFiles/cmTC_7afe6.dir/testCCompiler.c.o  -o cmTC_7afe6 
[cmake]     /home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
[cmake]     /home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o: No such file or directory
[cmake]     /home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lsim
[cmake]     /home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhandlers-sim
[cmake]     /home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhal
[cmake]     collect2: error: ld returned 1 exit status
[cmake]     CMakeFiles/cmTC_7afe6.dir/build.make:86: recipe for target 'cmTC_7afe6' failed
[cmake]     make[1]: *** [cmTC_7afe6] Error 1
[cmake]     make[1]: Leaving directory '/home/permal/electronics/IO-Card-G3/software/build/CMakeFiles/CMakeTmp'
[cmake]     Makefile:121: recipe for target 'cmTC_7afe6/fast' failed
[cmake]     make: *** [cmTC_7afe6/fast] Error 2
[cmake]     
[cmake] 
[cmake]   
[cmake] 
[cmake]   CMake will not be able to correctly generate this project.
[cmake] 
[cmake] 
[cmake] Configuring incomplete, errors occurred!
[cmake] See also "/home/permal/electronics/IO-Card-G3/software/build/CMakeFiles/CMakeOutput.log".
[cmake] See also "/home/permal/electronics/IO-Card-G3/software/build/CMakeFiles/CMakeError.log".
[cms-driver] Error during CMake configure: [cmake-server] Configuration failed.
```

huming2207, 2018-12-08:
Hello @PerMalmberg, thanks for your report.

The VSCode CMT plugin maintainer suggests me that I should also add all header files `COMPONENT_SRCS`. I've tried it and it works. Occasionally there is still a warning for missing `sdkconfig.h` but at least it's usable now. Please see here for what he replied: https://github.com/vector-of-bool/vscode-cmake-tools/issues/604#issuecomment-443488809

What I'm confused now is that, since we have added all the header's directories to `COMPONENT_ADD_INCLUDEDIRS`, why we still need to add those headers again into `COMPONENT_SRCS` to make it works? Although adding headers to `COMPONENT_SRCS` is a workaround, according to what I have understood from the sample projects provided by ESP-IDF, I don't think we need to do that. There might be something wrong either I misunderstood the ESP-IDF documentation, or there is a bug in ESP-IDF CMake build system.

So... @projectgus do you have any ideas about this? Thanks in advance!

Jackson

PerMalmberg, 2018-12-08:
@huming2207 I'm using IDF as a library so I'm not even using all the COMPONENT_*-stuff, just[ three lines](https://github.com/PerMalmberg/Smooth/blob/9a97f204e7ee8c01960a92767d879e983f288378/lib/smooth/CMakeLists.txt#L185). That said, perhaps adding additional include directories helps, but having to do that points to something missing in the tool chain; the compiler needs something extra from IDF.

I belive that is what is [being stated in this post](https://www.esp32.com/viewtopic.php?f=13&t=7535&start=20#p32958) by @igrr. In that post he argues for there not being enough use-cases to warrant the increased hassle of updating both toolchain and IDF from a user-perspective. Personally, I'll take gladly take the extra hassle when updating just to be able to integrate with IDEs; productivity-wise integration is worth a thousand times more than the cost of doing a git pull and replacing some unzipped files.

huming2207, 2018-12-10:
Just got the reply from @vector-of-bool from here: https://github.com/vector-of-bool/vscode-cmake-tools/issues/604 . 

So...is it could be the case that there is a bug in ESP-IDF's CMake build system? @projectgus is there anything I can do to debug for this?

igrr, 2018-12-10:
Hi @huming2207, sorry for the late reply.
IDF build system tells CMake about source files, include directories, and compile flags, which need to be passed to the compiler. CMake uses this information to generate the build script (Ninja or CMake). It also uses the same information to generate compile_commands.json. My hypothesis is that if the information supplied to CMake is sufficient to build the project using Ninja or Make, then it should also be sufficient for the IDE to find all source and header files. I might be wrong here — in this case please let me know which information is missing.
If you are seeing a parsing issue in IDE for some specific file, you can locate the compilation flags for this file inside compile_commands.json file, and see if any paths which should be included are missing.

@PerMalmberg Noted your issue with vscode-cmake-tools due to toolchain being broken, as well.  No ETA yet, but will try to prioritise this over other compiler related work. In the meantime you can add a custom launch target which would run the build via `idf.py`.

huming2207, 2018-12-10:
Hi @igrr, great thanks for your reply! 

According to @vector-of-bool said, his VSCode CMake plugin uses [CMake server](https://cmake.org/cmake/help/v3.7/manual/cmake-server.7.html). It fetches the information where CMake "understood" from what it has parsed.

Here is the stuff previously under my `build` directory. That's what it looks like when I haven't added any header files in my CMakeList: [build.tar.gz](https://github.com/espressif/esp-idf/files/2661962/build.tar.gz)

It looks like there is no any headers included into the `compile_commands.json`. So I assume that the CMake server might not provides any header information either.

If there is something I can help to debug, please tell me soon. Thanks.

Regards,
Jackson

PerMalmberg, 2018-12-10:
@igrr That would be very appreciated, it'd solve integration with CLion too. Let me know if there is something I can help with.

igrr, 2018-12-10:
@huming2207 From compile_commands.json, here is the line corresponding to CalLoader.cpp:
`  "command": "/home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -DESP_PLATFORM -DGCC_NOT_5_2_0=1 -DHAVE_CONFIG_H -DIDF_VER=\\\"v3.3-dev-97-gfc1995437\\\" -DMBEDTLS_CONFIG_FILE=\\\"mbedtls/esp_config.h\\\" -DWITH_POSIX -I/home/hu/Projects/QueSera/build/config -I/home/hu/Projects/QueSera/main -I/home/hu/Projects/QueSera/main/ical -I/home/hu/Projects/QueSera/main/hal -I/home/hu/Projects/QueSera/main/storage -I/home/hu/esp/esp-idf/components/app_trace/include -I/home/hu/esp/esp-idf/components/app_update/include -I/home/hu/esp/esp-idf/components/spi_flash/include -I/home/hu/esp/esp-idf/components/asio/asio/asio/include -I/home/hu/esp/esp-idf/components/asio/port/include -I/home/hu/esp/esp-idf/components/lwip/include/apps -I/home/hu/esp/esp-idf/components/lwip/lwip/src/include -I/home/hu/esp/esp-idf/components/lwip/port/esp32/include -I/home/hu/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/hu/esp/esp-idf/components/lwip/include_compat -I/home/hu/esp/esp-idf/components/vfs/include -I/home/hu/esp/esp-idf/components/bootloader_support/include -I/home/hu/esp/esp-idf/components/coap/port/include -I/home/hu/esp/esp-idf/components/coap/port/include/coap -I/home/hu/esp/esp-idf/components/coap/libcoap/include -I/home/hu/esp/esp-idf/components/coap/libcoap/include/coap -I/home/hu/esp/esp-idf/components/console -I/home/hu/esp/esp-idf/components/driver/include -I/home/hu/esp/esp-idf/components/esp_ringbuf/include -I/home/hu/esp/esp-idf/components/esp-tls -I/home/hu/esp/esp-idf/components/mbedtls/port/include -I/home/hu/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/hu/esp/esp-idf/components/esp32/include -I/home/hu/esp/esp-idf/components/tcpip_adapter/include -I/home/hu/esp/esp-idf/components/esp_event/include -I/home/hu/esp/esp-idf/components/log/include -I/home/hu/esp/esp-idf/components/esp_adc_cal/include -I/home/hu/esp/esp-idf/components/esp_http_client/include -I/home/hu/esp/esp-idf/components/nghttp/port/include -I/home/hu/esp/esp-idf/components/nghttp/nghttp2/lib/includes -I/home/hu/esp/esp-idf/components/esp_http_server/include -I/home/hu/esp/esp-idf/components/esp_https_ota/include -I/home/hu/esp/esp-idf/components/ethernet/include -I/home/hu/esp/esp-idf/components/expat/expat/expat/lib -I/home/hu/esp/esp-idf/components/expat/port/include -I/home/hu/esp/esp-idf/components/fatfs/src -I/home/hu/esp/esp-idf/components/wear_levelling/include -I/home/hu/esp/esp-idf/components/sdmmc/include -I/home/hu/esp/esp-idf/components/freemodbus/modbus/include -I/home/hu/esp/esp-idf/components/freemodbus/modbus_controller -I/home/hu/esp/esp-idf/components/freertos/include -I/home/hu/esp/esp-idf/components/heap/include -I/home/hu/esp/esp-idf/components/idf_test/include -I/home/hu/esp/esp-idf/components/jsmn/include -I/home/hu/esp/esp-idf/components/json/cJSON -I/home/hu/esp/esp-idf/components/libsodium/libsodium/src/libsodium/include -I/home/hu/esp/esp-idf/components/libsodium/port_include -I/home/hu/esp/esp-idf/components/mdns/include -I/home/hu/esp/esp-idf/components/micro-ecc/micro-ecc -I/home/hu/esp/esp-idf/components/mqtt/esp-mqtt/include -I/home/hu/esp/esp-idf/components/tcp_transport/include -I/home/hu/esp/esp-idf/components/newlib/platform_include -I/home/hu/esp/esp-idf/components/newlib/include -I/home/hu/esp/esp-idf/components/nvs_flash/include -I/home/hu/esp/esp-idf/components/openssl/include -I/home/hu/esp/esp-idf/components/protobuf-c/protobuf-c -I/home/hu/esp/esp-idf/components/protocomm/include/common -I/home/hu/esp/esp-idf/components/protocomm/include/security -I/home/hu/esp/esp-idf/components/protocomm/include/transports -I/home/hu/esp/esp-idf/components/pthread/include -I/home/hu/esp/esp-idf/components/smartconfig_ack/include -I/home/hu/esp/esp-idf/components/soc/esp32/include -I/home/hu/esp/esp-idf/components/soc/include -I/home/hu/esp/esp-idf/components/spiffs/include -I/home/hu/esp/esp-idf/components/ulp/include -I/home/hu/esp/esp-idf/components/wifi_provisioning/include -I/home/hu/esp/esp-idf/components/wpa_supplicant/include -I/home/hu/esp/esp-idf/components/wpa_supplicant/port/include -I/home/hu/esp/esp-idf/components/xtensa-debug-module/include    -Og -std=gnu++11 -fno-rtti -fno-exceptions -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -o CMakeFiles/main.dir/ical/CalLoader.cpp.obj -c /home/hu/Projects/QueSera/main/ical/CalLoader.cpp"`

It contains `-I/home/hu/esp/esp-idf/components/log/include` which is the location where `esp_log.h` file can be found, and `ESP_LOGE` definition should come from that file. I think this information should be sufficient for the IDE to resolve this macro?

huming2207, 2018-12-10:
@igrr  Ahhh ok sorry I missed that part. Yes that probably should be enough. Then probably some IDE/plugins didn't handle that correctly. I will ask them.

dtboy1995, 2018-12-25:
Are there any articles developed using clion? Thank you

huming2207, 2018-12-25:
@dtboy1995 Using CLion is easy. You just need to copy a boilerplate from ESP-IDF examples and use CLion to open it.

huming2207, 2018-12-25:
Hi @igrr , I would like to know are you still planning to work on your vscode plugin (i.e. https://github.com/espressif/vscode-esp-idf-extension) ?

igrr, 2018-12-26:
@huming2207 Yes, we are working on it.

jaanvi, 2019-01-04:
Hi, all. We are using CLion also for most of our embedded development. I think CLion is, in my opinion, the best option if you are willing to pay for it. 
Firstly, @PerMalmberg  the problem you have/had is related to esp32 weird need to have things defined in the global environment. Personally, I do not like this approach at all. But this can be easily solved in the CMakeLists.txt where you can add needed environmental variables (both IDF_PATH & PATH). 
Secondly, I have to admit there are still a bunch of integration problems. Such as moving between usages and declaration. But at the same time, I have to admit that we are running quite an experimental setup. C++, CLion, cmake & clion remote build.

piotrkundu, 2019-08-16:
@jaanvi I'm trying to build esp-idf from source with CLion, without any luck. CMake generates targets for each lib, but I can't build any of these. Also I would expect a target that builds the entire image for ESP32.
[image]
Trying to build libbt.a fails like this:
```
====================[ Build | ___idf_bt | Debug ]===============================
/Applications/CLion.app/Contents/bin/cmake/mac/bin/cmake --build /Users/piotrk/_git/esp-idf/cmake-build-debug --target ___idf_bt -- -j 4
make[3]: EXCLUDE_FROM_ALL: No such file or directory
make[3]: *** [CMakeFiles/___idf_bt] Error 1
make[2]: *** [CMakeFiles/___idf_bt.dir/all] Error 2
make[1]: *** [CMakeFiles/___idf_bt.dir/rule] Error 2
make: *** [___idf_bt] Error 2
```

renzbagaporo, 2019-08-21:
@piotrkundu  To build a specific library, use the target with only 2 underscores preceeding the name, i.e. `__idf_bt` vs `___idf_bt`. The three underscore one is an internal phony target, we'll see if we can clean them up.

With regards to building an image, I think the `app` target will suit this need?

piotrkundu, 2019-08-23:
@renzbagaporo That's a great tip. Didn't notice that there where both double and tripple underscore targets. The double ones do work as you point out.

securez, 2019-12-09:
I think that the main problem to this issue is simply that idf.py -v build, doesn't call ninja with -v option, so no build command in output is show, mostly IDEs use this output to discover include paths, macros, etc..

projectgus, 2019-12-09:
@securez CMake build system generates a `build/compile_commands.json` file instead. This is a [standard way to document the build commands](https://clang.llvm.org/docs/JSONCompilationDatabase.html) which is much more reliable than trying to parse the build runner output.

We also assume most IDEs will want to interact with CMake & the build generator (in this case ninja) directly rather than via the idf.py frontend.

Do you have an example of an IDE that only supports build output parsing and can't work with compile_commands.json?

closed 2020-03-11 by renzbagaporo