espressif/esp-idf #2773: lwIP: TCP handshake is not completed

yaqwsx, 2018-12-01:
## Environment

- Development Kit: ESP32-DevKitC
- Build System: PlatformIO, platform 1.5.0, (ESP-IDF framework 3.30101.0 )

## Problem Description

When opening TCP connection using raw API between two ESP32, TCP handshake does not complete. The client sends SYNC; the server sends SYNC+ACK; the client never responds. The problem is probably in condition `tcp_in.c:734` in `tcp_process`:

```cpp
if ((flags & TCP_ACK) && (flags & TCP_SYN)
     && ackno == ntohl(pcb->unacked->tcphdr->seqno) + 1 ) {
```
Upstream version of lwIP (from Savannah) has the condition changed to:
```cpp
if ((flags & TCP_ACK) && (flags & TCP_SYN)
     && (ackno == pcb->lastack + 1)) {
```
### Context

Bug discovered during implementation of custom network interface for lwIP. The bug was discovered using [1]. It was tested only with my netif, it was not tested on WiFi. However, the components should be separate, therefore, I think the bug should be reproducible even on WiFi. [2] and [3] are logs of the example code run with LwIP debug messages. As can been seen from the log:
```
SYN-SENT: ackno 6510 pcb->snd_nxt 6510 unacked 1
```
The values of `ackno` and `pcb->snd_nxt` feature of-by-one error.

If a workaround, i.e, commenting out the ackno check, is applied, the example works as expected.

[1] Example code: https://pastebin.com/RRSXyDbL
[2] LwIP debug log from server: https://pastebin.com/TRGa8Ng6
[3] LWIP debug log from client: https://pastebin.com/gi5LR36S

yaqwsx, 2018-12-27:
Could you please tell me what is the versions of lwIP in ESP-IDF? I see there is a fork of lwIP at  [1]. The fork has this issue fixed, however, ESP-IDF uses a different version and I was not able to update ESP-IDF to [1] (several compilation errors).

[1] https://github.com/espressif/esp-lwip

zhangyanjiaoesp, 2018-12-27:
@yaqwsx The versions of lwIP in ESP-IDF is lwip_STABLE_2_0_3, but it is not only the copy of the lwip_2.0.3, we add our own code for it.

zhangyanjiaoesp, 2019-03-15:
@yaqwsx The  esp-lwip fork is used in ESP-IDF v3.2 and the later versions, you can use `release/v3.2` .

Alvin1Zhang, 2019-03-15:
@yaqwsx Hi. Would you help share if any updates for this issue? Thanks.

Alvin1Zhang, 2019-03-19:
@yaqwsx Feel free to reopen if the issue still happens. Thanks.

closed 2019-03-19 by Alvin1Zhang