espressif/esp-idf #2774: Classical BT SPP doesn't ask PIN code when pairing

copercini, 2018-12-01:
## Environment

- Development Kit:      ESP32-DevKitC
- IDF version:              v3.2-dev-1800-g1cffc90e5 (last git version)
- Build System:            Make
- Compiler version:     crosstool-NG crosstool-ng-1.22.0-80-g6c4433a
- Operating System:    Linux
- Power Supply:           USB

## Problem Description
When using legacy paring on BT SPP, it's expected to ESP32 ask a PIN code before pair with another device, but even setting a code on `esp_bt_gap_set_pin()`, no PIN is requested for pairing

### Expected Behavior
ESP32 asks for a PIN when the smartphone is pairing, if this PIN is right, then pair

### Actual Behavior
ESP32 is pairing without any proof of possession, allowing anyone to pair and send commands

### Relevant Logs

```cpp
I (933) SPP_ACCEPTOR_DEMO: ESP_SPP_INIT_EVT
I (953) SPP_ACCEPTOR_DEMO: ESP_SPP_START_EVT
I (167523) SPP_ACCEPTOR_DEMO: authentication success: Android phone
I (167523) SPP_ACCEPTOR_DEMO: 3f 02 11 55 5a
W (169573) BT_RFCOMM: rfc_find_lcid_mcb LCID reused LCID:0x41 current:0x0
W (169573) BT_RFCOMM: RFCOMM_DisconnectInd LCID:0x41
W (179723) BT_APPL: new conn_srvc id:26, app_id:255
I (179723) SPP_ACCEPTOR_DEMO: ESP_SPP_SRV_OPEN_EVT
I (195373) SPP_ACCEPTOR_DEMO: speed(179.504532s ~ 195.156723s): 0.003067 kbit/s
```

### Code to reproduce this issue
Basically it's the spp_acceptor example with some changes in the end: [Full code is here](https://pastebin.com/LtxLWxXH)

blueMoodBHD, 2018-12-03:
Hi, @copercini , When ESP32 pair with smartphone, they will use Secure Simple pairing instead of Legacy Pairing. And fixed pin is used for Legacy Pairing,  so, fixed pin will not used.
Then, you set `iocap` as `ESP_BT_IO_CAP_OUT`,  and the default  `iocap` of smartphone is `ESP_BT_IO_CAP_IO`,  so they will pair with  `justwork`. 
You can set   `iocap` as `ESP_BT_IO_CAP_IO`, they will compare the numeric value; or  set   `iocap` as `ESP_BT_IO_CAP_IN`, it will show a passkey in smartphone then input it in ESP32

chegewara, 2018-12-03:
Hi @blueMoodBHD 
what we would like to achieve is bt SPP with legacy pairing. We know now it cant be done, now question, is it an option to implement it as feature in some feature?

blueMoodBHD, 2018-12-03:
Hi, @chegewara, We now support both Secure Simple pairing(SSP) and  Legacy Pairing. If both devices support SSP, they will use SSP for encryption. 
    According to your demand， We will consider providing a configuration to turn off SSP.

chegewara, 2018-12-03:
@blueMoodBHD That would be great, thanks.

TylerHallPureEngineering, 2018-12-04:
yes! cool. ok i've been looking around on this exact same issue. Glad to see others have the exact same result as i have found through experimentation.

https://www.esp32.com/viewtopic.php?f=2&t=8266&p=34916#p34916

blueMoodBHD, 2019-01-25:
Hi , all. Sorry for later reply,  we can turn on/off SSP in menuconfig, now. Refer to commit 65142bc59edb9fbf97b98b7aa867c68c32e15cb1

TylerHallPureEngineering, 2019-02-12:
Just an update. Works for what we need. I can get PIN code requests now as expected, when disabling SSP from make menuconfig. Thank you for the fix!

pacifickashyap, 2019-02-27:
@TylerHallPureEngineering  Can you share the example code of how to get it work?

pacifickashyap, 2019-03-02:
@blueMoodBHD @TylerHallPureEngineering  SSP is already compiled as enabled in arduino core.
So how to change it in runtime?

blueMoodBHD, 2019-03-04:
@pacifickashyap We can't change it in runtime.

closed 2019-03-15 by Alvin1Zhang