espressif/esp-idf #2775: UART0 not works after setting UART1 as console output

keelung-yang, 2018-12-02:
* ESP32-PICO-Kit
* Lastest esp-idf for windows
------
I Changed UART port from 1 to 0 and set all pins to UART_PIN_NO_CHANGE in uart_echo example. It works.
But after setting UART1(Tx:18, Rx23) as console output by menuconfig.
It doesn't work: no any response received after sending data to UART0 & UART1.

I also tried UART1(Tx:4, Rx:5) as in uart_echo example, it doesn't work too.

keelung-yang, 2018-12-04:
@koobest Please check following snapshots, sdkconfig and uart_echo_example_main.c:

[image]

[image]

[image]

[uart_echo_example_main.zip](https://github.com/espressif/esp-idf/files/2642551/uart_echo_example_main.zip)
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2642553/sdkconfig.zip)

koobest, 2018-12-04:
@keelung-yang 
When calling `uart_set_pin`, the pin number of UART0 should be specified to put the GPIO in the correct mode. e.g.
```
uart_set_pin(UART_PORT, 1, 3, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
```
Otherwise you will not be able to accept or send data.  
When UART0 is set to console output, we have already configured  the gpio to the correct mode.

keelung-yang, 2018-12-04:
@koobest It's works now! Thank you very very much!

closed 2018-12-04 by keelung-yang