espressif/esp-idf #2776: Missing distclean target

keelung-yang, 2018-12-02:
I want to reset all configurations in menuconfig, so I execute this:
> $ make distclean
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
make: *** No rule to make target 'distclean'.  Stop.

There is no distclean target in project_config.mk. 
I also tried `make config-clean`, still not working: my changes still in menuconfig

igrr, 2018-12-02:
Hi @keelung-yang, could you please point to the documentation page which mentioned this target? There is only `clean` target for build output. You can cancel config changes by delete sdkconfig file.

keelung-yang, 2018-12-02:
> Hi @keelung-yang, could you please point to the documentation page which mentioned this target? There is only `clean` target for build output. You can cancel config changes by delete sdkconfig file.

Hi @igrr There is no distclean target indeed, according search results of official documents. 

But it's a standard target for "_delete all files in the current directory (or created by this makefile) that are created by configuring or building the program_", see [https://www.gnu.org/prep/standards/html_node/Standard-Targets.html](https://www.gnu.org/prep/standards/html_node/Standard-Targets.html). And most of projects using gnu toolchain also support distclean target, such as u-boot, linux kernel, ffmpeg and so on.

Literally, the config-clean target is doing what distclean doing. So, why not just rename config-clean to distclean as the 'standart' target in gnu toolchain?

BTW, the config-clean did not clean my menuconfig changes in the lastest esp-idf.

igrr, 2018-12-02:
sdkconfig is of a project is often checked into git, so it is not so certain that it should be considered a build output. There are some build system targets which help you manipulate this settings file, but we treat it differently from the files generated in `build` directory. If you find `distclean` target useful, you may add it to your project makefile, as `rm -f sdkconfig`. However for some projects, sdkconfig file is included into the "distribution", so removing it would not be correct.

Alvin1Zhang, 2018-12-20:
@keelung-yang Feel free to reopen if the issue still exists, thanks.

keelung-yang, 2018-12-21:
@Alvin1Zhang I cannot re-open this issues since it's not closed by me and I'm not a collaborator.

There are several questions still not clear, so I don't think it should be closed.

- Literally, the config-clean target is doing what distclean doing, but it's not documented
- Need a clear and "standard" way to reset configurations and remove temp files, as same as distclean doing
- If there is a "standard" way to do something, then using "standard" way in SDK but delivery work to end users: custom makefile themself.

Alvin1Zhang, 2018-12-21:
Issue reopened. @keelung-yang

Alvin1Zhang, 2019-03-22:
@keelung-yang Thanks for reporting this.