espressif/esp-idf #2777: Can't store values after nvs_flash_erase()

BrianPugh, 2018-12-02:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     macOS]
- Power Supply:         USB

## Problem Description

After running `nvs_flash_erase()`, I am unable to store key/value pairs (namely blobs) until I restart. Weirdly, all writing functions return "successful". Removing `nvs_flash_erase()` allows for proper rights. In my application i'm trying to perform a "factory reset" and then write in a special values.

igrr, 2018-12-02:
Could you please share the code snippet which illustrates the issue? Particularly interesting is the sequence in which sequence you call nvs_flash_erase, nvs_flash_init, nvs_open, nvs_write, nvs_commit.

igrr, 2018-12-03:
Oh, I see now, thanks for the example. Documentation does not mention that `nvs_flash_erase` can only be used when NVS is not initialized, i.e. before `nvs_flash_init` or after `nvs_flash_deinit`. Will update the docs and add corresponding error check.

alyf80, 2020-02-07:
Just a friendly reminder: there is still no mention of this in either the documentation or the API reference.

Also, in the current IDF 4.0 (haven't checked with 4.1) calling `nvs_flash_erase()` on an initialized partition still returns `ESP_OK` while managing to break things in non-obvious ways.

In my specific case, after calling `nvs_flash_erase()` followed by `nvs_flash_init()` I would be able to write `u8`'s _and read them back successfully_ , with corruption only becoming apparent after the first reboot.

closed 2020-02-23 by espressif-bot