espressif/esp-idf #2779: ESP BT SPP: Interrupt wdt timeout upon disabling then enabling gap_scan_mode

Xasin, 2018-12-02:
## Environment

- Development Kit:none
- Module or chip used:  ESP32-PICO-D4
- IDF version: `v3.3-dev-337-g1cffc90e5`
- Build System:  CMake
- Compiler version: `1.22.0-80-g6c4433a`
- Operating System: Linux
- Power Supply:  USB

## Problem Description

In order to save power, I want to be able to disable, then re-enable the BT SPP module. Doing so, however, repeatably causes a crash and restart of the system, effectively making it impossible to restart the Bluetooth system.
More specifically, the crash occurs shortly after a second call to `esp_bt_gap_set_scan_mode` (in the order of a few ms); without setting any scan mode the system works. This is also undesirable, as without a scan mode enabled, nothing can connect to the device.

### Expected Behavior

The calls to `esp_bt_gap_set_scan_mode` and `esp_bt_controller_enable/disable` should not crash. They should simply restart the Bluetooth system, and thusly allow my system to save power by disabling Bluetooth when not needed.

### Actual Behavior

The second call to `esp_bt_gap_set_scan_mode`, which re-enables the discoverable_connectable mode, seems to cause an Interrupt to hang up. A few ms later, the interrupt watchdog is triggered, causing a Guru Meditation Error on Core 0 and resetting the chip.

### Steps to reproduce

Simply set up the build environment to have BT SPP mode enabled, and run the code below.

### Code to reproduce this issue

```cpp
  // Minimal initialisation above (i.e. #include's, nvs, tcp_ip and event loop init)

  esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
  bt_cfg.mode = ESP_BT_MODE_CLASSIC_BT;
  esp_bt_controller_init(&bt_cfg);
  esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT);
  esp_bluedroid_init();
  esp_bluedroid_enable();

  while (true) {
		puts("SPP: Disabling");
		esp_bt_controller_disable();

		vTaskDelay(3000 / portTICK_PERIOD_MS);

		puts("SPP: Enabling");
		esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT);
		while(esp_bt_controller_get_status() != ESP_BT_CONTROLLER_STATUS_ENABLED)
				vTaskDelay(10);
		esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE);

		vTaskDelay(3000/portTICK_PERIOD_MS);
	}
```

## Debug Logs

```
SPP: Disabling
SPP: Enabling
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40087d7f  PS      : 0x00060034  A0      : 0x80088cc6  A1      : 0x3ffbe6c0  
0x40087d7f: r_rwble_isr at ??:?

A2      : 0x00000004  A3      : 0x3ff71210  A4      : 0x800817b3  A5      : 0x3ffbbba0  
A6      : 0x00000000  A7      : 0x00000002  A8      : 0x00000020  A9      : 0x00000010  
A10     : 0x00000000  A11     : 0x3ffb7fec  A12     : 0x800907cd  A13     : 0x3ffbbb90  
A14     : 0x00000003  A15     : 0x00060323  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x40153fa6  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40087d7f
0x40153fa6: esp_pm_impl_waiti at /opt/ESP/esp-idf/components/esp32/pm_esp32.c:487

0x40087d7f: r_rwble_isr at ??:?

Backtrace: 0x40087d7f:0x3ffbe6c0 0x40088cc3:0x3ffbe6e0 0x40082972:0x3ffbe700 0x40153fa3:0x00000000
0x40087d7f: r_rwble_isr at ??:?

0x40088cc3: r_rwbtdm_isr_wrapper at intc.c:?

0x40082972: _xt_lowint1 at /opt/ESP/esp-idf/components/freertos/xtensa_vectors.S:1105

0x40153fa3: esp_pm_impl_waiti at /opt/ESP/esp-idf/components/esp32/pm_esp32.c:487

Core 1 register dump:
PC      : 0x40153fa6  PS      : 0x00060f34  A0      : 0x800d32c6  A1      : 0x3ffbc350  
0x40153fa6: esp_pm_impl_waiti at /opt/ESP/esp-idf/components/esp32/pm_esp32.c:487

A2      : 0x3ffcba49  A3      : 0x00060f20  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d2662  A9      : 0x3ffbc340  
A10     : 0x00000003  A11     : 0x00060f23  A12     : 0x00060f20  A13     : 0x3ffbc260  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40153fa6:0x3ffbc350 0x400d32c3:0x3ffbc370 0x40090da9:0x3ffbc390 0x4008fded:0x3ffbc3b0
0x40153fa6: esp_pm_impl_waiti at /opt/ESP/esp-idf/components/esp32/pm_esp32.c:487

0x400d32c3: esp_vApplicationIdleHook at /opt/ESP/esp-idf/components/esp32/freertos_hooks.c:86

0x40090da9: prvIdleTask at /opt/ESP/esp-idf/components/freertos/tasks.c:3537

0x4008fded: vPortTaskWrapper at /opt/ESP/esp-idf/components/freertos/port.c:403
```

blueMoodBHD, 2018-12-03:
Hi, @Xasin , Before disable controller, you should disable and deinit SPP and bluedroid. Then after enable controller, init and enable bluedroid and SPP again.

blueMoodBHD, 2018-12-04:
@Xasin ，Yes, (de)initialisation be executed that often is not a good idea. 
We are working on light sleep and DFS,  when it is done, it can save much power when there is nothing to do.

Xasin, 2018-12-04:
@blueMoodBHD Automatic light sleep compatibility would be fantastic, especially combined with BLE.
I suppose that in that case this issue should be closed since technically I wasn't running into a bug, I was just doing it wrong.

I'm looking forward to the Light Sleep functionality! Until then, I'll just have to charge my system a bit more often.

Alvin1Zhang, 2018-12-20:
@Xasin Thanks for reporting, feel free to reopen the issue if the issue reappears. Thanks.

closed 2018-12-20 by Alvin1Zhang