espressif/esp-idf #2784: Move UART RS458 mode and flow control from freemodbus port in SDK to user code

keelung-yang, 2018-12-03:
I‘m struggling to official modbus_slave example.

Please just Keep UART as UART mode in SDK for **design** principle reason.
UART is in SDK, so just manipulate UART in SDK and leave other options, like mode & flow control to user code. Because they are options for user but SDK. 

Users maybe use UART as native UART or RS232/RS485/RS422 connections. This should be determined by user but SDK.

I know the necessary hardware connections are well documented in freemodbus example. But the RS485 mode and RTS pin are not configured in this example but in SDK! That means we have to buy a RS485 device to run modbus_slave example if no checking hardcoded UART mode in SDK!

AGAIN, THEY ARE USER's OPTITONS BUT SDK!

keelung-yang, 2018-12-03:
Please take attentions for what SDK **should** or **shouldn't** do.

alisitsyn, 2018-12-12:
Hi keelung-yang,

The issue has been confirmed and corrected in ESP-IDF. 
In new revision the port pin and mode configuration will be part of example KConfig.
The changes will be merged soon.

keelung-yang, 2018-12-17:
@alisitsyn THX! Then I should close this issuse.

closed 2018-12-17 by keelung-yang