espressif/esp-idf #2786: Bluetooth GAP/A2DP affecting Pulse Counter

markingle, 2018-12-04:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it): v3.3-dev-337-g1cffc90
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     macOS
- Power Supply:         USB

## Problem Description
I am using the pulse counter functions to monitor a signal coming in on GPIO4.  When the program starts up it creates an average based on the difference between the first and second reading, which is one sample.  Ten (10) samples are taken to determine the average difference.  Here is the serial output from the first sample:

Start calibration....pass #0
Startup Calibration Pulse Counter is on - Core: 0
First counter value for calibration:171
Second counter value for calibration:183
Count Difference: 12.000000

Once the sampling completes, subsequent readings are taken to determine if there is a change in frequency.  If the difference between the first and second reading is > or < the average by 1 then I play a "beep" as an audible notification.  I am using a potentiometer to change the frequency.

The code worked perfectly until I enabled Bluetooth GAP and A2DP.  When GAP is initialized the BT_APPL layer does something to cause a delay in reading the second value thus causing the logic in the code to think the frequency has changed.....causing a beep.  This should not happen.

[image]

### Expected Behavior

Initializing BT GAP and A2DP should not impact the timing on checking the pulse counter.  A beep should only play when the frequency changes.

### Actual Behavior

The section where the problem occurs is in **BOLD.**

rst:0x1 (POWERON_RESET),boot:0x3a (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6164
load:0x40078000,len:10168
load:0x40080400,len:6552
entry 0x40080764
I (28) boot: ESP-IDF v3.3-dev-337-g1cffc90-dirty 2nd stage bootloader
I (28) boot: compile time 20:53:04
I (33) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x40038 (262200) map
I (185) esp_image: segment 1: paddr=0x00050060 vaddr=0x3ff80000 size=0x00000 (     0) load
I (186) esp_image: segment 2: paddr=0x00050068 vaddr=0x3ff80000 size=0x00000 (     0) load
I (192) esp_image: segment 3: paddr=0x00050070 vaddr=0x3ffbdb60 size=0x03d60 ( 15712) load
I (207) esp_image: segment 4: paddr=0x00053dd8 vaddr=0x3ffc18c0 size=0x00000 (     0) load
I (210) esp_image: segment 5: paddr=0x00053de0 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/Mark/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (219) esp_image: segment 6: paddr=0x000541e8 vaddr=0x40080400 size=0x0be28 ( 48680) load
I (248) esp_image: segment 7: paddr=0x00060018 vaddr=0x400d0018 size=0xa2118 (663832) map
0x400d0018: _flash_cache_start at ??:?

I (481) esp_image: segment 8: paddr=0x00102138 vaddr=0x4008c228 size=0x083fc ( 33788) load
0x4008c228: esp_rom_spiflash_write at /Users/Mark/esp/esp-idf/components/spi_flash/spi_flash_rom_patch.c:475

[... 675 lines cut ...]
The frequnecy difference is :#14
Frequency +++++++**
D (14259) ledc: LEDC_PWM CHANNEL 0|GPIO 23|Duty 0000|Time 0
Analyzing frequency data to detect metal....
Running first counter value :1059
Running second counter value :1071
The frequnecy difference is :12
Frequency -------

.
### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue
Here is a gist link to functions for detecting the change in frequency.  Let me know if more code is needed.

https://gist.github.com/markingle/0524d0c946f9d30593aef598ea61b33f

## Debug Logs

I have been able to create a SystemView DAT file.  But I cannot tell that anything is wrong.  I can post it if needed.

blueMoodBHD, 2018-12-17:
Hi @markingle ,
It seems like BT task block your task. As BT task running on Core 0, you can startup calibration pulse counter on  Core 1 and test again.
Thanks.

markingle, 2018-12-17:
Thanks @blueMoodBHD!  I have already tried Core 1 for the pulse counter but it did not correct the problem. I’ll test again and post the results along with my code changes for your review.

markingle, 2018-12-23:
@blueMoodBHD Here the function I am using to pin the detect_metal function to Core 1:

**xTaskCreatePinnedToCore(detect_metal, "DetectMetal_Task", 3000, NULL, 1, &DetectMetal_Task, 1);**

Below are calibration logs and the interference.  The interference with the pulse counter doesn't start until sound starts coming out of the speaker.

Also here is a portion of the log showing the calibration and pulse counter running on Core 1....

Determining frequency average on start up
Start calibration....pass #0
Startup Calibration Pulse Counter is on - Core: 1
First counter value for calibration:161
Second counter value for calibration:184
Count Difference: 23.000000
Start calibration....pass #1
Startup Calibration Pulse Counter is on - Core: 1
First counter value for calibration:185
Second counter value for calibration:208
Count Difference: 23.000000
Start calibration....pass #2
Startup Calibration Pulse Counter is on - Core: 1
First counter value for calibration:208
Second counter value for calibration:232
Count Difference: 24.000000
Start calibration....pass #3
Startup Calibration Pulse Counter is on - Core: 1
First counter value for calibration:232
Second counter value for calibration:256
Count Difference: 24.000000
Start calibration....pass #4
Startup Calibration Pulse Counter is on - Core: 1
First counter value for calibration:256
Second counter value for calibration:280
Count Difference: 24.000000
Start calibration....pass #5
Startup Calibration Pulse Counter is on - Core: 1
First counter value for calibration:280
Second counter value for calibration:303
Count Difference: 23.000000
Start calibration....pass #6
Startup Calibration Pulse Counter is on - Core: 1
First counter value for calibration:303
Second counter value for calibration:327
Count Difference: 24.000000
Start calibration....pass #7
Startup Calibration Pulse Counter is on - Core: 1
First counter value for calibration:327
Second counter value for calibration:351
Count Difference: 24.000000
Start calibration....pass #8
Startup Calibration Pulse Counter is on - Core: 1
First counter value for calibration:351
Second counter value for calibration:375
Count Difference: 24.000000
Start calibration....pass #9
Startup Calibration Pulse Counter is on - Core: 1
First counter value for calibration:375
Second counter value for calibration:399
Count Difference: 24.000000
SUM: 237
Average: 23

Partial log of BT interfering with the pulse counter....

Analyzing frequency data to detect metal....
Running first counter value :6371
Running second counter value :6394
The frequnecy difference is :23
Frequency -------
Analyzing frequency data to detect metal....
Running first counter value :6394
Running second counter value :6418
The frequnecy difference is :24
Frequency -------
Analyzing frequency data to detect metal....
Running first counter value :6418
Running second counter value :6430
The frequnecy difference is :12
Frequency +++++++
Analyzing frequency data to detect metal....
Running first counter value :6430
Running second counter value :6430
The frequnecy difference is :0
Frequency +++++++
Analyzing frequency data to detect metal....
Running first counter value :6430
Running second counter value :6430
The frequnecy difference is :0
Frequency +++++++
Analyzing frequency data to detect metal....
Running first counter value :6430
Running second counter value :6430
The frequnecy difference is :0
Frequency +++++++
Analyzing frequency data to detect metal....
Running first counter value :6430
Running second counter value :6430
The frequnecy difference is :0
Frequency +++++++
Analyzing frequency data to detect metal....
Running first counter value :6430
Running second counter value :6430
The frequnecy difference is :0
Frequency +++++++
Analyzing frequency data to detect metal....
Running first counter value :6430
Running second counter value :6430
The frequnecy difference is :0
Frequency +++++++
Analyzing frequency data to detect metal....
Running first counter value :6430
Running second counter value :6430
The frequnecy difference is :0
Frequency +++++++
Analyzing frequency data to detect metal....
Running first counter value :6430

markingle, 2018-12-23:
I increased the priority to 3 and it helped a little. The frequency change trigger isn’t tripped as often but it gets worse over time. Here is a video demonstrating the issue. 

https://youtu.be/pevl79casdo

markingle, 2019-01-05:
This issue was fixed by pinning the pulse_count function to Core 1 and eliminating a gpio handler function that was not needed

closed 2019-01-05 by markingle