espressif/esp-idf #2787: pthread doesn't allow setting task name

jeremyherbert, 2018-12-04:
In `pthread.c`, the name of the task is hardcoded:

`BaseType_t res = xTaskCreate(&pthread_task_func, "pthread", stack_size, task_arg, prio, &xHandle);`

It would be useful if this could be changed via either the new `pthread_attr_t` struct introduced in  https://github.com/espressif/esp-idf/commit/f27db1f24194f9a40f4a9d50f3471160c645f4fb or via the `esp_pthread_cfg_t` struct

jeremyherbert, 2018-12-04:
Actually, I think it is better done via `esp_pthread_cfg_t` so that it can be used with `std::thread`

This functionality would be useful when using `vTaskList`, otherwise the pthreads are hard to distinguish from each other:

```
I (60607) HEAP: Heap memory free: 4481020 (global min was 4481020)
I (60607) HEAP: TASK LIST:
pthread        	R	5	780	13
main           	R	1	4972	5
IDLE1          	R	0	1036	7
IDLE0          	R	0	1044	6
pthread        	B	5	2184	12
Tmr Svc        	B	1	1556	8
esp_timer      	B	22	3608	1
ipc0           	B	24	536	2
ipc1           	B	24	528	3
```

igrr, 2018-12-04:
Related: https://github.com/espressif/esp-idf/pull/2743

jeremyherbert, 2018-12-04:
great! looking forward to seeing it in master. You are welcome to close this in the meantime.

closed 2018-12-04 by igrr