espressif/esp-idf #2788: 怎么通过i2s_read_bytes()函数读取mic数据啊？

xiaoseyihe, 2018-12-04:
I2S配置如下：
i2s_config_t i2s_config = {

    .mode = I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_TX,
    .sample_rate = 48000,
    .bits_per_sample = SUPPOERTED_BITS,
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
    .communication_format = I2S_COMM_FORMAT_I2S,
    
    .dma_buf_count = 3,                            /*!< amount of the dam buffer sectors*/
    .dma_buf_len = 300,                            /*!< dam buffer size of each sector (word, i.e. 4 Bytes) */
    .intr_alloc_flags = I2S_INTER_FLAG,
    .use_apll = 1,
};

costaud, 2018-12-06:
What kind of microphone are you using ?

xiaoseyihe, 2018-12-06:
Analog Microphone

koobest, 2018-12-11:
Hi,  @xiaoseyihe 
Because you are using  analog microphone, so, it should use  I2S ADC to read mic data. you can refer to the example of  `examples/peripherals/i2s_adc_dac`

koobest, 2018-12-17:
Hi, @xiaoseyihe 
 Did you solve the issue?

Alvin1Zhang, 2018-12-20:
@xiaoseyihe Hi, could you help share if any updates for this issue? Thanks.

Alvin1Zhang, 2018-12-28:
@xiaoseyihe Hi, feel free to reopen the issue if it still exists. Thanks.

closed 2018-12-28 by Alvin1Zhang