espressif/esp-idf #2796: error during "make all" of blink project (Linux machine)

ziolelle, 2018-12-05:
## Environment

- Development Kit:      ESP32-DevKitC

- Module or chip used:  [ESP32]

- IDF version v3.3-dev-386-gf1b4c18

- Build System:         [Make]
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Linux]

## Problem Description

during compilation of "blink" project at the end got the following error:

Generating esp32.common.ld
Traceback (most recent call last):
  File "/opt/esp32/esp-idf/tools/ldgen/ldgen.py", line 111, in <module>
    main()
  File "/opt/esp32/esp-idf/tools/ldgen/ldgen.py", line 86, in main
    with open(sections_info_file) as sections_info_file_obj:
IOError: [Errno 36] File name too long: '/root/esp32/workspace/blink/build/app_trace/libapp_trace.a.sections_info\\n /root/esp32/workspace/blink/build/app_update/libapp_update.a.sections_info\\n /root/esp32/workspace/blink/build/asio/libasio.a.sections_info\\n /root/esp32/workspace/blink/build/aws_iot/libaws_iot.a.sections_info\\n /root/esp32/workspace/blink/build/bootloader_support/libbootloader_support.a.sections_info\\n /root/esp32/workspace/blink/build/bt/libbt.a.sections_info\\n /root/esp32/workspace/blink/build/coap/libcoap.a.sections_info\\n /root/esp32/workspace/blink/build/console/libconsole.a.sections_info\\n /root/esp32/workspace/blink/build/cxx/libcxx.a.sections_info\\n /root/esp32/workspace/blink/build/driver/libdriver.a.sections_info\\n /root/esp32/workspace/blink/build/esp-tls/libesp-tls.a.sections_info\\n /root/esp32/workspace/blink/build/esp32/libesp32.a.sections_info\\n /root/esp32/workspace/blink/build/esp_adc_cal/libesp_adc_cal.a.sections_info\\n /root/esp32/workspace/blink/build/esp_event/libesp_event.a.sections_info\\n /root/esp32/workspace/blink/build/esp_http_client/libesp_http_client.a.sections_info\\n /root/esp32/workspace/blink/build/esp_http_server/libesp_http_server.a.sections_info\\n /root/esp32/workspace/blink/build/esp_https_ota/libesp_https_ota.a.sections_info\\n /root/esp32/workspace/blink/build/esp_https_server/libesp_https_server.a.sections_info\\n /root/esp32/workspace/blink/build/esp_ringbuf/libesp_ringbuf.a.sections_info\\n /root/esp32/workspace/blink/build/ethernet/libethernet.a.sections_info\\n /root/esp32/workspace/blink/build/expat/libexpat.a.sections_info\\n /root/esp32/workspace/blink/build/fatfs/libfatfs.a.sections_info\\n /root/esp32/workspace/blink/build/freemodbus/libfreemodbus.a.sections_info\\n /root/esp32/workspace/blink/build/freertos/libfreertos.a.sections_info\\n /root/esp32/workspace/blink/build/heap/libheap.a.sections_info\\n /root/esp32/workspace/blink/build/idf_test/libidf_test.a.sections_info\\n /root/esp32/workspace/blink/build/jsmn/libjsmn.a.sections_info\\n /root/esp32/workspace/blink/build/json/libjson.a.sections_info\\n /root/esp32/workspace/blink/build/libsodium/liblibsodium.a.sections_info\\n /root/esp32/workspace/blink/build/log/liblog.a.sections_info\\n /root/esp32/workspace/blink/build/lwip/liblwip.a.sections_info\\n /root/esp32/workspace/blink/build/main/libmain.a.sections_info\\n /root/esp32/workspace/blink/build/mbedtls/libmbedtls.a.sections_info\\n /root/esp32/workspace/blink/build/mdns/libmdns.a.sections_info\\n /root/esp32/workspace/blink/build/micro-ecc/libmicro-ecc.a.sections_info\\n /root/esp32/workspace/blink/build/mqtt/libmqtt.a.sections_info\\n /root/esp32/workspace/blink/build/newlib/libnewlib.a.sections_info\\n /root/esp32/workspace/blink/build/nghttp/libnghttp.a.sections_info\\n /root/esp32/workspace/blink/build/nvs_flash/libnvs_flash.a.sections_info\\n /root/esp32/workspace/blink/build/openssl/libopenssl.a.sections_info\\n /root/esp32/workspace/blink/build/protobuf-c/libprotobuf-c.a.sections_info\\n /root/esp32/workspace/blink/build/protocomm/libprotocomm.a.sections_info\\n /root/esp32/workspace/blink/build/pthread/libpthread.a.sections_info\\n /root/esp32/workspace/blink/build/sdmmc/libsdmmc.a.sections_info\\n /root/esp32/workspace/blink/build/smartconfig_ack/libsmartconfig_ack.a.sections_info\\n /root/esp32/workspace/blink/build/soc/libsoc.a.sections_info\\n /root/esp32/workspace/blink/build/spi_flash/libspi_flash.a.sections_info\\n /root/esp32/workspace/blink/build/spiffs/libspiffs.a.sections_info\\n /root/esp32/workspace/blink/build/tcp_transport/libtcp_transport.a.sections_info\\n /root/esp32/workspace/blink/build/tcpip_adapter/libtcpip_adapter.a.sections_info\\n /root/esp32/workspace/blink/build/ulp/libulp.a.sections_info\\n /root/esp32/workspace/blink/build/unity/libunity.a.sections_info\\n /root/esp32/workspace/blink/build/vfs/libvfs.a.sections_info\\n /root/esp32/workspace/blink/build/wear_levelling/libwear_levelling.a.sections_info\\n /root/esp32/workspace/blink/build/wifi_provisioning/libwifi_provisioning.a.sections_info\\n /root/esp32/workspace/blink/build/wpa_supplicant/libwpa_supplicant.a.sections_info\\n /root/esp32/workspace/blink/build/xtensa-debug-module/libxtensa-debug-module.a.sections_info\\n'
make: *** [/root/esp32/workspace/blink/build/esp32/esp32.common.ld] Error 1

num-lock, 2018-12-05:
This issue happens on basically every make. Seems it has been introduced with a7a1c32a8eae57c0cdcbd2177dbb89d9e2e18026

https://github.com/espressif/esp-idf/blob/f1b4c18c2a5bc4be83c260605424880e4093b9c5/tools/ldgen/ldgen.py#L82

Seems like split doesn't properly split the lines, so all file paths are interpreted as one.

```
$ make
...
IOError: [Errno 36] File name too long: '/home/bla/a\\n /home/bla/b\\n ...' etc.
```

projectgus, 2018-12-05:
Thanks for reporting this, @num-lock . As mentioned on the linked PR, this is an issue with shell escaping when using GNU Make and bash as the system shell.

A fix will be merged very shortly.

closed 2018-12-06 by igrr