espressif/esp-idf #2799: aws-iot-device-sdk-embedde-c function declaration mismatch problem

IBeRyUS, 2018-12-06:
Function declaration is not match each other on network_mbedtls_wrapper unit. 

AWS function declaration for iot_tls_init in network_interface.h is 
```cpp
IoT_Error_t iot_tls_init(Network *pNetwork, char *pRootCALocation, char *pDeviceCertLocation,
						 char *pDevicePrivateKeyLocation, char *pDestinationURL,
						 uint16_t DestinationPort, uint32_t timeout_ms, bool ServerVerificationFlag);
```
Same function in network_mbedtls_wrapper.c is 
```cpp
IoT_Error_t iot_tls_init(Network *pNetwork, const char *pRootCALocation, const char *pDeviceCertLocation,
                         const char *pDevicePrivateKeyLocation, const char *pDestinationURL,
                         uint16_t destinationPort, uint32_t timeout_ms, bool ServerVerificationFlag) {
```

Because of this `const` declaration mismatch compiling is ending with error.

mahavirj, 2018-12-07:
@IBeRyUS Which network_interface.h file you are referring to? One in ESP-IDF https://github.com/espressif/aws-iot-device-sdk-embedded-C/blob/299183238ffe7a3e6a5ca0af9db19c10eaca62cf/include/network_interface.h#L96 looks correct to me.

IBeRyUS, 2018-12-07:
@mahavirj it's the v3.0.1 tag version.
https://github.com/espressif/aws-iot-device-sdk-embedded-C/blob/d039f075e1cc2a2a7fc20edc6868f328d8d36b2f/include/network_interface.h#L96

mahavirj, 2018-12-07:
@IBeRyUS (That fairly old release) Is it possible for you to use latest stable ESP-IDF v3.1.1 release?

mahavirj, 2018-12-09:
@IBeRyUS Aha, my bad, misunderstood your comment. We have few patches on top of official `aws-iot-device-sdk` release, e.g. https://github.com/espressif/aws-iot-device-sdk-embedded-C/commit/a70e2cc9502ce79ced76688979bc29252a16b811. Can you please help to check why your `aws-iot-device-sdk` sub-module does not include above fix (or help to send across `git submodule` output)?

chegewara, 2018-12-09:
Hi,
i am currently build application based on this example (esp-idf master):
https://github.com/espressif/esp-idf/tree/master/examples/protocols/aws_iot/subscribe_publish

It work great, except one thing. I had to use different CA certificate. Instead of `VeriSign Class 3 Public Primary G5 root CA certificate` (which is included in example)
https://www.symantec.com/content/en/us/enterprise/verisign/roots/VeriSign-Class%203-Public-Primary-Certification-Authority-G5.pem
i had to use `Amazon Root CA 1`:
https://www.amazontrust.com/repository/AmazonRootCA1.pem

> We recommend that all customers create an Amazon Trust Services (ATS) endpoint and load these CA certificates onto their devices to avoid any issues with the upcoming widespread distrust by browsers of Symantec CAs (including VeriSign) in October 2018

mahavirj, 2018-12-09:
@chegewara 
> It work great, except one thing. I had to use different CA certificate. Instead of VeriSign Class 3 Public Primary G5 root CA certificate (which is included in example)
https://www.symantec.com/content/en/us/enterprise/verisign/roots/VeriSign-Class%203-Public-Primary-Certification-Authority-G5.pem
i had to use Amazon Root CA 1:
https://www.amazontrust.com/repository/AmazonRootCA1.pem

Thanks for reporting, probably this needs to be fixed. Will keep you posted.