espressif/esp-idf #2800: Stopping monitor on Mac OS

ItsRom, 2018-12-06:
## Actions already completed

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- IDF version : v3.3-dev-396-g8c7a6be62
- Build System: Make
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System: Mac OS
- Power Supply:  USB

## Problem Description

When a do make monitor to see the hello word flashed on my ESP32, it's worked but the programe restart in infinite loop and I must do Ctrl + ] to stop it but I'm on Mac and Ctrl + ] doesn't worked.
I use Shift + Alt + ) for ] so Ctrl + Shift + Alt + ) but it's doesn't worked. 
I tried Ctrl + C or Cmd + C and lot of others tricks but doesn't work too.

I have this screen in loop.
[image]

How can I quit the monitor properly?
Thanks for your help.

Romain.

igrr, 2018-12-06:
Hi @ItsRom, which terminal application are you using, the default Terminal.app or something else?

Nathaniel7687, 2018-12-07:
> Hi @ItsRom, which terminal application are you using, the default Terminal.app or something else?

I'm suffering the same problem.
The problem has occurred in "screen" and "make monitor".
The serial communication just stuck while displaying ESP_LOGs and even booting.
So I have to re-connect the serial communication with screen command such as
"screen /dev/cu.SLAB_USBtoUART 115200" to monitor continuously without reboot the ESP32.
Unfortunately, the screen application also doesn't last very long.

Nathaniel7687, 2018-12-07:
> Hi @ItsRom, which terminal application are you using, the default Terminal.app or something else?

I'm suffering similarly a problem.
The problem has occurred in "screen" and "make monitor".
The serial communication just stuck while displaying ESP_LOGs and even booting (I don't know what about printf).

So I have to re-connect the serial communication with screen command such as
"screen /dev/cu.SLAB_USBtoUART 115200" to monitor continuously without reboot the ESP32.
Unfortunately, the screen application also doesn't last very long.

X-Ryl669, 2018-12-10:
For me it's `Ctrl + $`(the key that's just on the left of return key, top row)

lvx3, 2018-12-11:
`Ctrl+]` works for me in iterm2

ItsRom, 2018-12-12:
Thanks @X-Ryl669 : on Terminal.app Ctrl + $ works too!

closed 2018-12-12 by ItsRom

GilDev, 2020-02-15:
Was having the same issue, I type in [Bépo](http://bepo.fr).
For anyone interested, `Ctrl + T` `Ctrl + X` works for me.

georgik, 2021-06-02:
CTRL+] works for iTerm and Terminal on Big Sur. Thanks @lvturner 
@igrr  Is there a reason why CTRL+C is not accepted?

dobairoland, 2021-06-02:
@georgik I'm not sure which version are you using but CTRL+C should interrupt the application and invoke GDB Stub.

georgik, 2021-06-02:
@dobairoland Some additional info:
`idf.py --version`
`ESP-IDF v4.4-dev-1404` - master https://github.com/espressif/esp-idf

Observed behavior on OSes: 
- Linux and macOS is the same. CTRL+C is not accepted
- Windows throws Python exception `KeyboardInterrupt` and terminates

sverrisson, 2021-11-23:
Note I had the same issue as I am NOT using a USA keyboard.  
I saw the comment above pressing the key next to return at top, so I did press that key on my keyboard. For me this key is '  and it will be different on other international keyboards. 
Please pick a better key for stopping the monitor, that works on other than US keyboards.

gavalierm, 2024-01-16:
CTRL + T, then X

jonasbeltoft, 2024-03-11:
Danish keyboards: CTRL + OPTION + Å