espressif/esp-idf #2805: esp_mqtt reconnect error: TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected

rustyx, 2018-12-07:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32
- IDF version: **master** v3.3-dev-236-gfa59b1b1c
- Build System:         Make
- Compiler version: (crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     Windows 10
- Power Supply:         USB

## Problem Description

I'm trying out the new [**esp_mqtt**](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/mqtt.html) library with an **mqtts://** (TSL) connection.

My MQTT server is mosquitto v1.5.3.

The issue is whenever the **mqtt server is restarted**, the **esp_mqtt** task hangs with the following messages in the log:

```
E (56962) TRANS_SSL: esp_tls_conn_read error, errno=Bad file number
E (56967) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (56967) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (56977) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (56982) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (56992) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (56997) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57007) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57012) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57022) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57027) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57037) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57042) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57052) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57062) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57067) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57077) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57082) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57092) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57097) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57107) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57112) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57122) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57127) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57137) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57142) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57152) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57157) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57167) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57172) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57182) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57192) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57197) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57207) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57212) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57222) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57227) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57237) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57242) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57252) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57257) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57267) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57272) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57282) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57287) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57297) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57302) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57312) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57317) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57327) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57337) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57342) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57352) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57357) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57367) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57372) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57382) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57387) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57397) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57402) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57412) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57417) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57427) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57432) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57442) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57447) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57457) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57467) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57472) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57482) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57487) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57497) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57502) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57512) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57517) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57527) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57532) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57542) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57547) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57557) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (57562) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
. . .
```

Other error scenarios like Wi-Fi reconnection, are handled correctly; **esp_mqtt** is then able to reconnect properly. It is only when the MQTT server connection is closed is it not able to reconnect.

david-cermak, 2018-12-11:
Hi @rustyx 

It is a bug in ssl transport component. There is already a fix ready in Espressif internal repo, but that still needs to be verified, merged and published to GitHub.

I am attaching a oneliner patch that should address the issue (before it is available in GitHub)
[0001-tcp_transport-fixed-ssl-reconnection-issue.patch.txt](https://github.com/espressif/esp-idf/files/2667143/0001-tcp_transport-fixed-ssl-reconnection-issue.patch.txt)

Thanks, David

AadiMehta, 2018-12-23:
@david-cermak 
Thanks david for the patch.

I confirm it works for me.

syed00727, 2020-03-27:
@david-cermak  How to solve it in ESP-MDF.. 
Im facing this error when my internet router turns off, then ESP crashes.

`I (419335) wifi: n:11 0, o:11 2, ap:11 2, sta:11 2, prof:11
I (419341) [mwifi, 99]: Parent is disconnected, reason: 200
I (419347) mesh: [wifi]disconnected reason:200(beacon timeout), continuous:0/max:12, root, vote(,stopped)<><>
I (419347) [mqtt_tasks, 340]: event_loop_cb, event: 8
E (419347) esp-tls: read error :-76:
I (419359) [mqtt_tasks, 353]: Parent is disconnected on station interface
E (419365) TRANS_SSL: esp_tls_conn_read error, errno=Software caused connection abort
E (419377) MQTT_CLIENT: Read error or end of stream
I (419389) [mesh_mqtt, 175]: MQTT_EVENT_DISCONNECTED
W (419395) MQTT_CLIENT: Client asked to stop, but was not started
I (419395) [mqtt_tasks, 340]: event_loop_cb, event: 24578
I (419401) [mqtt_tasks, 475]: MQTT disconnected
I (419407) mesh: <MESH_NWK_ROOT_TODS_STATE>toDS:0
I (419413) [mwifi, 149]: State represents: 0
I (419413) [mqtt_tasks, 340]: event_loop_cb, event: 11
assertion "res == coreID || res == portMUX_FREE_VAL" failed: file "../esp-mdf/esp-idf/components/freertos/portmux_impl.inc.h", line 105, function: vPortCPUAcquireMutexIntsDisabledInternal
abort() was called at PC 0x400d2da7 on core 0
0x400d2da7: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
`

I tired to put delay in callbacks

`case MDF_EVENT_MWIFI_PARENT_DISCONNECTED:
            MDF_LOGI("Parent is disconnected on station interface");
                vTaskDelay(5000 / portTICK_PERIOD_MS);

            if (esp_mesh_is_root()) {
                mesh_mqtt_stop();
            }

            break;
`

david-cermak, 2020-04-03:
Hi @syed00727 

One reason for this kind of error could be a memory corruption.
Also, there's been a recent fix of a race condition on closing mqtt connection here https://github.com/espressif/esp-mqtt/commit/dc1a635a97bbbebd74d965447384c20d34d08526. Could you please cherry-pick this commit and check if it resolves the issue?