espressif/esp-idf #2806: ESP-IDF release ZIP file has the .git folder in it.

nuket, 2018-12-07:
Is there a reason to include the `.git` folder in the ZIP files from the [Releases](https://github.com/espressif/esp-idf/releases) page? 

This adds about 200MB worth of compressed data that no one really needs if they're using the release as a point-in-time snapshot anyways.

igrr, 2018-12-10:
Hi @nuket, we do keep .git directory in those archives on purpose. The intention is that users still use `git` to update IDF or switch between versions. Packaging the entire working copy into a release archive is a workaround for users who can't perform `git clone --recursive https://github.com/espressif/esp-idf.git` due to unreliable Internet connectivity.

nuket, 2018-12-10:
Ahh, I see. That makes sense.

WayneKeenan, 2018-12-10:
Are you already using the `--depth` parameter to reduce the `.git` sub-folder size when `git clone`ing prior to zipping?

Alvin1Zhang, 2018-12-18:
@nuket Hi, could you help share if any updates for this issue? Can this issue be closed? Thanks.

nuket, 2018-12-18:
No updates from my side, this issue can be closed.

closed 2018-12-18 by nuket