espressif/esp-idf #2808: Multi-Core handling of gpio interrupts (gpio_install_isr_service / gpio_isr_handler) - missing or double calls to handler

martinportmann, 2018-12-07:
## Environment

- Development Kit:     No (own hardware based on ESP32-PICO-D4
- Kit version: na
- Module or chip used:  ESP32-PICO-D4
- IDF version v3.1 
- Build System: Make
- Compiler version 1.22.0-80
- Operating System: Linux
- Power Supply:   DC/DC 2.8V

## Problem Description

This is about code in 'esp-idf/components/driver/gpio.c'.
I checked the code in master and but I am at the moment in the 3.1 release branch.

I am using GPIO interrupts a lot.
I am porting an application to the esp32 with existing code that uses an own HAL layer and is already based on FreeRTOS.

I do call once 'gpio_install_isr_service' and then 'gpio_isr_handler' and 'gpio_intr_enable' for every IO.
The threads that call gpio_isr_handler/gpio_intr_enable will get distributed randomly to a free core through FreeRTOS because the legacy code does not use xTaskCreatePinnedToCore.

The problem is now that I get duplicated calls to gpio_isr_handler because the code in 'gpio_intr_service' does not check if a GPIO is assigned to the core the ISR is running.

This resulted in double calls and sometimes in lost interrupts.
I see two solutions:
- only handle in gpio_intr_service GPIOs assigned to the current core
- assign all IRQs to a specific core

I choose the second solution and changed function 'gpio_intr_enable_on_core' to only assign to one CPU and had no more problems.

### Expected Behavior
No missing or double call to the gpio interrupt handler

### Actual Behavior
Missing or double calls to gpio interrupt handler if initialisation is done from both cores.

koobest, 2018-12-12:
Hi, @martinportmann 
Can you provide a example that I can reproduce your issue?
thanks!!

koobest, 2018-12-17:
Hi, @martinportmann 
Have you solved the problems?

martinportmann, 2018-12-17:
Hi koohoo

As I work on my own hardware I have no development kit around to
demonstrate the problem.
Me for me I have solved the problem in the way that I changed the esp-idf
to just issue the IRQ on one CPU.
Therefore I changed components/driver/gpio.c function
gpio_intr_enable_on_core to:

static esp_err_t gpio_intr_enable_on_core (gpio_num_t gpio_num, uint32_t
core_id)
{
    GPIO_CHECK(GPIO_IS_VALID_GPIO(gpio_num), "GPIO number error",
ESP_ERR_INVALID_ARG);
    gpio_intr_status_clr(gpio_num);
     GPIO.pin[gpio_num].int_ena = GPIO_PRO_CPU_INTR_ENA;     //enable pro
cpu intr
    return ESP_OK;
}

This means that all the GPIO IRQ will be executed on the first core. In my
case this is acceptable but the original idea was that
the GPIO interrupt will be enabled on the calling core. But the function
'gtpio_intr_service' does not check if the event is associated to the core
running the actual IRQ.
If two GPIO events happen at the same time two GPIO IRQ routines run at the
same time reading/writing on the same registers ...

Best regards
- Martin

wingunder, 2018-12-29:
@martinportmann, Could you please look at PR https://github.com/espressif/esp-idf/pull/2873 
With these patches, you should be able to:
   - call gpio_intr_disable from anywhere and expect the interrupt not to occur again.
   - rely on gpio_intr_disable to disable an interrupt consistently.
   - call gpio_intr_enable from any task, regardless on which core it runs.

martinportmann, 2018-12-29:
Hi Pieter

I had no time to test BUT:
I just call enable once at startup and therefore chance is that this is not
solving anything for me.
Please refer to the description of the problem. The problem happs within
the IRQ handling events for both both CPUs.

Best regards
- Martin

wingunder, 2018-12-30:
Hi Martin,

> I just call enable once at startup and therefore chance is that this is not solving anything for me.

I'm not sure of this. The difference is that gpio_intr_enable now enables the interrupt on the core, where gpio_install_service was called. Furthermore gpio_intr_enable_on_core (where you had to make a change to get your program to work) was removed, as gpio_intr_enable now consistently enables the interrupt on the right core.

For a demonstration of this, you could look at the unit test that I created for:
   - testing the enabling of a gpio interrupt (calling gpio_isr_handler_add, gpio_set_intr_type and/or gpio_intr_enable) on a different core to the one on which gpio_install_isr_service(0) was called.
   - testing disabling and re-enabling interrupts, using gpio_intr_disable for disabling and gpio_intr_enable and/or gpio_set_intr_type for enabling on a different core to the one on which gpio_install_isr_service(0) was called.

See:
https://github.com/wingunder/esp-idf/blob/a89e8213502007a0718fc3d5a99dd979677951f2/components/driver/test/test_gpio.c#L625-L735

> I do call once 'gpio_install_isr_service' and then 'gpio_isr_handler' and 'gpio_intr_enable' for every IO.
The threads that call gpio_isr_handler/gpio_intr_enable will get distributed randomly to a free core through FreeRTOS because the legacy code does not use xTaskCreatePinnedToCore.

Do you mean that you call gpio_isr_handler_add instead of gpio_isr_handler? (gpio_isr_handler is not in master:HEAD).
Would it be possible to supply a piece of code demonstrating the problem, similar to the test case that I pointed to above?

koobest, 2019-07-18:
Hi, Thank you for feedback on this issue. we will fix this bug soon.
@wingunder thanks for your PR, but we can use `esp_ipc_call` to fix this bug.

thanks !!

closed 2019-08-05 by igrr