espressif/esp-idf #2809: When SDIO and WiFi receive work at the same time, the packet loss rate of WiFi receive increases?

jixunx, 2018-12-08:
## Environment

- Development Kit:     
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|
- IDF version v3.1
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
     1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [external 3.3V]

## Problem Description
1. One esp32 is used as WiFi transmission and one esp32 is used as WiFi reception, using esp_80211_tx. The payload size of each package is 960 bytes.
2·Send data packets continuously. Within every 60 packets, there is no time interval between data packets and data packets.
3. The receiver has a sdio_transfer_task. For each packet received by the receiver, it enters wifi_sniffer_packet_handler and notifies sdio_transfer_task.Sdio_transfer_task writes two blocks to SDIO slave device. The receiver esp32 acts as the master device of the receiver.
4. Wifi task is allocated to CPU1 and sdio_transfer_task to cpu0.
5·If WiFi and SDIO transmissions do not work at the same time, the packet loss rate is very low.

//Detailed problem description goes here.

### Expected Behavior
Lower Packet Loss Rate

### Actual Behavior
higher Packet Loss Rate

### Steps to repropduce

1. Start the esp32 receiver.
2. Start the esp32 sender.

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue
receiver code excerpt：

```cpp
void wifi_sniffer_packet_handler(void* buff, wifi_promiscuous_pkt_type_t type)
{
 
    if (type != WIFI_PKT_DATA)
        return;
 
    const wifi_promiscuous_pkt_t *ppkt = (wifi_promiscuous_pkt_t *)buff;
    const wifi_ieee80211_packet_t *ipkt = (wifi_ieee80211_packet_t *)ppkt->payload;
    const wifi_ieee80211_mac_hdr_t *hdr = &ipkt->hdr;
 
    if(!((hdr->addr2[4]==0x00)&&(hdr->addr2[5]==0x00)))
    {
        return ;
    }
 
 
       
 
        if(NORMALSENDPACKETS==ipkt->payload[0])//it means it is not the retrans packet
        {
 
            pack_count++;
            register uint32_t i=0;
            register uint8_t pac_seq=ipkt->payload[1];
            if(pac_seq==59)
            {              
                ets_printf("cb=%d\n",60-pack_count);
                pack_count=0;
            }
            xEventGroupSetBits(frame_event_group,FRAME_RECEIVED_BIT);
        }
   
}
static void  sdio_transfer_task(void *pvParameters)
{
 
    while(true)
    {
 
        xEventGroupWaitBits(frame_event_group,FRAME_RECEIVED_BIT, pdTRUE, pdTRUE,portMAX_DELAY);
       
       
                uint8_t sdiotmp[1024]
        esp_err_t user_sdmmc_write_error=sdmmc_write_sectors(card,sdiotmp,1,2);
        if(user_sdmmc_write_error)
        {
            esp_restart();
           }
    }
}

```
sender code excerpt：
```cpp
int32_t user_esp_wifi_80211_tx(void * data,size_t datasize,uint32_t picture_count)
{
        wifi_internal_rate_t rate;  
       
        //  struct timeval test_promis_start;
        //  struct timeval test_promis_end;
        //  struct timeval promis_result_time;
        int32_t fail_count=0;
        int pack_num=datasize/PACKSIZE;
 
 
 
       
#ifdef CHANGERTATE
 
        rate.fix_rate = RATE_MCS7;
        esp_wifi_internal_set_rate(100, 1, 4, &rate);
#endif
 
        for(int i=0;i<pack_num;i++)
        {
 
            packetH[DATAOFFSET]=NORMALSENDPACKETS;
            packetH[DATAOFFSET+1]=i;
            memcpy(packetH+DATAOFFSET+2,data+i*PACKSIZE,PACKSIZE);
            int result = esp_wifi_80211_tx(WIFI_MODE_NULL, packetH, sizeof(packetH), true);
           
            if(result)
            {
                fail_count++;
 
            }
        }
       
 
        return fail_count;
       
}
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

[sdio and wifi work at the same time.zip](https://github.com/espressif/esp-idf/files/2659355/sdio.and.wifi.work.at.the.same.time.zip)

MaxwellAlan, 2020-10-22:
Hi @jixunx, does this issue still exist? If yes, could you help to reproduce this issue in idf v4.0 above, because we can't reproduce it. Thanks.

MaxwellAlan, 2021-01-19:
Hi @jixunx , thanks for reporting this issue.
In our internal test, only SDIO and WiFi work together, there is only a little performace loss.
But test with your code, we found when use the SDIO to write data to SD card, the bottleneck of performace is SD card writing, according to the SD specification, performance will decrease as the card fills up.
Last but not least, there are some ways to improve performace:

- optimize wifi config according to [import wifi performace](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#how-to-improve-wi-fi-performance)
-  format the card before starting the packet capture
- optimize SDIO configuration according to SD Card example

jack0c, 2022-05-17:
@MaxwellAlan @jixunx It should be related with Flash impact Wi-Fi performance issue.
The root cause should be same, the solution should also be same:
put some code into IRAM.
The fix is added since idfv.3.3.

closed 2022-05-17 by jack0c