espressif/esp-idf #2812: Error Kconfig and Makefile when build eclipse

kakaducsy, 2018-12-10:
[image]

I try to build project with eclipse but error as bellow:

`14:45:32 **** Incremental Build of configuration Default for project HelloWorld ****
python "C:\\msys32\\home\\sy.nguyen-duc\\esp\\esp-idf/tools/windows/eclipse_make.py" -j4 all 
Running make in 'C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld'
including C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/app_update/Makefile.projbuild...
including C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/bootloader/Makefile.projbuild...
including C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/bootloader_support/Makefile.projbuild...
including C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/coap/Makefile.projbuild...
including C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/esp32/Makefile.projbuild...
including C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/mbedtls/Makefile.projbuild...
including C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/nghttp/Makefile.projbuild...
including C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/partition_table/Makefile.projbuild...
including C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/ulp/Makefile.projbuild...
including C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/unity/Makefile.projbuild...
Toolchain path: C:/msys32/opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
PROJECT_PATH= COMPONENT_DIRS= make -C C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/bootloader/subproject V=1 BUILD_DIR_BASE=C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/bootloader TEST_COMPONENTS= TESTS_ALL= EXCLUDE_COMPONENTS= C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/bootloader/bootloader.bin
python C:/msys32/home/sy.nguyen-duc/esp/esp-idf/tools/check_python_dependencies.py
make -C C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/app_trace -f C:/msys32/home/sy.nguyen-duc/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/app_trace/component.mk COMPONENT_NAME=app_trace build
make -C C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/app_update -f C:/msys32/home/sy.nguyen-duc/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/app_update/component.mk COMPONENT_NAME=app_update build
make[1]: Entering directory 'C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/bootloader/subproject'
make[1]: Entering directory 'C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/app_update'
make[1]: Entering directory 'C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/app_trace'
make[1]: Nothing to be done for 'build'.
make[1]: Leaving directory 'C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/app_trace'
make -C C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/asio -f C:/msys32/home/sy.nguyen-duc/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/asio/component.mk COMPONENT_NAME=asio build
make[1]: Entering directory 'C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/asio'
make[1]: Nothing to be done for 'build'.
make[1]: Leaving directory 'C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/asio'
make -C C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/aws_iot -f C:/msys32/home/sy.nguyen-duc/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/aws_iot/component.mk COMPONENT_NAME=aws_iot build
make[1]: Entering directory 'C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/aws_iot'
make[1]: Nothing to be done for 'build'.
make[1]: Leaving directory 'C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/aws_iot'
make -C C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/bootloader_support -f C:/msys32/home/sy.nguyen-duc/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/bootloader_support/component.mk COMPONENT_NAME=bootloader_support build
make[1]: Entering directory 'C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/bootloader_support'
make[1]: Nothing to be done for 'build'.
make[1]: Leaving directory 'C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/bootloader_support'
App "app-template" version: df62230
make -C C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/bt -f C:/msys32/home/sy.nguyen-duc/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/bt/component.mk COMPONENT_NAME=bt build
Python requirements from C:/msys32/home/sy.nguyen-duc/esp/esp-idf\requirements.txt are satisfied.
make[1]: Entering directory 'C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/bt'
make[1]: Nothing to be done for 'build'.
make[1]: Leaving directory 'C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/bt'
make -C C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/coap -f C:/msys32/home/sy.nguyen-duc/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/coap/component.mk COMPONENT_NAME=coap build
make -C C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/console -f C:/msys32/home/sy.nguyen-duc/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/console/component.mk COMPONENT_NAME=console build
make[1]: Entering directory 'C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/console'
make[1]: Nothing to be done for 'build'.
make[1]: Leaving directory 'C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/console'
make -C C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/build/cxx -f C:/msys32/home/sy.nguyen-duc/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/cxx/component.mk COMPONENT_NAME=cxx build
xtensa-esp32-elf-gcc -std=gnu99 -Og -ggdb -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -DESP_PLATFORM -D IDF_VER=\"v3.3-dev-424-g8b885fb93\" -MMD -MP   -DGCC_NOT_5_2_0=0 -DWITH_POSIX -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -DUNITY_INCLUDE_CONFIG_H -D PROJECT_VER=\"df62230\" -D PROJECT_NAME=\"app-template\" -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/app_update/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/app_trace/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/asio/asio/asio/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/asio/port/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/bootloader_support/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/bt/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/coap/port/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/coap/port/include/coap -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/coap/libcoap/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/coap/libcoap/include/coap -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/console -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/driver/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/esp-tls -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/esp32/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/esp_adc_cal/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/esp_event/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/esp_http_client/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/esp_http_server/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/esp_https_ota/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/esp_https_server/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/esp_ringbuf/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/ethernet/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/expat/expat/expat/lib -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/expat/port/include -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/fatfs/src -I C:/msys32/home/sy.nguyen-duc/esp/esp-idf/componen [... cut ...] home/sy.nguyen-duc/esp/esp-idf/components/bootloader/Kconfig.projbuild                C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/esptool_py/Kconfig.projbuild             C:/Users/sy.nguyen-duc/Desktop/workpace/HelloWorld/main/Kconfig.projbuild         C:/msys32/home/sy.nguyen-duc/esp/esp-idf/components/partition_table/Kconfig.projbuild                 " --env           "IDF_CMAKE=n"

14:46:13 Build Failed. 2 errors, 0 warnings. (took 40s.700ms)

`

Can you help me to resolve? Thanks all

ziolelle, 2018-12-11:
Same for me

kakaducsy, 2018-12-12:
Hi @ziolelle @SMHRambo 
What is the your folder name? ( I scared "sy.nguyen-duc" contains special characters . and -)

SMHRambo, 2018-12-12:
my path is "E:\Entwicklungsumgebung\ESP32\home\sascha\esp\esp-idf"
But if check out the "ESP-IDF Release v3.1.1" branch i have no errors.

ulfalizer, 2018-12-12:
Maybe `$COMPONENT_KCONFIGS_PROJBUILD` is set to the MSYS2-internal *nix path, while Python expects the Windows path. This is the line the error points to:

```
source "$COMPONENT_KCONFIGS_PROJBUILD"
```

Kconfiglib itself doesn't do any special path processing. It's probably some kind of mismatch between the Python version being used and MSYS2.

DMPatod, 2018-12-12:
I've got this error too

DMPatod, 2018-12-12:
The master branch is broken right now. I've checkout on 964f5a91f7fe2d2168d58db5314e11c91289aa17 commit, then the build returned to work

kakaducsy, 2018-12-13:
I try " https://github.com/espressif/esp-idf/releases" ESP-IDF Release v3.1.1 as @SMHRambo comment ,
and build ok, it only have 2 warnings
c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:esp32.common.ld:39: warning: memory region `rtc_data_seg' not declared
c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:esp32.common.ld:66: warning: memory region `rtc_data_location' not declared

JOO200, 2018-12-13:
Same problem.
I add the reference to #2722

renzbagaporo, 2018-12-13:
Hi all, @ulfalizer is correct. The kconfig paths are not passed to the linker script generator (ldgen.py) properly unlike when build is invoked from MSYS. 

I've submitted a fix internally, the patch file is provided below for you to verify if it works.

[0004-ldgen-fix-issue-when-not-built-in-msys2.patch.txt](https://github.com/espressif/esp-idf/files/2675962/0004-ldgen-fix-issue-when-not-built-in-msys2.patch.txt)

@kakaducsy Although just warnings, those seem to be important ones. These might have been fixed by later commits, though. If you tried the patch on the latest master and those warnings still persist, I encourage you to file a separate issue for that.

closed 2018-12-14 by igrr

noyelseth, 2019-09-15:
build failed on Ubuntu

Generating esp32.project.ld
/home/noyelseth/esp/esp-idf/Kconfig:21: warning: Kconfiglib expands environment variables in strings directly, meaning you do not need 'option env=...' "bounce" symbols. For compatibility with the C tools, rename IDF_TARGET_ENV to IDF_TARGET (so that the symbol name matches the environment variable name).
Traceback (most recent call last):
  File "/home/noyelseth/esp/esp-idf/tools/ldgen/ldgen.py", line 150, in <module>
    main()
  File "/home/noyelseth/esp/esp-idf/tools/ldgen/ldgen.py", line 114, in main
    sdkconfig = SDKConfig(kconfig_file, config_file)
  File "/home/noyelseth/esp/esp-idf/tools/ldgen/sdkconfig.py", line 50, in __init__
    self.config = kconfiglib.Kconfig(kconfig_file)
  File "/usr/local/lib/python2.7/dist-packages/kconfiglib.py", line 1038, in __init__
    self._parse_block(None, self.top_node, self.top_node)
  File "/usr/local/lib/python2.7/dist-packages/kconfiglib.py", line 2875, in _parse_block
    if self.srctree else "unset or blank"))
kconfiglib.KconfigError: /home/noyelseth/esp/esp-idf/Kconfig:67: '/home/noyelseth/esp/esp-idf/components/app_update/Kconfig.projbuild /home/noyelseth/esp/esp-idf/components/bootloader/Kconfig.projbuild /home/noyelseth/esp/esp-idf/components/esptool_py/Kconfig.projbuild /home/noyelseth/esp/esp-idf/components/partition_table/Kconfig.projbuild' not found (in 'source "$COMPONENT_KCONFIGS_PROJBUILD"'). Check that environment variables are set correctly (e.g. $srctree, which is set to '/home/noyelseth/esp/esp-idf'). Also note that unset environment variables expand to the empty string.
/home/noyelseth/esp/esp-idf/components/esp32/Makefile.projbuild:17: recipe for target '/home/noyelseth/esp/esp_workspace/hello_world/build/esp32/esp32.project.ld' failed
make: *** [/home/noyelseth/esp/esp_workspace/hello_world/build/esp32/esp32.project.ld] Error 1

[image]