espressif/esp-idf #2813: BLE Data loss when running for many hours

jakkra, 2018-12-10:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP32 chip]
- IDF version f11ac037c4958fc68e67a50134a0ce9a1634008c
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description
We have had problems since always that we lose some data over BLE. It seems as it's most of the time one MTU of data. If we run with 23 MTU (20 bytes data) we loose 20 bytes or MTU 245 we lose 242 bytes of data. The problem seem to be worse with low connection interval, i.e. 7.5ms. But we also have the problem with 30ms connection interval. 

The data loss has been improving over each esp-idf version.
Sending data to  NRF 52832 with ESP-IDF versions:
- 2.1 data loss after 30 seconds
- 3.0 data loss after 5 minutes
- 3.1 without modem-sleep data loss after 20 minutes
- 3.1 + BT modem sleep/power save we can run for multiple hours before loss.

So all that progress made with the **same application code**. Note that enabling modem sleep caused less data loss.
Another thing we noticed is that if we put the BT controller on core 1 instead of 0, we start loosing alot more data. **Maybe that could be some hint that there could be some issue with the concurrency issue in the BT controller?**

Also with recent improvements in the ESP-IDF we have started seeing a few, not many RX errors, i.e that ESP32 does not receive all data sent to it. This has not showed up earlier, but is a new issue which we see now.

We have not yet managed to reproduce our problems with any example application either, which is very unfortunate. We also have other problems which we have only managed to reproduce with our application, i.e. https://github.com/espressif/esp-idf/issues/2467. Maybe our application could be triggering some scenario/case/timing issues which can't be seen with example applications.

### Expected Behavior
No data loss over BLE
### Actual Behavior
We see lost BLE data
### Steps to repropduce

## Debug Logs

## Other items if possible
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2662432/sdkconfig.1.zip)

Dual mode BLE/Classic
The application and Bluetooth stack is running on core 1
BT Controller is running on 0

Weijian-Espressif, 2018-12-11:
@jakkra  how to reproduce the issue ? which command do you use,   `read`, `write`, `prepare write` or 'notify '?

jakkra, 2018-12-11:
Hi @Weijian-Espressif, there is not a straight forward way to reproduce it. I have not been able to reproduce it by tweaking the example applications.
What we do is sending duplex data over BLE, throughput is about 250 kbps in each direction, connection interval 17.5ms and latency 0 (the lower the worse it gets)

I have reproduced the same issue (I hope, it's the same issue, it looks the same) using two instances of our application running on Windows and using two ESP32's as HCI over UART. Where on one side our application acts as central and on the other peripheral. 

Also we are not using Bluedroid stack, but Bluekitchen, however I don't think it's related to the stack as I have logged every HCI packet both on Central and Peripheral side and by studying the two HCI logs, I can clearly see that all data the peripheral application sent is shown in the HCI log, but on the HCI log for the Central I can see that one packet is missing (244 bytes as we negotiate 247 MTU).

If the issue found using ESP32 only as HCI dongle is the same I see running our application between two ESP32's or between ESP32 and a NRF chipset, we can conclude that it's not our application configuration that triggers the error, but maybe instead the behaviour of it. As I seem to have reproduced the same issue when our application is running on Windows with ESP32 as HCI dongle. 

From what I have seen it seems more common that it's the Central which is missing a packet. So problem can be either that the peripheral never sends the packet, or that the Central receives it, but does not report it over HCI to the host/stack for some reason. 

I have two HCI dump files each about 500MB I can send you and also point you to the error in them, but not sure if that will help you.

I'm not sure how I can assist you in debugging this issue. Have you got any idea? One idea is that I compile and modify our application as a windows .exe
If I can get that to work then you would be able to run two instances of our application and providing 2 ESP32's as HCI over UART you should be able to reproduce the issue at your side too.

Weijian-Espressif, 2018-12-12:
@jakkra,  please provide me some log, thanks.

jakkra, 2018-12-12:
@Weijian-Espressif Do you have an email I can send to?

Weijian-Espressif, 2018-12-12:
@jakkra  zhiweijian@espressif.com

jakkra, 2018-12-12:
@Weijian-Espressif I sent you an email. I can add that data is lost also from central to peripheral. So the problem does not seem to have anything to do with which role ESP32 has.

Weijian-Espressif, 2018-12-13:
@jakkra, I have received the mail, thanks.

Weijian-Espressif, 2018-12-17:
@jakkra , Do you use flow control? Do you check  flow control before sending the ACL packet?

Weijian-Espressif, 2018-12-17:
@jakkra, Sorry, `esp_vhci_host_check_send_available()`  is not a function of flow control.
you can refer to 4.1 in vol 2 part E of core_v4.2 .

1. The LE_Read_Buffer_Size command is used to read the maximum size of the data portion of HCI LE ACL Data Packets sent from the Host to the Controller. The Host will segment the data transmitted to the Controller according to these values, so that the HCI Data Packets will contain data with up to this size.
2. The Number Of Completed Packets event is used by the Controller to indicate to the Host how many HCI Data Packets have been completed (transmitted or flushed) for each Connection_Handle since the previous Number Of Completed Packets event was sent to the Host. This means that the corresponding buffer space has been freed in the Controller.
3. you can refer to  `l2c_link_check_send_pkts ()`
and `btu_hcif_num_compl_data_pkts_evt()` in ESP_IDF

Weijian-Espressif, 2018-12-17:
@jakkra, it is `HOST TO CONTROLLER DATA FLOW CONTROL`, Before sending an ACL packet, the peripheral device should check whether the controller still has free buffers.

Weijian-Espressif, 2018-12-21:
@jakkra,  you can send ACL packet on  core 1 or on core 0, but  can't switch dynamically.  I test our example, I don't reproduce data loss. I suggest you use our example to do the test, and capture the hci log.

Weijian-Espressif, 2019-03-22:
@jakkra,  After the connection parameter update is fixed, Does this issue still appear?

jakkra, 2019-03-22:
Hi @Weijian-Espressif thank you, I'll verify in our long running tests over the weekend if issues still exists. It's my fault I have not updated you with more info on this issue, hopefully we will priortize at our side and I'll get more time to sit down with this issue. I have pretty much confirmed that the data is lost after we passed it to the BT Controller and that it's not a flow control issue. I will gather all data possible for you when I get the time. I think we can close this issue on Github, as we have this issue also created at your gitlab. Then we don't need to keep the information updated at two locations. So when I get time I'll come back to you regarding this issue.

Alvin1Zhang, 2019-04-18:
@jakkra Hi, would you help advise if any updates for the issue? Thanks.

jakkra, 2019-04-18:
Hi @Alvin1Zhang We still see problems with data loss. We have actualy made some more investigation. We will try to summurize our finding and collect more data for you next week. But I can give you a short summary of what we see:
Previously we had to run for many hours before we got lost packet, but now we have found that it's alot easier to reproduce using 23 MTU (default MTU) No Data Length Extention and Duplex BLE data transfer between two ESP32 with shortest connection interval 6 (7.5ms). Sent packets are mostly 20 bytes long, but every ~10th packet is 1 byte only.

Now we see after just a minute or so data error. As it now is alot less data that needs to be sent before error, we can sniff and easier find if the packet was sent at all or not.
What we see is that the sniffer picks up the "missing packet" but the ESP32 receiver does not. The packet does not arrive to the stack from BT Controller. 

Our current theory is that when the ESP 32 BT Controller and Radio receives alot of packets, somehow one packet get lost. As this happens with shortest connection interval and small packets => lots of packets. 

We have reproduced it with 6, 8 connection interval, for 10  (1.25ms units) we have not yet seen it yet. But the time it takes for error is alot longer with 8 connection interval than 6. 

Also allowing DLE and MTU negotiation the problem goes away (or at least takes alot of time befor eit happens)

Alvin1Zhang, 2019-04-18:
@jakkra Hi, thanks for the updates.

jakkra, 2019-05-02:
Hi, I also posted this on our/your internal gitlab issue, so let's keep further discussion there, ok?

We have found a faster way to reproduce the data loss issue now and have collected information for you. 

### Setup
Python script generates and sends data over UART-**>ESP32_Central->BLE->ESP32_Peripheral**->UART->Python script validates the data is correct.
Script is also ran the other way around UART->**ESP32_Peripheral->BLE->ESP32_Central**->UART->Python script validates the data is correct.

Meaning data transfer is duplex.

**Connection interval is 6 (7.5ms), no MTU or Data Length Extention negotiation.**

The data sent looks like "6442PACKET_NUM_FFFF\n" to make packets easy to find in the sniffer. Where the number is the packet number, increased by 1 for each packet sent.

In addition to the check in python that the received data is correct, we added an internal check on both ESP32's directly in the packet callback:

```
static const esp_vhci_host_callback_t vhci_host_cb = {
    .notify_host_send_available = host_send_pkt_available_cb,
    .notify_host_recv = host_recv_pkt_cb,
};

static int host_recv_pkt_cb(uint8_t *data, uint16_t len){

    if (xPortInIsrContext()){
        report_recv_called_from_isr();
        return 0;
    }

    xSemaphoreTake(ring_buffer_mutex, portMAX_DELAY);
    if (data[7] == 4) { // ATT CID
		// Here we verify that all data received is correct. i.e. that we get data looking like "6442PACKET_NUM_FFFF\n"
    }
    // Omitted code
}
```

And also right before we call `esp_vhci_host_send_packet` we validate that the packet number and data is correct.

In addition we also have the sniffer log of all data sent.

By combining all data from python scripts, internal logging on ESP32 and the sniffer log we can draw a few conclutions:

- As we can see the data which is not received on the ESP32 in the sniffer and also don't see two packets the same (doublette), which we see on RX side in the ESP32.
We can conclude that the data is successfully sent.

- By checkig incoming data straight in the `notify_host_recv `callback, we can rule out the issue is somewhere in our application on the RX side.

- We also see that the internal check in the `notify_host_recv `callback detectes the same issues as the python script error checking.

- We have a data check before `esp_vhci_host_send_packet` that also verifies that there is no issue in our application that we send two identical packets.

We see two different kind of errors, the most common is that a packet sent by the Central is not received on the Peripheral ESP32 (we see it in the sniffer).
Second problem is that we get two copies of the same packet (a doublet), in the sniffer we actually see two packets which have same data.

## Logs
Logs from python scripts when we get one missing packet and one doublet:
Those logs will probably not tell you much, but might help to explain the issue.

[logs_and_py_files.zip](https://github.com/espressif/esp-idf/files/3138807/logs_and_py_files.zip)

### ESP32 Peripheral python log
```
$ python tools/data_checker_generator.py 98
COM98 open
Starting read thread
Starting write thread
DATA ERROR. CURRUPT OR MISSING PACKET
Expected packet num: 5554 but was 5555
5555PACKET_NUM_FFFF

DATA ERROR. CURRUPT OR MISSING PACKET
Expected packet num: 6443 but was 6442
6442PACKET_NUM_FFFF

All sent, send STOP and flush
Sent 400200 bytes
Found STOP
Received 400200 bytes
All Done!
```

### ESP32 Central python log
```
$ python tools/data_checker_generator.py 159
COM159 open
Starting read thread
Starting write thread
Found STOP
Received 400200 bytes
All sent, send STOP and flush
Sent 400200 bytes
All Done!
```

### ESP32 Peripheral target check
```
btstack_port_esp32.c:00552: DATA ERROR. CURRUPT OR MISSING PACKET
btstack_port_esp32.c:00553: Expected packet num: 5554 but was 5555
btstack_port_esp32.c:00552: DATA ERROR. CURRUPT OR MISSING PACKET
btstack_port_esp32.c:00553: Expected packet num: 6443 but was 6442
```

Sniffer log:
[Sniffer log found here](https://github.com/espressif/esp-idf/files/3138807/logs_and_py_files.zip)

Packet number 29482 shows the packet which was not received from ESP32 BT Controller in Peripheral

Packet number 33912 and 33924 shows two packets with same data beeing sent twice.

## Summary
From what it looks like there are two issues, lost packet and double packet. They seem not really related as one seems to be an RX issue (packet sent correctly) and the other a TX issue (packet sent twice).
We also see similar problems running with longer connection interval and with MTU and DLE negotiation, however it takes longer for the issue to appear.

The issue with a packet duplicate is alot less common than the lost packet, so I'd focus on the lost packet first.
We can reproduce lost packet issue in under 30s, so it's quick to test.

We can only reproduce the issue with our application, maybe if you can provide us with a bt lib with some relevant printout when things seems to go wrong it will be easier for you to debug? Is there some case when your BT Controller will throw away an incoming packet?

jakkra, 2019-11-21:
Finally we can close this issue! Fixed in https://github.com/espressif/esp-idf/commit/039f9d0a20c35e7f3535975e5ccef1dcaf6e4a5a

Thanks!

closed 2019-11-21 by jakkra