espressif/esp-idf #2815: aws iot watchdog triggered during connect to aws

chegewara, 2018-12-10:
@mahavirj my code is based on aws iot example:
```
E (10340) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10340) task_wdt:  - IDLE1 (CPU 1)
E (10340) task_wdt: Tasks currently running:
E (10340) task_wdt: CPU 0: IDLE0
E (10340) task_wdt: CPU 1: aws_iot_task
```
this issue is always in somewhere in this code:
```
    do {
        rc = aws_iot_mqtt_connect(&client, &connectParams);
        if(SUCCESS != rc) {
            ESP_LOGE(TAG, "Error(%d) connecting to %s:%d", rc, mqttInitParams.pHostURL, mqttInitParams.port);
            vTaskDelay(1000 / portTICK_RATE_MS);
        }
    } while(SUCCESS != rc);

    /*
     * Enable Auto Reconnect functionality. Minimum and Maximum time of Exponential backoff are set in aws_iot_config.h
     *  #AWS_IOT_MQTT_MIN_RECONNECT_WAIT_INTERVAL
     *  #AWS_IOT_MQTT_MAX_RECONNECT_WAIT_INTERVAL
     */
    rc = aws_iot_mqtt_autoreconnect_set_status(&client, true);
    if(SUCCESS != rc) {
        ESP_LOGE(TAG, "Unable to set Auto Reconnect to true - %d", rc);
        abort();
    }
```

chegewara, 2018-12-10:
To not open new issue, this is related to aws iot, here are logs. Could someone explain this:
```
I (772892) wifi: bcn_timout,ap_probe_send_start
I (775397) wifi: ap_probe_send over, resett wifi status to disassoc
I (775397) wifi: state: run -> init (100)
I (775399) wifi: pm stop, total sleep time: 707105600 us / 771459313 us

I (775403) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (775410) esp32_ap_eventHandler: wifi event no: 5
E (775949) aws_iot: failed! mbedtls_ssl_write returned -0x4e
W (775953) subpub: MQTT Disconnect
I (775953) subpub: Auto Reconnect is enabled, Reconnecting attempt will start now
I (777830) esp32_ap_eventHandler: wifi event no: 5
I (777952) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (777953) wifi: state: init -> auth (b0)
I (777957) wifi: state: auth -> assoc (0)
I (777963) wifi: state: assoc -> run (10)
I (777979) wifi: pm start, type: 1

I (777982) esp32_ap_eventHandler: wifi event no: 4
I (778695) event: sta ip: 192.168.0.10, mask: 255.255.255.0, gw: 192.168.0.1
I (778696) esp32_ap_eventHandler: wifi event no: 7
I (778698) subpub: AWS IoT SDK Version 3.0.1-
I (778703) subpub: Connecting to AWS...
Heap summary for capabilities 0x00000800:
  At 0x3ffca2d4 len 32767 free 12200 allocated 20492 min_free 0
    largest_free_block 9356 alloc_blocks 7 free_blocks 3 total_blocks 10
  At 0x3ffae6e0 len 6432 free 0 allocated 6300 min_free 0
    largest_free_block 0 alloc_blocks 25 free_blocks 0 total_blocks 25
  At 0x3ffc69a0 len 104032 free 200 allocated 103152 min_free 0
    largest_free_block 76 alloc_blocks 157 free_blocks 5 total_blocks 162
  At 0x3ffe0440 len 15072 free 76 allocated 14876 min_free 0
    largest_free_block 40 alloc_blocks 19 free_blocks 3 total_blocks 22
  At 0x3ffe4350 len 113840 free 28644 allocated 75920 min_free 872
    largest_free_block 21464 alloc_blocks 2265 free_blocks 46 total_blocks 2311
  At 0x40097cdc len 33572 free 33536 allocated 0 min_free 33528
    largest_free_block 33536 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 74656 allocated 220740 min_free 34400 largest_free_block 33536
E (780918) aws_iot: failed! mbedtls_ssl_setup returned -0x7f00
E (780917) aws_iot: failed! mbedtls_ssl_setup returned -0x7f00
W (781931) subpub: MQTT Disconnect
W (781932) subpub: Auto Reconnect not enabled. Starting manual reconnect...
```
Basically i know whats going on, i have memory leak, then mbedTLS failed with -0x7f00, because lack of heap. What is strange to me its this:
```
W (775953) subpub: MQTT Disconnect
I (775953) subpub: Auto Reconnect is enabled, Reconnecting attempt will start now
```
```
W (781931) subpub: MQTT Disconnect
W (781932) subpub: Auto Reconnect not enabled. Starting manual reconnect...
```

mahavirj, 2018-12-11:
@chegewara  (At first look) it looks like some form of memory corruption, how about putting watchpoint around `pClient->clientStatus.isAutoReconnectEnabled`?

chegewara, 2018-12-11:
@mahavirj could you help me with this?
I am subscribing to few topics:
```
I (14309) subscribe_topic: Subscribing... => /test/FNT/4
I (14403) subscribe_topic: Subscribing... => /test/ota
I (14504) subscribe_topic: Subscribing... => /all/ota
I (14607) subscribe_topic: Subscribing... => /test/reset
```
`test/ota` and `all/ota` are using the same callback function:
```
    memset(TOPIC, 0, 100);
    sprintf(TOPIC, "/%s/%s", name, "ota");
    rc = subscribe_topic(&client, TOPIC, ota_subscribe_callback_handler);
    if(SUCCESS != rc) {
        ESP_LOGE(TAG, "Error subscribing : %d ", rc);
        abort();
    }

    rc = subscribe_topic(&client, "/all/ota", ota_subscribe_callback_handler);
    if(SUCCESS != rc) {
        ESP_LOGE(TAG, "Error subscribing : %d ", rc);
        abort();
    }
```
The issue is that all topics works but `/test/ota`. I cant find the problem, the is no error during subscription, even if i stop using `/all/ota` i dont receive messages from `/test/ota`. I could assume that maybe `/test` topic is somehow filtered by aws, but `/tst/reset` works.

Alvin1Zhang, 2018-12-18:
@chegewara Hi, could you help share if any updates for this issue? Thanks.

chegewara, 2018-12-18:
Yes, i can close it now. Most likely it was problem with my app (aws iot example code) or sdkconfig settings. Thanks.

closed 2018-12-18 by chegewara