espressif/esp-idf #2816: `esp_mesh_send` does not send big buffers

mandreyel, 2018-12-10:
## Environment

- IDF version (run ``git describe --tags`` to find it): v3.1.1-74-g0b3fdf420 (but the same behavior occurs on the latest master commit)
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     Windows|Linux
- Power Supply:         external 5V

## Problem Description

[`esp_mesh_send`](https://espressif-docs.readthedocs-hosted.com/projects/esp-idf/en/latest/api-reference/mesh/esp_mesh.html#_CPPv213esp_mesh_sendPK11mesh_addr_tPK11mesh_data_tiA_K10mesh_opt_ti) fails to send large buffers (haven't had enough time to figure out the precise number but ours failed at around 5000 bytes) with the `ESP_ERR_MESH_ARGUMENT` error code, however, the documentation seems not to mention any upper bound. We're using a somewhat modified code of the internal communication example found in this repo.
Is this a bug in the implementation, or is the enforced upper bound valid but is  not mentioned in the documentation?

ESP-MESH-QI, 2018-12-11:
esp_mesh_send does not support Fragmentation now. One packet size is limited to MESH_MTU( defined in ./esp-idf/components/esp32/include/esp_mesh.h).

ESP-MESH-QI, 2018-12-11:
Thanks for your advice. We will consider adding this part of the comment.

Alvin1Zhang, 2018-12-18:
@mandreyel Will temporarily close this issue, feel free to reopen. Thanks.

closed 2018-12-18 by Alvin1Zhang

Suraj-singh999898, 2020-08-10:
Does it resolve?
Problem > Root can't send more than 1400 char approximate to other nodes!

IDF version > release/v4.0 (v4.0.1)
Compiler version > xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Mesh example using > [https://github.com/espressif/esp-idf/tree/master/examples/mesh/internal_communication](url)