espressif/esp-idf #2818: gdbstub should also report other task status

X-Ryl669, 2018-12-10:
When using gdbstub (nice work by the way!) over UART via `make monitor`, one can observe the current stack trace for the current *task*.

Would it be possible to have the other threads/tasks listed too (by updating the gdbstub command parser ?) and ideally their callstack ?

That would involve dealing with FreeRTOS internals to get their addresses, but since the CPU are stalled and the scheduler is stopped, it should be possible to fetch them, right ?

Also, what is the reason we can't resume ? As I understand it, if the interruption was based on an abort, we could be resuming the CPU by setting the PC on the next instruction and restarting the other CPU, right ?

I was wondering if it would be possible to have a "debug point" (that is: a code that's triggering a hw breakpoint) so that it would only stop a single CPU and could be resumed by the other CPU via ERI
That way, instead of:
```
exceptionHandler() {
disable interrupt for CPU
disable watchdog
stall other CPU
start gdbStub
```
We could have:
```
userCode() {
    [whatever]
    debug_point(); // CPUX stalled in breakpoint
}
userConnectedViaGDBAndTriggeredAHighLevelInterruptViaWhateverWay() {
   // This could only be called on the other CPU, since CPUX is stalled
   disableInterrupt();
   disableWatchdog();
   start smartGDBStub() with extended command;
}

In smartGDBStub(), support setting the PC or resuming from breakpoint for the other CPU (or setting the next breakpoint) to unstall it 
```

This way, we could debug a single CPU core using the other CPU core. This would probably break time sensitive stuff like wifi, but for most work that would be very useful anyway.

igrr, 2018-12-11:
Hi @X-Ryl669, Gdb Stub can indeed be modified to be thread-aware. This makes it a bit more complex and possibly fragile, it would need to check all the pointers as it walks the task lists, otherwise it might itself cause an exception if some of FreeRTOS memory was corrupted.

Will track this as an enhancement request, but it might take a while before it gets handled (there are some other pending debugging related tasks right now).

X-Ryl669, 2018-12-12:
I don't know enough (yet) about ERI.
I supposed there was some kind of communication between the CPU via registers, but you seems to imply there's none ?

I've made a PR for the implementation of this task support in gdbstub. If you have time, please review it. It's working well, but for post-mortem analysis only (no resuming yet).

I'll need more help to implement live debugging since I'm new to Xtensa assembly / debug level interrupt. If you intend to work on this, I can probably help.

BTW, a useful command to see what's going on in GDB's remote protocol is this:
`xtensa-esp32-elf-gdb -ex "set debug remote 1" -ex "set serial baud 115200" -ex "target remote /dev/cu.SLAB_USBtoUART" -ex "interrupt /path/to/app.elf"`

X-Ryl669, 2019-01-18:
BTW, when the PR will be merged, I think it'll be possible to implement real live debugging via GDB over UART. 
My idea being to install a debug breakpoint with some function (for example `esp_enter_gdb`).
When the function is called, we detect this in this exception handler (so `make monitor` starts gdb), and we install a DEBUG level exception handler. We stall the other CPU to avoid memory corruption like it's done now. We then have to add the breakpoint writing GDB commands in the code and let GDB resumes when it's done placing the breakpoints and continueing. 
Upon continueing, we'll have to unstall both CPU (is it really required, or can we move all tasks back to the remaining CPU ?) and then everything should work smoothly.

We can have a shared "mailbox" between CPU by using 2 arrays (each dedicated to a CPU) and use DEBUG level exception to force a CPU to read its mail. It's a shame we can't read other CPU core's register, but this can be overcome by triggering a exception on that CPU and having it dump its register in the handler. Said differently, I think it's totally possible to implement realtime debugging on ESP32 with GDB over the serial link.

igrr, 2022-03-14:
Looks like we forgot to close this issue when the PR got merged (https://github.com/espressif/esp-idf/commit/b797f05558f5c902f40ea4b9f24e31e1ffdf8106). Thanks again for your contribution @X-Ryl669!

> BTW, when the PR will be merged, I think it'll be possible to implement real live debugging via GDB over UART.

Some time has passed since you gave this suggestion, and it's partially implemented now. There is an example over at https://github.com/espressif/esp-idf/tree/master/examples/system/gdbstub which demonstrates very minimal run-control GDB debugging via UART.

This is going to be extended soon with the ability to set breakpoints and do other operations you'd expect to be able to do using GDB.

closed 2022-03-14 by igrr

X-Ryl669, 2022-03-15:
Thanks! I'm waiting for the breakpoint setting feature though!