espressif/esp-idf #2819: esp-idf htu21d i2c timeout

PawWaw, 2018-12-10:
I'm trying to write program on ESP32-PICO-D4 using htu21d, but I got some troubles with code. I'm using esp-idf.
```

int htu21d_init() {
	
	esp_err_t ret;
	
	// setup i2c controller
	i2c_config_t conf;
	conf.mode = I2C_MODE_MASTER;
	conf.sda_io_num = 18;
	conf.scl_io_num = 19;
	conf.sda_pullup_en = GPIO_PULLUP_DISABLE;
	conf.scl_pullup_en = GPIO_PULLUP_DISABLE;
	conf.master.clk_speed = 100000;
	ret = i2c_param_config(I2C_NUM_0, &conf);
	if(ret != ESP_OK) return HTU21D_ERR_CONFIG;
	
	// install the driver
	ret = i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0);
	if(ret != ESP_OK) return HTU21D_ERR_INSTALL;
	
	// verify if a sensor is present
	i2c_cmd_handle_t cmd = i2c_cmd_link_create();
	i2c_master_start(cmd);
	i2c_master_write_byte(cmd, (HTU21D_ADDR << 1) | I2C_MASTER_WRITE, true);
	i2c_master_stop(cmd);
        ret = i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
	if( ret == ESP_OK)
        ESP_LOGI(TAG, "OK");
        else
		ESP_LOGE(TAG, "I2C Timeout");

        cmd = i2c_cmd_link_create();
	i2c_master_start(cmd);
	i2c_master_write_byte(cmd, (HTU21D_ADDR << 1) | I2C_MASTER_WRITE, true);
	i2c_master_write_byte(cmd, 0xF3, true);
	i2c_master_stop(cmd);
	ret = i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
	i2c_cmd_link_delete(cmd);
	if( ret == ESP_OK)
        ESP_LOGI(TAG, "OK");
       else
		ESP_LOGE(TAG, "I2C Timeout");
	
	// wait for the sensor (50ms)
	vTaskDelay(50 / portTICK_RATE_MS);
	
	// receive the answer
	uint8_t msb, lsb, crc;
	cmd = i2c_cmd_link_create();
	i2c_master_start(cmd);
	i2c_master_write_byte(cmd, (HTU21D_ADDR << 1) | I2C_MASTER_READ, true);
	i2c_master_read_byte(cmd, &msb, 0x00);
	i2c_master_read_byte(cmd, &lsb, 0x00);
	i2c_master_read_byte(cmd, &crc, 0x01);
	i2c_master_stop(cmd);
	ret = i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
	i2c_cmd_link_delete(cmd);
	if( ret == ESP_OK)
        ESP_LOGI(TAG, "OK");
        else
		ESP_LOGI(TAG, "I2C Timeout");

	
	return HTU21D_ERR_OK;
}

```
Everything is okay (ret is ESP_OK), until program goes to "receive the answer" part. The "i2c_master_cmd_begin" returns "I2C Timeout". I tried everything I know and what I found in the Internet, but nothing works... Anyone know, what am I doing wrong?

koobest, 2018-12-11:
@PawWaw 
Can you please try the  `receive the answer` part as the follows?
```
    uint8_t raw[3] = {0};
    cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, (HTU21D_ADDR << 1) | I2C_MASTER_READ, true);
    i2c_master_read(cmd, raw, 3, I2C_MASTER_LAST_NACK);
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
```
Can you provide a waveform of the communication? I think the  I2C bus broken caused the timeout.

PawWaw, 2018-12-11:
I don't know how to check the waveform :/ The problem didn't solve when I pasted your code.

koobest, 2018-12-11:
OK, We will buy this sensor for testing.

PawWaw, 2018-12-11:
Thank you very much. Close this thread?

PawWaw, 2018-12-11:
Don't buy this sensor. I'm afraid I'm just making some mistake somewhere in my code and because of that I get this timeout :(

koobest, 2018-12-11:
Do you have an oscilloscope that can capture waveforms for communication?

PawWaw, 2018-12-11:
No, I don't have.

koobest, 2018-12-11:
May be I can use your code to do a test, but use a sensor I can find now.

koobest, 2018-12-11:
Can you try to enable the  GPIO internal pull?  
```  
conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
```

PawWaw, 2018-12-11:
Sure, finally my code looks like that:

```
int htu21d_init() {
	
	esp_err_t ret;
	
	// setup i2c controller
	i2c_config_t conf;
	conf.mode = I2C_MODE_MASTER;
	conf.sda_io_num = 18;
	conf.scl_io_num = 19;
	conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
	conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
	conf.master.clk_speed = 100000;
	ret = i2c_param_config(I2C_NUM_0, &conf);
	if(ret != ESP_OK) return HTU21D_ERR_CONFIG;
	
	// install the driver
	ret = i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0);
	if(ret != ESP_OK) return HTU21D_ERR_INSTALL;
	
	// verify if a sensor is present
	i2c_cmd_handle_t cmd = i2c_cmd_link_create();
	i2c_master_start(cmd);
	i2c_master_write_byte(cmd, (HTU21D_ADDR << 1) | I2C_MASTER_WRITE, true);
	i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
	if( ret == ESP_OK)
        ESP_LOGI(TAG, "OK");
    else
		ESP_LOGE(TAG, "I2C Timeout");

    cmd = i2c_cmd_link_create();
	i2c_master_start(cmd);
	i2c_master_write_byte(cmd, (HTU21D_ADDR << 1) | I2C_MASTER_WRITE, true);
	i2c_master_write_byte(cmd, 0xF3, true);
	i2c_master_stop(cmd);
	ret = i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
	i2c_cmd_link_delete(cmd);
	if( ret == ESP_OK)
        ESP_LOGI(TAG, "OK");
    else
		ESP_LOGE(TAG, "I2C Timeout");
	
	// wait for the sensor (50ms)
	vTaskDelay(50 / portTICK_RATE_MS);
	
	// receive the answer
	uint8_t raw[3] = {0};
    cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, (HTU21D_ADDR << 1) | I2C_MASTER_READ, true);
    i2c_master_read(cmd, raw, 3, I2C_MASTER_LAST_NACK);
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
	if( ret == ESP_OK)
        ESP_LOGI(TAG, "OK");
    else
		ESP_LOGE(TAG, "I2C Timeout");

	
	return HTU21D_ERR_OK;
}
```
`
But it still gives "timeout" at receive the answer.

koobest, 2018-12-11:
Just find that  `verify if a sensor is present` may cause sensor status error, can you remove this part and take a try?

PawWaw, 2018-12-11:
No difference, still in the receiving is timeout.

koobest, 2018-12-11:
Ok, maybe this issue is caused by the sensor itself, and its timing is special. I will buy this sensor and try to support it.

koobest, 2018-12-11:
Can you try to wait for more time when reading the sensor?
```
// wait for the sensor (100ms)
vTaskDelay(100 / portTICK_RATE_MS);
```

PawWaw, 2018-12-11:
That was it... Thank you very much koobest, I would never think about it!
[image]

closed 2018-12-12 by costaud