espressif/esp-idf #2821: modbus_slave example crash while closing serial port

keelung-yang, 2018-12-11:
* ESP32-PICO-Kit
* Lastest esp-idf for windows on master branch
* Windows 7 SP1 64 bit
* Python 3.7.1 64 bit out of esp-idf
* pymodbus 2.1.0
* [sdkconfig.zip](https://github.com/espressif/esp-idf/files/2666260/sdkconfig.zip)

------
I modified uart mode and pin in modbus_slave example since I have no RS485 device connected to ESP32.
It runs successfully and I can read registers in that example. But ESP32 crash and reboot after closing serial port.
[image]

[image]
[image]
[image]

alisitsyn, 2018-12-12:
Hello keelung-yang,

It was possible to reproduce your issue. In your configuration the USB-serial bridge U4 is configured as a Modbus port. 
https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/get-started-pico-kit.html#get-started-pico-kit-v4-pin-notes-cmake

The board ESP32-PICO-KIT-V4 has USB bridge adapter connected to IO3 - TXD, IO1 - RXD. These pins are used by boot loader. Once your software flashed you use these pins for Modbus communication.
The bridge has also RTS and DTR lines which are used to automatically enter the bootloader. 
When you close the port from client software it makes a pulse on the EN pin which resets chip. 
So, the configuration that you use for Modbus connection is not recommended because of this.

The Modbus example has been checked with RS485 adapter first and then with your changes and pymodbus script similar to yours. It works just fine and does not have any issues in standard configurations with pymodbus or other clients.

Would you agree that the issue can be closed?

keelung-yang, 2018-12-18:
@alisitsyn Thanks for your infomation with details!

We want to reuse USB-Serial bridge to reduce connected devices. 

If we don't reuse it we must buy another USB-Serial device to communicate with PC on modbus RTU. This means there are two USB devices connected to PC, two USB ports occupied on PC, and only one of them as daily using. 

I know we can use WiFi and modbus TCP. But this solution need too many operations for our users. And WiFi is not always available on PC. 

IMO, bootloader is gone after booting, it shouldn't run if no requesting, so it shouldn't occupy or response UART. This is another reason we want to reuse USB-Serial bridge safely in our applications.

keelung-yang, 2018-12-19:
Maybe I should close this issue and fire new one:  Reuse USB-Serial bridge in application.

alisitsyn, 2018-12-19:
@keelung-yang,
I understand your needs but with the ESP32-PICO-Kit board will reset the ESP32 when you close the port and I don't see any way to avoid it. This is behavior defined by USB bridge, its driver and Q1, Q2 on the schematic. You can design your own board with the different algorithm to enter into boot loader mode. There are several ways to do it and I think it will work in your approach.

0rus26, 2022-10-13:
is any solution directly on hardware, like cutting out the trace betwen EN and the transistor S8050?

alisitsyn, 2022-10-14:
@0rus26,

It should not e a problem to use the USB bridge pins if you disconnect these two transistors. In this case you need to manually put chip into bootloader mode for programming.

0rus26, 2022-10-15:
Yes, thanks, that was the solution :3