espressif/esp-idf #2823: Not able to open port /dev/ttyUSB0

Kottapalli-Naveen, 2018-12-11:
INSTRUCTIONS
============

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    v3.3-dev-424-g8b885fb
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc --version
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

Not able to open serial port using putty or minicom.  

### Expected Behavior

### Actual Behavior

### Steps to repropduce - Can see below logs on console

Welcome to minicom 2.7                                                                                       
                                                                                                             
OPTIONS: I18n                                                                                                
Compiled on Nov 15 2018, 20:18:47.                                                                           
Port /dev/ttyUSB0, 22:07:20                                                                                  
                                                                                                             
Press CTRL-A Z for help on special keys                     +-----------------------------+                  
                                                            |                             |                  
ets Jun  8 2016 00:22:57                                    |  Cannot open /dev/ttyUSB0!  |                  
                                                            |                             |                  
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)     +-----------------------------+                  
configsip: 0, SPIWP:0x00                                                                                     
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00                                      
mode:DIO, clock div:1                                                                                        
load:0x3fff0008,len:8                                                                                        
load:0x3fff0010,len:1848                                                                                     
load:0x40078000,len:6712                                                                                     
load:0x40080000,len:252                                                                                      
entry 0x40080034

koobest, 2018-12-12:
Hi, @Kottapalli-Naveen
It is strange, but you got the console output log from ESP32. Restart the development board without exiting minicom, will this log (Cannot open /dev/ttyUSB0) still appear?

Alvin1Zhang, 2018-12-18:
@Kottapalli-Naveen Hi, could you help share if any updates for this issue? Thanks.

Kottapalli-Naveen, 2018-12-19:
> Hi, @Kottapalli-Naveen
> It is strange, but you got the console output log from ESP32. Restart the development board without exiting minicom, will this log (Cannot open /dev/ttyUSB0) still appear?

Extremely sorry for the late reply.  I got pulled into another priority task and didn't get time check on this again.  Will check and let you all know.

Thank you very much for the support.

closed 2018-12-30 by Kottapalli-Naveen