espressif/esp-idf #2824: how to set app_main stack size?

ashareye, 2018-12-12:
Hi All:

I run a test in app_main, it showed error:
> A stack overflow in task main has been detected.

so I want to change the main stack size ,  and I searched in menuconfig but nothing found, only a pthread stack size can be set. Is this the right one?

dhrishi, 2018-12-12:
@ashareye You can change it through menuconfig:
make menuconfig -> Component config -> ESP32-specific -> Main task stack size

ashareye, 2018-12-12:
I found it, thank you, dhrishi.

closed 2018-12-12 by ashareye

cima, 2019-12-06:
Hello can You please tell us where can we find _main_ thread stack size in esp-idf framework 4.0? 4.0 beta2 specifically. Mentioned menu item no longer exists in _menuconfig_. The closest I found is in pthread but there I assume the value is applied on new threads. Based on my experiments the _main_ thread stack size is different (bigger) than newly created threads.
[image]

dhrishi, 2019-12-18:
@cima It has been moved to:
`make menuconfig -> Component config -> Common ESP-related -> Main task stack size`

shawty, 2024-10-24:
I know this is closed... but I REALLY wish espressif would stop changing things between versions of the IDF.  I've just wasted 3 hours trying to track down what ended up being a 30 second fix!