espressif/esp-idf #2825: Problem compiling with can.h because of CAN_IO_UNUSED

arrowcircle, 2018-12-12:
## Environment

- Development Kit:      [none]
- Module or chip used:  [none]
- IDF version: v3.3-beta1-4-g23b6d40c5
- Build System: [Make]
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: [macOS]
- Power Supply: [none]

## Problem Description

Hi! I think I found the problem with CAN driver. It's impossible to compile the project because of CAN_IO_UNUSED is defined as int (not gpio_num_t).

### Expected Behavior

Project compiles successfully

### Actual Behavior

Error during the compilation

### Steps to repropduce

Use example from documentation https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/can.html#examples

### Code to reproduce this issue

```cpp
static const can_general_config_t g_config = {
  .mode = CAN_MODE_NORMAL,
  .tx_io = GPIO_NUM_18, .rx_io = GPIO_NUM_19,
  .clkout_io = CAN_IO_UNUSED, .bus_off_io = CAN_IO_UNUSED,
  .tx_queue_len = 0, .rx_queue_len = 5,
  .alerts_enabled = CAN_ALERT_NONE,
  .clkout_divider = 0
};
```

## Debug Logs

```
/Users/ionbuggy/esp/esp-idf/components/driver/include/driver/can.h:39:131: error: invalid conversion from 'int' to 'gpio_num_t' [-fpermissive]
                                                                    .alerts_enabled = CAN_ALERT_NONE,  .clkout_divider = 0,        }
                                                                                                                                   ^
```

## Other items if possible

I found a hack how to fix this, but for me it does not look great.
I added CAN_IO_UNUSED to GPIO definitions in gpio.h file and removed definition from can.h.
I am new to C and C++ programming, so don't know yet how to make it properly. If someone advice proper solution I can make pull request.

igrr, 2018-12-12:
You including can.h from a C++ source file, right?

arrowcircle, 2018-12-12:
@igrr Yes

arrowcircle, 2018-12-14:
@igrr any ideas?

igrr, 2018-12-14:
C allows implicit conversions between ints and enum values, C++ does not. Will fix by defining CAN_IO_UNUSED as `((gpio_num_t) -1)`.

arrowcircle, 2019-02-17:
@igrr Looks like its not been fixed yet. Any news on this?

Civilduino, 2020-03-14:
In the Miscellaneous macros section of the can.h file you are using, modify the line:
#define CAN_IO_UNUSED                   (-1)

to

#define CAN_IO_UNUSED                   (gpio_num_t)(-1)

stuartpittaway, 2021-01-08:
This issue still isn't fixed for C++ users

homonto, 2023-03-04:
2023 and still the same problem...

igrr, 2023-03-04:
@homonto As far as i can tell, the original problem has been fixed in all recent IDF versions. If you are still seeing this issue, could you please open a new ticket, filling in the issue template? We'll take a look!

commit 099b83af (2019-02-26): CAN: Add functions to clear queues and fix multiple bugs

This commits adds the functions can_clear_transmit_queue() and
can_clear_receive_queue(). Closes #2906

The following bug are fixed:

- CAN_IO_UNUSED is now explicitly cast to enum type. Closes #2825
- Fix multiple documentation errors. Closes #2898, Closes #2794
- can_reconfigure_alerts() returns incorrect current_alerts. Closes #3028
- Add missing header file. Closes #3065