espressif/esp-idf #2826: compilition fail if PROJECT_VER exceeds 32 bytes

vwitti, 2018-12-12:
In current master the application version is read in esp-idf/components/app_update/component.mk, line 14 to: 
GET_PROJECT_VER := $(shell cd ${PROJECT_PATH} && git describe --always --tags --dirty || echo "Not found git repo")

compilation fails if GET_PROJECT_VER exceeds limit of 32 characters as defined in esp_image_format.h
If you truncate it to first 31 significant bytes compilations works
...
GET_PROJECT_VER := $(shell cd ${PROJECT_PATH} && git describe --always --tags --dirty || echo "Not found git repo")
# due to limit in esp_app_desc_t for element version[32]
GET_PROJECT_VER := $(shell echo ${GET_PROJECT_VER}|cut -c1-31)
...

We encountered this issue on some special tags. Or you increase version field to 48 bytes
best regards
volker

projectgus, 2018-12-20:
Hi @vwitti ,

Thanks for reporting this. As it happens, a fix was just merged to master in our internal repo - 78487123bf80852f4ce11c788a3ee7f44b873833 . It should appear on GitHub in the next day or so.

Angus

projectgus, 2018-12-20:
Hi @vwitti ,

Thanks for reporting this. A fix was just merged to master in our internal repo. It should appear on GitHub in the next day or so.

Angus

vwitti, 2018-12-21:
Hi Angus,thank you very much .best regards and merry christmasvolkerAngus Gratton <notifications@github.com> hat am 21. Dezember 2018 um 00:47 geschrieben:  Hi @vwitti ,Thanks for reporting this. A fix was just merged to master in our internal repo. It should appear on GitHub in the next day or so.Angus

jimparis, 2019-03-06:
This can probably be closed thanks to c0f5e58bdc5aa01c04168fc0998fd100445bf081?

vwitti, 2019-03-10:
Hi,this issue can be closed, as it seems to be resolved. thanksvolkerjimparis <notifications@github.com> hat am 6. März 2019 um 04:26 geschrieben:  This can probably be closed thanks to c0f5e58?

closed 2019-03-10 by Alvin1Zhang