espressif/esp-idf #2827: Provisioning hangs on swift app

naderahmed, 2018-12-12:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

I downloaded:

https://github.com/espressif/esp-idf-provisioning-ios

ran pod install

clicked the provison button to do it by WiFi, nothing happened.
//Detailed problem description goes here.

### Expected Behavior
I expected it to provision

### Actual Behavior

Nothing happens.
I can see that it makes it's way to SendHTTPData , however nothing happens after that.
There is no error either. 

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

anujdeshpande, 2018-12-13:
Hi @naderahmed - I just verified that it works. Maybe an issue with the configuration in `menuconfig`?

Can you please check - 
- You are using `WiFi Sec 1 Debug` build in iOS
- In your IDF, you are using `examples/provisioning/softap_prov`
  - in the example, you have set `Example Configuration->Use Security Version 1`

If you haven't made changes to the PoP, the default `abcd1234` should be set on both the device as well as the smartphone app

naderahmed, 2018-12-13:
Hi Anuj  - infact if you can spend an hour with me, my company can offer you $30.00

naderahmed, 2018-12-13:
thank you for your response. My company is in a time crunch, as we plan on selling millions of these... so I am under pressure to test and verify if ESP32 really works. 

However.. I apologize, but I did not understand your instructions, I  need very basic steps. I am completely new. 

I also wanted to add a point which I forgot to mention - I got an error when I tried to build:

> error: Build input files cannot be found:
> esp-idf-provisioning-ios-master/EspressifProvision/EspressifProvision/proto/session.pb.swift'
> 
> 

it was actually for many files like that, therefore what I did was.

copy all the swift files from:

     /Users/nader/Documents/iOS/ESP32/esp-idf-provisioning-ios-master/EspressifProvision/EspressifProvision/proto

TO:

    /Users/nader/Documents/iOS/ESP32/esp-idf-provisioning-ios-master/proto

then it worked. I just wanted to let you know that, in case that could have been a problem.

Therefore, please help me understand your first instruction:

> Can you please check -
> 
>     You are using WiFi Sec 1 Debug build in iOS

I do not know how to do that, nor do I understand what it means. 
Can we chat briefly on skype ? I am available now. 

I am also available at 5:00 AM NYC time

or when do you suggest?

naderahmed, 2018-12-13:
Hello,
I will be available at 4:30 PM Pune, India time tomorrow(12-14) so we can quickly chat via skype.
skype id: naderwashere

naderahmed, 2018-12-13:
we are using the /esp/esp-idf/examples/wifi/smart_config

here is my skype id: naderwashere

Alvin1Zhang, 2019-03-21:
@naderahmed Hi, would you help share if any updates for the issue? Thanks.

closed 2019-03-21 by Alvin1Zhang