espressif/esp-idf #2829: websocket client in esp-idf

tharunjoy, 2018-12-13:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

JOO200, 2018-12-13:
Maybe checkout the cpp_utils from nkolban:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/WebSocket.cpp

tharunjoy, 2018-12-14:
Thanks for your response.
I am new to this, can you provide an example code for websocket client using this library.

JOO200, 2018-12-14:
I think you have first to setup a Socket connecting to your WebSocket Server.
( https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/Socket.h - use Socket::connect(..); )
At the second step you can make an instance from WebSocket with the Socket as parameter in the constructor.
I don't know how to exactly use the WebSocket because I only used the HttpServer before.
Maybe this [file ](https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/DesignNotes/WebSockets.md) helps or @nkolban can provide some examples.

chegewara, 2018-12-14:
https://github.com/nkolban/esp32-snippets/tree/master/sockets

projectgus, 2019-06-05:
Hi @tharunjoy ,

Sorry for the lack of an update on this issue. An official WebSockets client is being worked on for ESP-IDF V4.0, and this issue will be updated when it's available.

Angus

viettrungluu, 2019-06-17:
FWIW, I just wrote a WebSocket client, if this is of use to anyone else: https://github.com/tricot-io/xsp
(it's currently on top of a slightly patched esp-idf v3.2).

closed 2019-06-26 by igrr

tharunjoy, 2019-12-08:
@projectgus 

I am using older esp idf version. Can I manually add this new websocket component to my older version of esp idf?

projectgus, 2019-12-08:
@tharunjoy You haven't said what ESP-IDF version you are currently using, but anything is possible if you are prepared to some development work yourself and support the result yourself.

I would strongly recommend waiting for ESP-IDF v4.0 release, or trying the [v4.0-beta2](https://github.com/espressif/esp-idf/releases/tag/v4.0-beta2) version, though.

tharunjoy, 2019-12-09:
@projectgus  
Currently I am working on 'esptool.py v2.8-dev, 2.8-dev'. Since I am using 'Arduino over esp idf' component to support Arduino library and as this is not implemented over the latest IDF(https://github.com/espressif/arduino-esp32/issues/3201#issuecomment-530899455) version, I am forced to stick on this version. 

If I download the source code for official websocket library and add them to the my IDE version component file as an external library and add it in the component.mk file, will I be able to compile and use them?

projectgus, 2019-12-18:
@tharunjoy That is the esptool.py version not the ESP-IDF version, but I understand what you mean.

There is no easy way to integrate this into the IDF version used with Arduino, I'd recommend waiting for Arduino to update the IDF version. Sorry about that.

Ramakrishna247, 2023-04-12:
Hi I see there is IDF component esp_websocket_client in 4.4.3 , but it is not there in IDF v5.0 , why is that ? is it  got modified with new name ,  or else is it deprecated ?