espressif/esp-idf #2830: Heap corruption in negotiate security in blufi process

AadiMehta, 2018-12-13:
## Environment

- Development Kit:    [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v3]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32U]
- IDF version (run ``git describe --tags`` to find it): 
    v3.3-dev-424-g8b885fb
- Build System:         [Make]

## Problem Description
Intermittent issue while negotiating security in blufi process. This causes ESP32 to reboot. 

NOTE: This occurs intermittently. 1 out of 10 times this error is seen.

Following are the logs :
```
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
Core 0 register dump:
PC      : 0x400932d1  PS      : 0x00060233  A0      : 0x80094f72  A1      : 0x3ffe0640  
0x400932d1: uxPortCompareSet at /home/amehta/esp-idf/components/freertos/tasks.c:3537
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /home/amehta/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/amehta/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at /home/amehta/esp-idf/components/freertos/tasks.c:4231

A2      : 0x4008213c  A3      : 0x00060223  A4      : 0x00060220  A5      : 0x3ffe9254  
0x4008213c: _xt_user_exit at /home/amehta/esp-idf/components/freertos/xtensa_vectors.S:746

A6      : 0x00000001  A7      : 0x3ffe08d0  A8      : 0x0000abab  A9      : 0xb33fffff  
A10     : 0x0000cdcd  A11     : 0x00060220  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x0000cdcd  A15     : 0x3ffe96fc  SAR     : 0x0000001b  EXCCAUSE: 0x00000003  
EXCVADDR: 0x4008213c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
0x4008213c: _xt_user_exit at /home/amehta/esp-idf/components/freertos/xtensa_vectors.S:746

Backtrace: 0x400932d1:0x3ffe0640 0x40094f6f:0x3ffe0660 0x40094fa3:0x3ffe0680 0x400825dc:0x3ffe06b0 0x400827ef:0x3ffe06d0 0x40082a24:0x3ffe06f0 0x400daf6d:0x3ffe0710 0x400d906b:0x3ffe0730 0x400da23a:0x3ffe0750 0x400d880e:0x3ffe0dd0 0x400d7e28:0x3ffe0e00 0x40101b5f:0x3ffe0e50 0x40101043:0x3ffe0ea0 0x401011f3:0x3ffe0ee0 0x401025b1:0x3ffe1170 0x401171e9:0x3ffe11b0 0x40113a2d:0x3ffe11d0 0x40114029:0x3ffe1490 0x40114462:0x3ffe14f0 0x4011297d:0x3ffe1540 0x40112a35:0x3ffe1560 0x40119a77:0x3ffe1580 0x4010fcdd:0x3ffe15c0
0x400932d1: uxPortCompareSet at /home/amehta/esp-idf/components/freertos/tasks.c:3537
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /home/amehta/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/amehta/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at /home/amehta/esp-idf/components/freertos/tasks.c:4231

0x40094f6f: multi_heap_internal_lock at /home/amehta/esp-idf/components/heap/multi_heap.c:696

0x40094fa3: multi_heap_malloc_impl at /home/amehta/esp-idf/components/heap/multi_heap.c:696

0x400825dc: heap_caps_malloc at /home/amehta/esp-idf/components/heap/heap_caps.c:130

0x400827ef: heap_caps_calloc at /home/amehta/esp-idf/components/heap/heap_caps.c:130

0x40082a24: esp_mbedtls_mem_calloc at /home/amehta/esp-idf/components/mbedtls/port/esp_mem.c:25

0x400daf6d: mbedtls_calloc at /home/amehta/esp-idf/components/mbedtls/mbedtls/library/platform.c:59

0x400d906b: mbedtls_mpi_grow at /home/amehta/esp-idf/components/mbedtls/mbedtls/library/bignum.c:676

0x400da23a: mbedtls_mpi_exp_mod at /home/amehta/esp-idf/components/mbedtls/mbedtls/library/bignum.c:1716

0x400d880e: mbedtls_dhm_make_public at /home/amehta/esp-idf/components/mbedtls/mbedtls/library/dhm.c:292

0x400d7e28: blufi_dh_negotiate_data_handler at /home/amehta/ESP32-Node/webserver/main/src/utils/blufi_security.c:115

0x40101b5f: btc_blufi_protocol_handler at /home/amehta/esp-idf/components/bt/bluedroid/btc/profile/esp/blufi/blufi_protocol.c:124

0x40101043: btc_blufi_recv_handler at /home/amehta/esp-idf/components/bt/bluedroid/btc/profile/esp/blufi/blufi_prf.c:430

0x401011f3: blufi_profile_cb at /home/amehta/esp-idf/components/bt/bluedroid/btc/profile/esp/blufi/blufi_prf.c:186

0x401025b1: bta_gatts_send_request_cback at /home/amehta/esp-idf/components/bt/bluedroid/bta/gatt/bta_gatts_act.c:531

0x401171e9: gatt_sr_send_req_callback at /home/amehta/esp-idf/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

0x40113a2d: gatts_process_write_req at /home/amehta/esp-idf/components/bt/bluedroid/stack/gatt/gatt_sr.c:1159

0x40114029: gatts_process_attribute_req at /home/amehta/esp-idf/components/bt/bluedroid/stack/gatt/gatt_sr.c:1468

0x40114462: gatt_server_handle_client_req at /home/amehta/esp-idf/components/bt/bluedroid/stack/gatt/gatt_sr.c:1661

0x4011297d: gatt_data_process at /home/amehta/esp-idf/components/bt/bluedroid/stack/gatt/gatt_main.c:975

0x40112a35: gatt_le_data_ind at /home/amehta/esp-idf/components/bt/bluedroid/stack/gatt/gatt_main.c:562

0x40119a77: l2c_rcv_acl_data at /home/amehta/esp-idf/components/bt/bluedroid/stack/l2cap/l2c_main.c:267

0x4010fcdd: btu_hci_msg_process at /home/amehta/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:598
 (inlined by) btu_task_thread_handler at /home/amehta/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:229
```

### Steps to repropduce

1. Run blufi example.
2. Error seems to point to following line in [blufi_security ](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/blufi/main/blufi_security.c)file :
```
ret = mbedtls_dhm_make_public(&blufi_sec->dhm, (int) mbedtls_mpi_size( &blufi_sec->dhm.P ), blufi_sec->self_public_key, blufi_sec->dhm.len, myrand, NULL);
```

AadiMehta, 2018-12-16:
Any updates on this issue?

Thanks,
Aadi Mehta

Weijian-Espressif, 2018-12-17:
@AadiMehta,  I run blufi example with commit id 8b885fb, did not reproduce the issue.

Weijian-Espressif, 2018-12-17:
Please provide your test log, thanks.

AadiMehta, 2018-12-17:
Hello @Weijian-Espressif I have mentioned that the issue is intermittent(appears 1 out of 10 times approx). Issue is when it appears, esp32 gets paniced and reboots. I have attached the test log in the 1st comment.

I have not made any changes in the blufi example and am running that with my project. As per tracelogs, issues seems to be pointing towards LoadStoreError when blufi example tries to save private key(mbedtls_dhm_make_public function in blufi_security file).

Weijian-Espressif, 2018-12-17:
AadiMehta , I use blufi APP to test the blufi example, i did not reproduce the issue. Please provide a complete test log, starting from the first test, thanks.

Weijian-Espressif, 2019-02-15:
@AadiMehta, I use blufi APP to test the blufi example, but still haven't reproduce this issue. Do you test blufi demo with our blufi APP?  Android or iOS ?

AadiMehta, 2019-02-21:
@Weijian-Espressif I am getting still getting this issue through both ios and android. 

I think it fails when I am sending custom data which is large in size(around 1400 bytes). I am sending a certificate in the custom data.

Weijian-Espressif, 2019-02-22:
@AadiMehta, I found the bug, The length of data sent by the APP exceeds 1024. We will modify the APP code, and the data needs to be fragmented when sending custom data.

Weijian-Espressif, 2019-02-22:
@AadiMehta, I am not sure if the bug I found is the same as yours. 
1.  please update your blufi app source code from github 
2. pacth your ESP-IDF
[blufi.patch.zip](https://github.com/espressif/esp-idf/files/2892745/blufi.patch.zip)

AadiMehta, 2019-02-22:
@Weijian-Espressif With android the failures are intermittent(3 out of 10 attempts are failed). When passed following is the sample output:

```
I (51213) HA-Node.api_ap: Free HEAP: 113432 bytes
I (51216) esp_https_server: Starting server
I (51286) esp_https_server: Server listening on port 443
I (51290) HA-Node.api_ap: Registering URI handlers
I (51298) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (109868) HA.Node.blufi: BLUFI ble connect

I (112367) HA.Node.blufi: Recv Custom Data 1303

I (112371) Custom Data in char : {"gpio_pins":4,"
I (112374) Custom Data in char : gateway_id":"2fb
I (112378) Custom Data in char : 9ad636f1a4d10a6d
I (112382) Custom Data in char : 09080fc6d6587","
I (112387) Custom Data in char : gateway_ip":"192
I (112446) Custom Data in char : .168.2.4","ca":"
I (112450) Custom Data in char : -----BEGIN CERTI
I (112453) Custom Data in char : FICATE-----\nMII
I (112456) Custom Data in char : DNTCCAh2gAwIBAgI
I (112464) Custom Data in char : UT62kEuXXdmxTqPX
I (112468) Custom Data in char : zNwCEUQ+eT60wDQY
I (112472) Custom Data in char : JKoZIhvcNAQEL\nB
I (112477) Custom Data in char : QAwMTEvMC0GA1UEA
I (112483) Custom Data in char : wwmMmZiOWFkNjM2Z
I (112488) Custom Data in char : jFhNGQxMGE2ZDA5M
I (112494) Custom Data in char : DgwZmM2ZDY1ODcu\
I (112499) Custom Data in char : nbG9jYWwwHhcNMTk
I (112559) Custom Data in char : wMjIxMTEwNTI3Whc
I (112563) Custom Data in char : NMjkwMjE4MTEwNTI
I (112566) Custom Data in char : 3WjAxMS8wLQYDVQQ
I (112570) Custom Data in char : D\nDCYyZmI5YWQ2M
I (112575) Custom Data in char : zZmMWE0ZDEwYTZkM
I (112580) Custom Data in char : DkwODBmYzZkNjU4N
I (112590) Custom Data in char : y5sb2NhbDCCASEwD
I (112593) Custom Data in char : QYJ\nKoZIhvcNAQE
I (112596) Custom Data in char : BBQADggEOADCCAQk
I (112602) Custom Data in char : CggEAMWOQjYEXMbE
I (112607) Custom Data in char : Hma9V1krQXluhfgh
I (112613) Custom Data in char : qrOwZ\nKE1m2+sly
I (112667) Custom Data in char : 6B7UJyrbwaM0WIZW
I (112669) Custom Data in char : Ow5DWWC94ShIQfh8
I (112672) Custom Data in char : \/wPgPOqUAaAf\/T
I (112677) Custom Data in char : wJk\/6waan\ntgFd
I (112682) Custom Data in char : i0bYfgYW\/yg2GS2
I (112688) Custom Data in char : NtMKDG\/yfMiRMh+
I (112694) Custom Data in char : JuiS2Hi\/9pO668h
I (112698) Custom Data in char : DoAfivC0hYXjHHh\
I (112704) Custom Data in char : nXgLcy9\/hg3I9Qu
I (112709) Custom Data in char : hzIXyZEd6zQWSq2t
I (112715) Custom Data in char : U8EIIiT43vAO0Kht
I (112720) Custom Data in char : 3f3nw3zwsO\/kTiM
I (112773) Custom Data in char : vD+\nJjqPvQ8RSFt
I (112776) Custom Data in char : G7NdOlf57o97r4GP
I (112779) Custom Data in char : Q2eMdjEETG9KIlzE
I (112783) Custom Data in char : pIWLOwWnmJhdfABh
I (112789) Custom Data in char : Bblma\npHVuVn8Py
I (112794) Custom Data in char : S2R22rprPpFYYoGJ
I (112799) Custom Data in char : GiAycL6\/QD5ZRAS
I (112805) Custom Data in char : pPrawSz\/u1u5PQI
I (112810) Custom Data in char : DAQABo0Yw\nRDAPB
I (112816) Custom Data in char : gNVHRMECDAGAQH\/
I (112821) Custom Data in char : AgEAMDEGA1UdEQQq
I (112826) Custom Data in char : MCiCJjJmYjlhZDYz
I (112881) Custom Data in char : NmYxYTRkMTBh\nNm
I (112884) Custom Data in char : QwOTA4MGZjNmQ2NT
I (112887) Custom Data in char : g3LmxvY2FsMA0GCS
I (112892) Custom Data in char : qGSIb3DQEBCwUAA4
I (112897) Custom Data in char : IBAQAGo6elGrBK\n
I (112902) Custom Data in char : Z2oXaUFRICtVSST0
I (112908) Custom Data in char : ltZTHQ9fOjXI6FO2
I (112913) Custom Data in char : qyeqyMsZpoaTJ2eU
I (112918) Custom Data in char : Vu4znnN3Pt2BDx+z
I (112924) Custom Data in char : \nr1C0aJ\/t4NVky
I (112929) Custom Data in char : Pus\/jxh+kZVDRll
I (112935) Custom Data in char : 4qNfkhSQZEQQonRR
I (112987) Custom Data in char : pn6DcUdKk8FDLeNA
I (112990) Custom Data in char : wWG5\nrOdZmrV9BO
I (112993) Custom Data in char : zEjGi+eF4HCIkpBA
I (112997) Custom Data in char : iw01HkJgMoa7ZP1M
I (113003) Custom Data in char : H2AmK+fvL30zdcX6
I (113008) Custom Data in char : OCxnaS\nFDCB4vX+
I (113013) Custom Data in char : 4Dj74MKIh8IolrzC
I (113019) Custom Data in char : LfoSrB81o4ny33B8
I (113024) Custom Data in char : s5lYmjOgaeOyBc3k
I (113029) Custom Data in char : k0lQuwHX\n1hmhY7
I (113035) Custom Data in char : r5q8Rvwt2TuWSr4F
I (113040) Custom Data in char : yjJXbE1lbEEa15GF
I (113094) Custom Data in char : h+F8z56BmRU2mvYy
I (113097) Custom Data in char : SQTBuEPznS\nAzj3
I (113099) Custom Data in char : dagYdKao\n-----E
I (113104) Custom Data in char : ND CERTIFICATE--
I (113109) Custom Data in char : ---\n"}
```

When failed, I have posted the error code above.

Note: It always fails with iOS. Also I am sending the whole data in one custom data call. For IOS we have filed another bug (https://github.com/EspressifApp/EspBlufiForiOS/issues/9)

Weijian-Espressif, 2019-02-22:
@AadiMehta, Our iOS engineers are checking their source code

Weijian-Espressif, 2019-02-25:
@AadiMehta , ios source cod has updated in github, please update and try.
The crash issue, we do not reproduce it with our blufi demo. Can you test our blufi demo?

AadiMehta, 2019-02-25:
@Weijian-Espressif sure, I will try it again and let you know the results by the end of the day. Also as mentioned before, this issue is intermittent(fails 3 out of 10 times approx). I will try to test the given example as well with our workflow and will check if issue still persists

Weijian-Espressif, 2019-02-27:
@AadiMehta, Do you reproduce the issue using our blufi demo ?

AadiMehta, 2019-02-28:
@Weijian-Espressif Following are some observations. We are using the IOS app provided by the Espressifs. Following are the points:
- We are sending all the data in one fragment
- All calls(connect -> send custom data -> send ssid -> send pwd) are made back to back

**While connecting to the ESP32 the following errors are observed(one of them occurs each time)
This happens 4 out of 5 times we try to connect**

**1############################################################**
I (8786) HA.Node.blufi: BLUFI ble connect

***ERROR*** A stack overflow in task btuT has been detected.
abort() was called at PC 0x400900d8 on core 0
0x400900d8: vApplicationStackOverflowHook at /Users/sheky/Documents/Developer/Esp/esp-idf/components/esp32/panic.c:707

Backtrace: 0x4008fdc7:0x3ffe03b0 0x400900c1:0x3ffe03d0 0x400900d8:0x3ffe03f0 0x400928da:0x3ffe0410 0x40093fa4:0x3ffe0440 0x40093f5a:0x00000000
0x4008fdc7: invoke_abort at /Users/sheky/Documents/Developer/Esp/esp-idf/components/esp32/panic.c:707

0x400900c1: abort at /Users/sheky/Documents/Developer/Esp/esp-idf/components/esp32/panic.c:707

0x400900d8: vApplicationStackOverflowHook at /Users/sheky/Documents/Developer/Esp/esp-idf/components/esp32/panic.c:707

0x400928da: vTaskSwitchContext at /Users/sheky/Documents/Developer/Esp/esp-idf/components/freertos/tasks.c:3537

0x40093fa4: _frxt_dispatch at /Users/sheky/Documents/Developer/Esp/esp-idf/components/freertos/portasm.S:406

0x40093f5a: _frxt_int_exit at /Users/sheky/Documents/Developer/Esp/esp-idf/components/freertos/portasm.S:206

Rebooting...

**2############################################################**
I (63988) HA.Node.blufi: BLUFI ble connect

Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
Core 0 register dump:
PC      : 0x40092b21  PS      : 0x00060433  A0      : 0x80094832  A1      : 0x3ffe04f0  
0x40092b21: uxPortCompareSet at /Users/sheky/Documents/Developer/Esp/esp-idf/components/freertos/tasks.c:3537
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /Users/sheky/Documents/Developer/Esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /Users/sheky/Documents/Developer/Esp/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at /Users/sheky/Documents/Developer/Esp/esp-idf/components/freertos/tasks.c:4231

A2      : 0x40081f88  A3      : 0x00060423  A4      : 0x00060420  A5      : 0x00000001  
0x40081f88: _xt_user_exit at /Users/sheky/Documents/Developer/Esp/esp-idf/components/freertos/xtensa_vectors.S:785

A6      : 0x67ae7ae1  A7      : 0xc20cd392  A8      : 0x0000abab  A9      : 0xb33fffff  
A10     : 0x0000cdcd  A11     : 0x00060420  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x0000cdcd  A15     : 0x00000008  SAR     : 0x0000001f  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40081f88  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  
0x40081f88: _xt_user_exit at /Users/sheky/Documents/Developer/Esp/esp-idf/components/freertos/xtensa_vectors.S:785

Backtrace: 0x40092b21:0x3ffe04f0 0x4009482f:0x3ffe0510 0x40094863:0x3ffe0530 0x40082448:0x3ffe0560 0x400825b3:0x3ffe0580 0x40082798:0x3ffe05a0 0x400db799:0x3ffe05c0 0x400d82c3:0x3ffe05e0 0x400d83a7:0x3ffe0600 0x400d8c2b:0x3ffe0620 0x400d8cfd:0x3ffe0670 0x400d904c:0x3ffe06a0 0x400d9197:0x3ffe0730 0x400d9399:0x3ffe0750 0x400da09e:0x3ffe0dd0 0x400d7d0c:0x3ffe0e00 0x4010242f:0x3ffe0e50 0x401018d3:0x3ffe0ea0 0x40101abf:0x3ffe0ee0 0x401294aa:0x3ffe1170 0x4011e0a5:0x3ffe11b0 0x4011c98d:0x3ffe11d0 0x4011cf89:0x3ffe1490 0x4011d3d2:0x3ffe14f0 0x4011b8dd:0x3ffe1540 0x4011b995:0x3ffe1560 0x40122c83:0x3ffe1580 0x40116dc1:0x3ffe15c0
0x40092b21: uxPortCompareSet at /Users/sheky/Documents/Developer/Esp/esp-idf/components/freertos/tasks.c:3537
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /Users/sheky/Documents/Developer/Esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /Users/sheky/Documents/Developer/Esp/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at /Users/sheky/Documents/Developer/Esp/esp-idf/components/freertos/tasks.c:4231

0x4009482f: multi_heap_internal_lock at /Users/sheky/Documents/Developer/Esp/esp-idf/components/heap/multi_heap.c:699

0x40094863: multi_heap_malloc_impl at /Users/sheky/Documents/Developer/Esp/esp-idf/components/heap/multi_heap.c:699

0x40082448: heap_caps_malloc at /Users/sheky/Documents/Developer/Esp/esp-idf/components/heap/heap_caps.c:231

0x400825b3: heap_caps_calloc at /Users/sheky/Documents/Developer/Esp/esp-idf/components/heap/heap_caps.c:231

0x40082798: esp_mbedtls_mem_calloc at /Users/sheky/Documents/Developer/Esp/esp-idf/components/mbedtls/port/esp_mem.c:25

0x400db799: mbedtls_calloc at /Users/sheky/Documents/Developer/Esp/esp-idf/components/mbedtls/mbedtls/library/platform.c:59

0x400d82c3: mbedtls_mpi_grow at /Users/sheky/Documents/Developer/Esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:676

0x400d83a7: mbedtls_mpi_copy at /Users/sheky/Documents/Developer/Esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:676
[... 101 lines cut ...]
0x4011c98d: gatts_process_write_req at /Users/sheky/Documents/Developer/Esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_sr.c:1159

0x4011cf89: gatts_process_attribute_req at /Users/sheky/Documents/Developer/Esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_sr.c:1468

[... cut ...]
-----------------------------------------------------------------------------------------

**Upon tweaking the below ESP32 settings, the Bluetooth connect get stable with only occasional error
    => Component config -> Bluetooth -> Bluetooth enable
        Bluetooth Bluedroid Host Stack task stack size: Changed from 3072 to 4096
        Use dynamic memory allocation in BT/BLE stack: Changed from False to True**

With above settings StoreProhibited error with above stack trace still occurs(1 out of 7-8 times). Please let me know if you require me to try out any more cases

Weijian-Espressif, 2019-02-28:
Hi @AadiMehta, 
1.   Please test our blufi demo in ESP_IDF and do not modify menuconfig file.
2.  Please use Android blufi app, do not modify the source code. (iOS app have a bug about sending custom data )
3.  Please provide  a complete log, from the beginning to the end.

AadiMehta, 2019-02-28:
@Weijian-Espressif In the above comment, I have mentioned that we have tried first without modifying the menuconfig file, which resulted in the above errors. Also we have tried iOS app with the new changes for sending custom data(which was committed 2-3 days back). We will try with Android too. 

We will try with the exact app and will provide you more details in the log.

Alvin1Zhang, 2019-03-15:
Would you help share if any updates? Thanks. @AadiMehta

AadiMehta, 2019-03-15:
@Alvin1Zhang We confirmed that blufi process is working as expected with iOS app. However we are trying the same with Android application and should get back once we have more info on this.

Weijian-Espressif, 2019-03-22:
@AadiMehta, We tested our native blufi demo and didn't reproduce the issue, how is your test result?

Weijian-Espressif, 2019-03-22:
@AadiMehta, Do you set Optimization Level (Release (-Os)) in menuconfig?

AadiMehta, 2019-03-25:
@Weijian-Espressif yes, the optimization level has been set to Release

AadiMehta, 2019-03-27:
Works fine with the espressif's android app as well. Closing this

closed 2019-03-27 by AadiMehta