espressif/esp-idf #2831: tzset for CountryName/CityName is not working

hvishu, 2018-12-13:
Hi
We are using IDF 3.0 release

What we see is that using "CountryName/CityName" format is not getting the correct timeinfo.  I have taken the sntp_example (attached) and run through all the timezones available and as you can see in the log file (output_onESP32.txt, attached), it is working only for some formats and not all.

I did the same example with gcc on ubuntu16.04 (timeZone2.c, and output file is output_onUbuntu16.txt) it works fine.  

Could you help in this regard?

[output_onESP32.txt](https://github.com/espressif/esp-idf/files/2675991/output_onESP32.txt)
[output_onUbuntu16.txt](https://github.com/espressif/esp-idf/files/2675992/output_onUbuntu16.txt)
[sntp_example_main.c.zip](https://github.com/espressif/esp-idf/files/2675998/sntp_example_main.c.zip)
[timeZone2.c.zip](https://github.com/espressif/esp-idf/files/2675996/timeZone2.c.zip)

Regards,
Vishwa

igrr, 2018-12-13:
Newlib tzset supports the second format described in https://www.gnu.org/software/libc/manual/html_node/TZ-Variable.html. If you need to map city names to timezone information, you can use something like https://github.com/nayarsystems/posix_tz_db, and convert standard tzdata database to the format which newlib can accept. Due to application size constraints, you may want to additionally filter out the regions from the database which aren't relevant to your product.

hvishu, 2018-12-14:
Thanks for the reply @igrr 
https://github.com/nayarsystems/posix_tz_db is helping.  But few things still doesn't work.  For instance
**Asia/Colombo                    <+0530>-5:30         is: Thu Dec 13 -498:06:08 2018**    
Looks like the ESP library doesn't do right way for the entries in the format <+0530>-5:30 or <+08>-8

Are you aware of any other database, which I can test and confirm?

Regards,
Vishwa

igrr, 2018-12-14:
Could you clarify which exact values of TZ variable do not produce the expected result?

hvishu, 2018-12-14:
As you could see from the attached log, Anything in this format <+0530>-5:30 is not correct.  
Ex
America/Argentina/Buenos_Aires  
America/Argentina/Catamarca     
America/Argentina/Cordoba       
America/Argentina/Jujuy         
America/Argentina/La_Rioja      
America/Argentina/Mendoza       
America/Argentina/Rio_Gallegos  
America/Argentina/Salta         
America/Argentina/San_Juan      
America/Argentina/San_Luis      
America/Argentina/Tucuman       
America/Argentina/Ushuaia       
Asia/Colombo
etc...

This log was taken sometime back, you can take the reference of NY and Shanghai team.
I (6081) example: The current date/time in New York is: Fri Dec 14 03:06:08 2018
I (6091) example: The current date/time in Shanghai is: Fri Dec 14 16:06:08 2018

[timezone_log.txt](https://github.com/espressif/esp-idf/files/2679576/timezone_log.txt)

Alvin1Zhang, 2019-03-22:
@hvishu Hi, would you help share if any updates for the issue? Thanks.

hvishu, 2019-04-04:
As mentioned by @igrr, replaced the angular bracket with some string and it works.

closed 2019-04-04 by hvishu