espressif/esp-idf #2832: Is this new HTTP Server is ready for production use

snahmad, 2018-12-13:
Hi,

Can I use this Web Server
[](https://github.com/espressif/esp-idf/tree/master/components/esp_http_server)

Does this support uploading files similar to mongoose web server.
Does this allow to browser file system?
Does this support user /cookie authentication?
Thanks,
Naeem

anurag-kar, 2018-12-17:
Hi @snahmad. The **esp_http_server** provides all these features. You can find a basic example here : https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/simple.

> Does this support uploading files

Yes. You must register a URI handler for file uploading and inside the handler access the contents of the request packet from httpd_req_t structure. Following is a very simplified implementation

```
esp_err_t upload_post_handler(httpd_req_t *req)
{
    char filename[32];
    char buffer[256];

    // Get file name length
    size_t filename_len = httpd_req_get_url_query_len(req);

    // Save the filename
    httpd_req_get_url_query_str(req, filename, sizeof(filename));

    // Receive contents into buffer
    int recv_len = httpd_req_recv(req, buffer, req->content_len);

    // Note : recv_len <= req->content_len always (-ve on error)
    // In a typical scenario httpd_req_recv() should 
    // be called in a loop till the complete file is received

    /* ---------- Code to store the file comes here ------------- */
    .
    .
    .

    // Send HTTP response with success message
    httpd_resp_send(req, "File uploaded successfully", -1);
    return ESP_OK;
}
```

Register the URI handler as follows

```
     httpd_uri_t file_upload = {
        .uri       = "/upload",
        .method    = HTTP_POST,
        .handler   = upload_post_handler,
        .user_ctx  = NULL
    };
    httpd_register_uri_handler(server, &file_upload);
```

Now you can upload a file by sending a POST command, eg. using curl `curl -X POST --data-binary @abcd.pdf http://10.0.0.152:80/upload?books/abcd.pdf`. The part `books/abcd.pdf` is received as the query string.

> Does this allow to browser file system?

You can use SPIFFS for storing the files on flash, or use SDMMC driver to store them on an SD card. See :

1. https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/sdmmc.html
2. https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/spiffs.html

After initializing your storage using either of the methods, you can access a file like `FILE *f = fopen("/sdcard/filename.ext", "r")` or `FILE *f = fopen("/spiffs/filename.ext", "r")`

Then, you must write a URI handler to accept GET requests for getting the files from the storage.

> Does this support user /cookie authentication?

This can be done easily by writing a URI handler for user authentication, inside which the username and password can be accessed from the request body, then generate a token and send that as the response header as follows :

```
    httpd_resp_set_hdr(req, "Cookie", "access_token=generated_token");
    httpd_resp_send(req, NULL, 0);  // Response body can be empty
```

The complete documentation is here : 
1. https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/esp_http_server.html
2. https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/esp_https_server.html

snahmad, 2018-12-17:
ok, Thanks for information. I will use forum https://esp32.com/

closed 2018-12-17 by mahavirj