espressif/esp-idf #2836: vTaskDelay(1) wrong first time

michprev, 2018-12-14:
## Environment

- IDF version: v3.3-dev-180-g4c881708d
- Compiler version: 1.22.0-80-g6c4433a

## Problem Description

`vTaskDelay(1)` does not wait 1 FreeRTOS tick for the first time.

### Code to reproduce this issue

CPU frequency must be set to 240 MHz
FreeRTOS tick rate must be 1000 Hz

```cpp
#include <esp_timer.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>

void app_main(void)
{
    uint32_t start, end;

    while (true)
    {
        asm volatile ("rsr.ccount %0" : "=r"(start));
        vTaskDelay(1);
        asm volatile ("rsr.ccount %0" : "=r"(end));

        // should be around 1000 us
        if ((end - start) / 240 < 500)
            printf("error %d\n", (end - start) / 240);
    }
}
```

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2681038/sdkconfig.txt)

igrr, 2018-12-15:
This seems to be standard FreeRTOS behavior: vTaskDelay(1) will wait for the closest tick, vTaskDelay(0) is equivalent to yielding.

https://www.freertos.org/FreeRTOS_Support_Forum_Archive/December_2012/freertos_vTaskDelay_0_vs_vTaskDelay_1_6518023.html

michprev, 2018-12-15:
That makes sense.

closed 2018-12-15 by michprev