espressif/esp-idf #2837: LoadStoreAlignment exception in http task

asweeney99, 2018-12-14:
### Environment
- Development Kit:      [Custom Board]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version:  [v3.2-beta1-182-g4c881708d]
- Build System: [Make]
- Compiler version: [5.2.0]
- Toolchain version: [crosstool-ng-1.22.0-80-g6c4433a] 
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
Two RTOS tasks are created in main(). The first is an AWS iot task using MQTT. The second task is a http task which checks a different server for a firmware update. The http task runs fine just once, before the connection to AWS iot. After AWS connection is setup, the http task throws a LoadStoreAlignment exception when called.

### Expected Behavior
The http task should successfully contact the server and check for firmware update even when the device is connected to AWS iot.

### Actual Behavior
A LoadStoreAlignment exception is thrown in the http task.

### Things I've tried
I tried using xTaskPinnedToCore assiging the tasks to separate cores and also the same cores but this didn't help. 

### Steps to repropduce
1. make -j flash
2. make monitor

###  Setup/wiring
GPIO pin 25 of esp32 is attached to a DHT22 thermostat.
Serial interface is via TX, RX, GND pins to CP2102 adapter.

### Code to partially reproduce this issue
- main() from thermostat.cpp
[https://gist.github.com/asweeney99/435169f85e7ff2849d728f59b0cdf3eb.js](url)
- firmware.cpp
[https://gist.github.com/asweeney99/73aeeddaad6c6af018d78fb08359363a.js](url)

### Debug Logs
- backtrace
[https://gist.github.com/asweeney99/64e39ca4745e9e3e91667ec0cc768d3b.js"](url)

### Other items 
- sdkconfig file 
[https://gist.github.com/asweeney99/dd3725638665f55bd77258355a7237a6.js](url)

mahavirj, 2018-12-15:
@asweeney99 

What happens if you change line at https://gist.github.com/asweeney99/73aeeddaad6c6af018d78fb08359363a#file-gistfile1-txt-L71 to `esp_http_client_config_t config = {0}`?

asweeney99, 2018-12-15:
Thank you for your suggestion. I'll try it as soon as possible.

asweeney99, 2018-12-15:
```esp_http_client_config_t config = {0}``` makes no difference. The backtrace is the same.

asweeney99, 2018-12-15:
Another clue. If I comment out the line ```xTaskCreate(&aws_iot_task, "aws_iot_task", 9216, NULL, 5, NULL);``` and just create the http task only, the task will run successfully once and then fail on the second try with the same backtrace. So it seems not to be related to the AWS iot task at all.

```[I][thermostat.cpp:120] event_handler(): Connected to AP!

I (2586) event: sta ip: 192.168.8.101, mask: 255.255.255.0, gw: 192.168.8.1
[I][thermostat.cpp:115] event_handler(): Our IP address is 192.168.8.101

Connected to wifi network
I (2593) codemaker_thermostat: Current firmware version: 0.2
I (2599) codemaker_thermostat: Checking if new firmware is available...

current firmware version (0.2) is greater or equal to the available one (0.2), nothing to do...

I (13410) codemaker_thermostat: Current firmware version: 0.2
I (13410) codemaker_thermostat: Checking if new firmware is available...

assertion "res == coreID || res == portMUX_FREE_VAL" failed: file "/home/andrew/esp/esp-idf/components/freertos/portmux_impl.inc.h", line 105, function: vPortCPUAcquireMutexIntsDisabledInternal
abort() was called at PC 0x400d3ad3 on core 1
```

asweeney99, 2018-12-15:
I thought that maybe I need to explicitly initialize each member of the struct ```esp_http_client_config_t``` in order like this, but this makes no difference.

```
esp_http_client_config_t ota_client_config = {
                                                                url : file->valuestring,
                                                                host : NULL,
                                                                port : NULL,
                                                                username : NULL,
                                                                password : NULL,
                                                                auth_type : HTTP_AUTH_TYPE_NONE,
                                                                path : "/",
                                                                query : NULL,
                                                                cert_pem : server_cert_pem_start,
                                                                method : HTTP_METHOD_GET,
                                                                timeout_ms : NULL,
                                                                disable_auto_redirect : NULL,
                                                                max_redirection_count : NULL,
                                                                event_handler : _http_event_handler,
                                                                transport_type : HTTP_TRANSPORT_UNKNOWN,
                                                                buffer_size : NULL,
                                                                user_data : NULL,
                                                                is_async : false
                                                        };
```

mahavirj, 2018-12-16:
@asweeney99 
There is definitely memory corruption somewhere. What is size of `evt->data_len` at https://gist.github.com/asweeney99/73aeeddaad6c6af018d78fb08359363a#file-gistfile1-txt-L48, any chance of buffer overflow here?

asweeney99, 2018-12-16:
Thanks Mahavir for looking over my code. I think you might be right about buffer overflow. I'll report back.

asweeney99, 2018-12-17:
Yes buffer overflow was the problem here:
```
// receive buffer
char rcv_buffer[200]; //too small
```

Luca Dentella has a nice tutorial how to use your ESP32 to check your app's firmware version here http://www.lucadentella.it/en/2018/10/27/esp32-37-ota-via-https/ Just don't forget to increase the buffer size for your JSON file if you need to!

Thanks to Mahavir. The issue can be closed.

closed 2018-12-17 by mahavirj