espressif/esp-idf #2838: BT: GATTS: Indication response missing attr handle

rojer, 2018-12-14:
If `esp_ble_gatts_send_indicate` is invoked with need_confirm = true (i.e., actual indication request), `ESP_GATTS_CONF_EVT` is raised with attr handle not set (esp_ble_gatts_cb_param_t.conf.handle = 0).

It works fine for notifications (`esp_ble_gatts_send_indicate(..., false)`) - handle is populated properly.

Tested on latest 3.2 release branch

rojer, 2018-12-14:
should add that i'm testing with "BLE Scanner" android app.

chegewara, 2018-12-15:
Its with master about month ago, still without answer:
https://github.com/espressif/esp-idf/issues/2552#issuecomment-437108715

Weijian-Espressif, 2018-12-17:
@rojer, @chegewara, sorry, I will fix the bug.
@chegewara,  If you find a new bug, please create a new issue.

Weijian-Espressif, 2019-02-14:
Hi @rojer @chegewara, this bug have fixed, please update to the latest IDF and try, thanks.

closed 2019-02-14 by Alvin1Zhang