espressif/esp-idf #2841: ble adv example code dont advertise in ios device

sjoteppagol, 2018-12-16:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x ] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:      [ESP32-Wrover-Kit
- Kit version (for WroverKit
- Module or chip used:  [|ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): ESP-IDF v3.0-dirty 2nd stage bootloader
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     [Windows]
- Power Supply:         [external 5V]

## Problem Description
The Bluetooth advertisement of the device is not  in ios mobile, where as on android it works fine. 

//Detailed problem description goes here.
The Bluetooth advertisement is visible on android devices but it just don't work on ios devices. On the same ios device if we use nRF connect app then we can see the device's Bluetooth advertisement. I am not getting what's wrong with ios devices why it is not showing Bluetooth advertisement.

### Expected Behavior
The Bluetooth advertisement should be visible in list of available Bluetooth devices. 

### Actual Behavior
On the ios devices the Bluetooth advertisement is not visible.

### Steps to repropduce
1. step1: Build ble_adv code given in example code of esp-idf
2. Turn on mobile Bluetooth and see available devices.
3. In android under list of available devices it shows ESP-BLE-HELLO but whereas in ios it is not listed.

Weijian-Espressif, 2018-12-17:
@sjoteppagol, you can use nrf connect app or lightblue APP to show adv. iOS does not display BLE adv in Bluetooth settings.

sjoteppagol, 2018-12-17:
@Weijian-Espressif  Thank you. 
Closing this thread.

closed 2018-12-17 by sjoteppagol